pcb2gcode/pcb2gcode #99: Request: stencil cutting

ldam42, 2018-02-08:
Would love to see it possible to generate stencils from a paste mask output.
...for that next step after prototyping, when you order small batches from a manufacturer

eyal0, 2018-02-09:
A stencil is a gerber file like any other, right, with cut-outs where the paste needs to go.

If you ran pcb2gcode with the B.Paste file where the B.Cu file goes, it would make gcode to cut out a stencil, right?  What tool would you load into your CNC for cutting the stencil?  What material is the stencil made from?

ldam42, 2018-02-10:
There's a small problem inputting the paste as route layout; as the cutting goes outside the copper.

E.g. having 2 pads at 2x2mm with a spacing of 1mm between, and using a 0.5mm endmill;  the outside of the pads that will get cut away it means i will end up with one hole of 3x6mm instead of the two holes of 2x2 mm with a 1mm barrier in between - just an example but it should be clear that template cutting is different than route cutting.

Besides: I did a quick trial on a .15mm alu with a 30deg vcutter [stepcraft 300]. Gave some rough edges but would be useful for a few boards - a proper endmill might give a cleaner cut; I also read that similar width mylar sheets would be cheap/quick  for a small batch; i'm not sure though how nice this would be cut (ordered a pack to test it out though)

eyal0, 2018-02-11:
That sounds like a problem with the CNC, not an issue with pcb2gcode, right?

What if you used a tool like this:

https://www.amazon.com/Silhouette-Cameo-Material-Cutting-Printer/dp/B0060D1LFO

?

What are stencils usually made from?

ldam42, 2018-02-11:
The CNC is not the problem, neither pcb2gcode.

The issue is that to cut a stencil, the path algorithm would have to be different than the routing algorithm for cutting pcb traces, as this algorithm in pcb2gcode removes material outside pads and traces, but the stencil algorithm should remove material inside pads (should be a simple algorithm even).

Normally a stencil is a steel plate 0.12mm, and with laser cut holes, and would cost from 30$ and up, with the usual waiting time, just like when ordering pcbs. For stencils that would only be used 1-5 times, this would be too expensive, and slow down the process a lot. 

For prototyping, using pcb2code, it would be nice feature to add, so it easily could support this part of the manufacturing process as well. And for a few usages, a vinyl/mylar stencil would in most cases be satisfactory.

Using a vinyl cutter as suggested may work as well, but buying a 300$ device for this purpose seems a bit overkill, if one already have a cnc cutter available.

corna, 2018-02-11:
It seems to me just a polarity issue: the Paste layers contains the pads where the soldering paste should be deposited, but to create a stencil we need a huge polygon with holes on the pads, which is the negative of the Paste layer.

It should be easily doable by adding an additional boolean parameter "invert" in GerberImporter::generate_layers: when "true" the [polarity](https://github.com/pcb2gcode/pcb2gcode/blob/master/gerberimporter.cpp#L600) of each item is inverted.

eyal0, 2018-02-11:
@ldam42 pcb2gcode already has an "offset" argument that lets you specify how far from the trace to mill.  If the milling tool is width 12 thou then you set offset to 6 thou and the milling will happen 6 thou outside from the trace.

However, if you set offset to negative 6 thou, it will mill *inside* the line.  This accomplishes the same as what @corna said, right?

I already have a PR that will allow negative offsets: https://github.com/pcb2gcode/pcb2gcode/pull/93

@ldam42 If it were possible to specify a negative milling offset and you set the milling layer to the F.Paste, would that solve the problem for you?

ldam42, 2018-02-11:
Using negative offset may do the trick; i'll checkout the PR, and give you some feedback

@christoph1703 did some drilling test; seems like drilling stencils is also a very viable way; would love to check your project, but the account is protected.

ldam42, 2018-02-11:
@eyal0 - Had to make a brief test;
Building PR93, and running twice, once with a positive offset, and second a negative offset;
It's clear that the output produced using negative offset will give the desired result - using a ballpoint pen in the cnc shows the correct outline of a stencil, so i am confident that the gcodes produced are fine.

The big issue for me now is of practical nature and is to do some test runs with bits/materials to produce a satisfactory result.

Thanks for the support, this request is fulfilled from my point :)

closed 2018-02-11 by ldam42

eyal0, 2018-02-11:
Fantastic!  I'm happy to hear it.

I think that a Silhouette machine like I linked uses a sharp tip that doesn't rotate.  But it just cuts paper and fabric.  If stencils are usually metal then you'll need something else.  I've never used a stencil, I don't know.