pcb2gcode/pcb2gcode #100: Unnecessary travel added to milling code

oyvindkinsey, 2018-02-13:
Here's a comparison between two files created by pcp2gcode and flatcam respectively, and as you can see, the one created by pcb2gcode contains large amounts air travel.

[image]

[image]

As the remainder of the settings were comparable (manually edited to have the same feed, heights etc), it seems the extra travel time is considerably adding to the total time (by a factor of 30%).

eyal0, 2018-02-13:
I have a PR to optimize the algorithm further but I haven't patched it in yet.

In the meanwhile, do you have sample input files that you can send me for testing?

eyal0, 2018-02-13:
Also, which version are you using?  That travel that you're showing is *really* bad.  Even the latest version without my changes should be doing better than that!

eyal0, 2018-02-13:
Oops, I found a glaring typo.  I hope that it solves your problem!  Please check and let me know.

oyvindkinsey, 2018-02-13:
This was using master. 

I've attached the gerber files.

[CNC PCB.zip](https://github.com/pcb2gcode/pcb2gcode/files/1720866/CNC.PCB.zip)

oyvindkinsey, 2018-02-13:
I'll update master when I get home and try again! Using pcb2gcode is still faster when configured correctly, since it only does one pass to separate two polygons, and not two like Flatcam. With more efficient travel, I could get this down way under an hour (currently doing very conservative feeds etc).

oyvindkinsey, 2018-02-13:
@eyal0, thanks for being super responsive on these things, it really shows the value from actively filing issues like this!

oyvindkinsey, 2018-02-14:
bingo!

closed 2018-02-14 by oyvindkinsey

eyal0, 2018-02-14:
> it only does one pass to separate two polygons

@oyvindkinsey Can you explain what you mean?

I'm working on optimizations to make it choose an even better path and I want to learn more.

Thanks!

oyvindkinsey, 2018-02-14:
@eyal0, normally two adjacent polygons gets isolated individually, even if they share one or more lines, this will cause a mill operation in e.g a copper fill to both mill out the contour of the trace, but also the contour of the surrounding copper fill. 

Isn't one of pcb2gcode's claims that it can avoid these redundant cuts?

eyal0, 2018-02-15:
With the new Eulerian paths update, overlapping paths are joined.

Thanks for the testing.

oyvindkinsey, 2018-02-16:
@eyal0, I saw, and the I really enjoyed watching how efficiently the machine moved around, moving between adjacent traces without even raising the spindle!

[image]
[image]

oyvindkinsey, 2018-02-16:
On a different note though, none of the thermal reliefs got routed, though they do show up on all the svg's. That partly explains why it took so much less time this time around - I'm pretty sure I just recompiled pcb2gcode and ran pcb2gcodeGUI without restarting it or changing settings, but somehow after the update, the reliefs are gone?

eyal0, 2018-02-17:
Let's open a new bug for this: https://github.com/pcb2gcode/pcb2gcode/issues/105

Now I understand what you mean by traces being milled twice.  It's because the outline of the trace was milled and then the outline of the ground plane was milled, too.  However, you've set the offset large enough that those two lines merged into one.  The downside is that the offset expansion also made the thermal relief holes disappear.

eyal0, 2018-02-21:
@oyvindkinsey Did you get to try https://github.com/pcb2gcode/pcb2gcode/pull/106 ?

oyvindkinsey, 2018-02-21:
@eyal0, no I haven't gotten around to that yet. I'll be creating some new boards soon though, and will give it a shot then!