pcb2gcode/pcb2gcode #105: Thermal reliefs not being milled

eyal0, 2018-02-17:
In versions before https://github.com/pcb2gcode/pcb2gcode/commit/07f4adc5a2a111929c54511b0f7e1ecae7a31bd2 , vectorial mode didn't correctly obey the polarity of layers in the gerber file so subtraction didn't work.

Here's the source:

[image]

A section with thermal relief used to look like this:

[image]

But after the fix, thermal reliefs starting working:

[image]

When `offset=0`, it looks fine.  But as the offset increases, pcb2gcode drills farther and farther away from the traces so that the tool doesn't cut the traces.  pcb2gcode does this by enlarging each region and then drilling around that.  The green, brown, and yellow regions are expanded and then the drilling happens on the outline.  For most traces, this works fine.  But for thermal relief it's a problem because the holes in the green shape shrink until they potentially disappear.  This is `offset=0.004`:

[image]

Here is `offset=0.006`:

[image]

The green and red layers have butt up against one another and the thermal reliefs have disappeared.  Setting `voronoi=true` would accomplish something similar, by drilling mid-line between the colored shapes.  In both cases, the thermal reliefs will disappear.

I'm not sure what the best solution to this problem is.  Does the PCB need to be made differently?  Or does pcb2gcode need to be modified for special handling of board with ground planes?

eyal0, 2018-02-17:
Possible solutions:

1. Larger thermal reliefs, bigger than the size of the milling tool.  That way, when the shape is expanded, the hole doesn't disappear.
2. Fill thermal reliefs, either in pcb2gcode or in CAD.  That would prevent pcb2gcode from expanding the holes into nothingness.
3. Let the thermal relief holes expand inward but instead of letting them disappear, let them become just lines.  That seems hard to do in code but would give a nice output.

Other ideas?  @oyvindkinsey looking at the first image above, with the colors, and knowing that the milling tool has a thickness, where do you expect milling to happen?  For the red and yellow shapes, it makes sense to mill *outside* the shape, that way the traces' width is preserved.  Where is the best place to mill for the thermal relief?  On the line, inside the hole, or outside the hole?

Option 2 is easy to do in pcb2gcode.  I'll add it as a flag and you can decide if you like it.

eyal0, 2018-02-18:
I wrote option 2.  It detects thermal reliefs by looking for holes that have no traces in them.

It works for thermal reliefs that don't overlap other areas but it doesn't work for others.

[image]

In the image above, the thermal reliefs work except for the one at the bottom right and the one above it. 
 Those two are overlapping the spacing around the brown area so they got merged into a single "hole". 
 And that hole has the brown trace inside of it so it's not empty and not recognized as a thermal relief.  :disappointed: I'm not sure how to detect those.

The code is here: https://github.com/eyal0/pcb2gcode/tree/preserve_thermal_relief

corna, 2018-02-19:
I think it's a problem that pcb2gcode shouldn't even try to solve: the user is asking here to create a thermal relief which is smaller than the minimum feature size (= the size of the milling tool). My opinion is that pcb2code should warn the user that some thermal reliefs have been removed (as it isn't possible to create them without milling away also a part of the track) and then continue.

eyal0, 2018-02-19:
What you say is true.  If he makes the thermal relief bigger, that will solve the problem, too.

It might still be useful to have a PR like mine for `voronoi=true` when people want voronoi milling but still want thermal relief.

closed 2018-06-10 by eyal0