pcb2gcode/pcb2gcode #108: milldrill should have separate z depth from outline cutting

planetclegg, 2018-02-25:
This is probably more of a (simple) enhancement request than a bug

It would be useful if milldrill either used --zdrill for it's final depth or it's own setting instead of the oultline --zcut depth.   My reasoning for this is to be able to milldrill completely thru the PCB and slightly into the waste board to ensure clean holes, but conversely leave a thin layer on outline cutting so that it doesn't start to cut into the wasteboard (the 'incomplete' outline cut can be easily finished with an exacto knife, but the milldrilling cannot)

For example, with a 1.6mm PCB, I would want to milldrill to -1.7mm, but only outline cut to -1.5mm, leaving the remaining .1mm to finish cutting manually.

eyal0, 2018-02-26:
That's absolutely right.

I'll work on it.

eyal0, 2018-02-26:
I looked at the outputs and the results look fine to me.  Let me know if you're able to test the code from this branch: https://github.com/eyal0/pcb2gcode.git

Thanks!

planetclegg, 2018-03-02:
Built branch milldrill_zdrill_depth on OSX.   This appears to address the issue. 

 I also see you fixed a problem where it would sometimes (based on value of cut-infeed) stop before hitting the target depth.   Although this sometimes changes the exact distance of each cut slightly (e.g. 0.15 cut-infeed with -1.6 depth results in 0.16 steps), I think this is vastly better than the previous behavior (would stop at -1.45 inexplicably).   I tried a few other values for cut-infeed and the difference always seemed negligible.

Can these fixes be merged into master?

eyal0, 2018-03-02:
The code was stopping early due to rounding issues.  If the cut-infeed is 0.3 and the total depth needed is -1.0, then it used to do: -0.1, then -0.4, then -0.7, then -1.0.  But due to rounding, sometimes the -1.0 would be -1.00000001 or something like that and it would skip it because it seems too low.  That I fixed.

I also cleaned up the depths.  Rather than have the first cut be thim (0.1 above) and then a bunch of 0.3, I changed it to evenly space the drilling.  In the case above, it would do -0.25, -0.5, -0.75, -1.0.  Each one is as big as possible without being larger than the cut-infeed.  I think that it's a good choice but I can do it differently if you have a better idea.

I'm surprised that you set cut-infeed to 0.15 steps and it did a 0.16 step!  It should only do smaller, never larger!  Let's fix this and them I'm ready to put it in, sure.  Maybe you can send me your millproject file so that I can debug the issue?  If it's really doing 0.16 steps when 0.15 is the programmed maximum, I'll fix it!

planetclegg, 2018-03-02:
The difference isn't large enough to matter to me, but here's my millproject and some sample output.

metric=true
metricoutput=true
basename=out
zwork=-0.15
zsafe=2.0
mill-feed=150

mill-vertfeed=40
mill-speed=1000
offset=0.25
extra-passes=2
zchange=2
nog81=true
zcut=-1.60
cutter-diameter=0.5
cut-feed=200

cut-vertfeed=40
cut-infeed=0.15
cut-speed=1000

drill-feed=50
drill-speed=1000
zdrill=-1.60
onedrill=true
milldrill=true
vectorial=true
postamble=millpostamble

---
Sample of drill output:

G0 X13.16200 Y-13.02200
G1 Z-0.16000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G1 Z-0.32000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G1 Z-0.48000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G1 Z-0.64000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G1 Z-0.80000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G1 Z-0.96000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G1 Z-1.12000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G1 Z-1.28000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G1 Z-1.44000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G1 Z-1.60000
G2  X13.16200 Y-13.02200 I-0.10000 J0.00000
G0 Z2.00000

eyal0, 2018-03-04:
I tested with your settings and it worked fine for me.  Yours does 10 steps of drilling and mine does 11.  Your steps are just larger than the 0.15 maximum cut-infeed that you specified and mine is just smaller.  Bizarre!

I think that the only way that it could happen is if the ceiling function in c++ is broken.  If you're good at debugging, I would like to understand it!  But it's not critical so if you don't how or can't be bothered, that's okay, too.

Are you on linux?

eyal0, 2018-03-04:
Ah, wait, maybe I sent you to the wrong branch?

It should have been: https://github.com/eyal0/pcb2gcode/tree/milldrill_zdrill_depth

Or maybe you figured that our already?

closed 2018-03-04 by eyal0

planetclegg, 2018-03-05:
I'm on OSX, building with libs from homebrew, so that could be the difference.   I did figure out that that was the branch to use.      Will this fix get merged into the mainline branch?

eyal0, 2018-03-05:
Already merged.  I'm still surprised that you didn't get the right answer and I can't figure out how because it means that std::ceil isn't working like it does on my system.  Makes no sense to me!  But if you're happy with the results, I guess that we're done here.