pcb2gcode/pcb2gcode #111: Mill away more (or all) unwanted copper

henryhallam, 2018-03-02:
New user, so forgive me if this already exists and I'm just not seeing it...

For ease of soldering, aesthetics and/or signal integrity, I would like the option to mill away the entirety of the copper that isn't specified as traces in the gerbers, rather than just the minimum amount to isolate the traces.  I understand this has downsides in machining time and tool life.

i.e. I would like the finished board to look as close as possible to the gerber - obviously inside corners with radius narrower than the milling tool won't be sharp, but apart from that it ought to be possible.

Alternatively, instead of going all the way to "remove all the unspecified copper", a variant on --extra-passes would be OK - the trouble with --extra-passes at the moment is that each extra pass expands blindly from the original contour, eating into neighboring traces. If it could be constrained to never expand far enough to encroach on other traces that would be OK.

Assuming this is not yet possible, I'd be happy to try to put together a pull request, but would appreciate suggestions for how to go about it - I'm not familiar with the codebase yet nor really the geometric/topological theories involved.

eyal0, 2018-03-03:
Extra passes doesn't eat into neighboring traces, it stops at the midline between neighbors, called the voronoi line. Can you check again?

The feature that you're talking about is called rubout in some programs, I've seen.

eyal0, 2018-03-07:
@henryhallam Were you able to check if using extra-passes doesn't eat into neighboring traces?  It shouldn't.

JozefHumaj, 2018-03-07:
Great idea would be possibility of **2 different tools** (with different diameter) 1st to remove most of cooper and second one for fine finish. I know you can generate two separate files for each diameter, but there is problem with probing after you etch pcb using first tool.

henryhallam, 2018-03-07:
Thanks @eyal0 - I don't have a good means of previewing / simulating the effects of the g-code, but on my physical board I noticed that when I increased --extra-passes from 1 to 3, in several regions where there were multiple closely-spaced parallel tracks, some of them ended up being milled almost entirely away, much thinner than the 10 mil track width specified in the gerbers.  However I now suspect that this was actually a result of the board and and CNC router not being very flat, so the tapered tool was penetrating deeper than I had expected on this region of the board and therefore cutting wider than intended.  I'll get a probe set up so I can try the auto-leveling function and report back.

eyal0, 2018-03-11:
I use bCNC to look at the milling path.  I also look at the .png files created in vectorial mode.  (I hope that you are using vectorial mode).  In both, the milling looks right.

chaotix-, 2018-10-15:
@henryhallam You can try CAMotics https://camotics.org/. It's a nice open source simulation software to check if the problem is with your machine or the generated G-Code.

eyal0, 2019-02-23:
@JozefHumaj Finally, pcb2gcode is able to use mulitple milling bits!  Just set --mill-diameters=2mm,1mm and it will use those, in that order.

closed 2019-02-23 by eyal0