pcb2gcode/pcb2gcode #125: Drill sizes reported in inches

laurivosandi, 2018-05-03:
Hi, in case of the latest Git commit even while using --metric and --metricoutput arguments the LinuxCNC for some reason shows drill sizes in inches. Some older G-code files produced with pcb2gcode show metric as expected. Here's an example of faulty G-code with latest revision:

```
( pcb2gcode 1.3.2 )
( Software-independent Gcode )

( This file uses 5 drill bit sizes. )
( Bit sizes: [0.031inch] [0.032inch] [0.039inch] [0.043inch] [0.091inch] )

G94       (Millimeters per minute feed rate.)
G21       (Units == Millimeters.)
G91.1     (Incremental arc distance mode.)
G90       (Absolute coordinates.)
S6000     (RPM spindle speed.)

G00 Z100.00000 (Retract)
T1
M5      (Spindle stop.)
G04 P2.00000
(MSG, Change tool bit to drill size 0.03100 inch)
M6      (Tool change.)
M0      (Temporary machine stop.)
M3      (Spindle on clockwise.)
G0 Z1.00000
G04 P2.00000
```

eyal0, 2018-05-04:
I'll take a look. This is probably related to the new feature that allows all inputs to be with units. For example, offset=0.01inches

eyal0, 2018-05-06:
@laurivosandi  The bit size units do not come from the --metric and --metricoutput arguments.  They come directly from the gerber input files.  If you are seeing bit sizes in inches, it is because the input files that you provided specified the drill sizes in inches.

Can you open your Excellon drill file (maybe it's called "___.drl" and see look at the first few lines to see if the format is METRIC or not?

silps, 2018-05-06:
@eyal0 thanks, I changed to mm now, looks metric 👍 

M48  
;DRILL file {KiCad 4.0.7} date 2018 May 06, Sunday 12:49:53  
;FORMAT={-:-/ absolute / metric / decimal}  
FMAT,2
METRIC,TZ
T1C1.000
T2C2.300
....

eyal0, 2018-05-06:
Okay.  So we confirmed that the output is based on the input file units.  Good!

Would you prefer that the output be based on `metricoutput`?  Maybe it makes more sense for the output drill sizes to be based on the `metricoutput` flag, and have them convert to/from metric based on that?

silps, 2018-05-06:
I have to test it out on the LinuxCNC and confirm it.

laurivosandi, 2018-05-06:
Hello, I think with --metricoutput flag it should output G-code in metric regardless of input units

eyal0, 2018-05-13:
Okay.  I'll work on that.  Unless you want to write the PR, which would be fine, too.  The fix is cosmetic, just changing how the comment outputs look.  The gcode itself is unaffected.

Here and other places in the file need to be fixed: https://github.com/pcb2gcode/pcb2gcode/blob/master/drill.cpp#L267

eyal0, 2018-05-27:
It's done.

closed 2018-05-27 by eyal0