pcb2gcode/pcb2gcode #140: Milldrill is wrong for small slots

eyal0, 2018-05-16:
This code tries to cut clear across a slot: https://github.com/pcb2gcode/pcb2gcode/blob/master/drill.cpp#L390

But it ignore zcut and does too deep a pass.  In the case of slots, it should be going back and forth, removing a little material at a time.

eyal0, 2018-05-18:
Fixed by https://github.com/pcb2gcode/pcb2gcode/pull/141

closed 2018-05-18 by eyal0

jbeuckm, 2018-06-09:
How do you make pcb2gcode generate slots? I'm using KiCad but could use Eagle...

eyal0, 2018-06-09:
I added slot support because I used a footprint in KiCad that had slots in it.  That slot shows up in the Excellon drill file with a G85 command.  The spec says that slots should be drilled as a series of holes and that's what pcb2gcode does in the case of slots.

You might need the latest version of gerbv for this to work because older gerbv doesn't support it.  Older gerbv will display a warning or an error when it sees the G85.