pcb2gcode/pcb2gcode #165: Copper planes broken

christoph1703, 2018-06-17:
Hi,
I found a bug which can be seen here:
[image]
The copper plane has some weird contours around the other traces. It seems to happen only with traces completely surrounded by the plane. I tried different versions and found that the issue appered first in dd7a96d1d2394e48f5b9f992cf9b58a5df78c6a3.
I can reproduce the bug with this file:
[gbr.zip](https://github.com/pcb2gcode/pcb2gcode/files/2109274/gbr.zip)
The file in the picture is the front layer. I have attached the back layer, because it shows the same issue, but not with all of the traces.

Best regards,
Christoph

eyal0, 2018-06-18:
Thanks for sending this. Looks like a real bug! I'll take a look.

eyal0, 2018-06-19:
I'm at a computer now and I've unzipped the gbr.zip file but I don't see any millproject file in there and I don't know what command-line options you used for pcb2gcode.  :disappointed:

I tried your files using the following `millproject` file and the results look okay so maybe the problem only appears with certain settings.  Please send the `millproject` file that you used and also the command-line that you used.  Thanks!

```
back=feeder-B.Cu.gbr
front=feeder-F.Cu.gbr

mill-feed=360 in/min
mill-speed=200rotations/second
voronoi=true
zwork=-0.04inch
drill-feed=30
drill-side=auto
drill-speed=12000
milldrill=true
onedrill=true
zdrill=-1.6
cut-feed=100
cut-infeed=100
cut-side=auto
cut-speed=10000
cutter-diameter=0.8
milldrill-diameter=0.8
zbridges=-0.6000
zcut=-0.13
fill-outline=true
zsafe=0.08
zchange=1.0

```

christoph1703, 2018-06-19:
Sorry, completely forgot about the command-line options. Here's a zip with the complete set of gerbers and the millproject. I just tested it with dd7a96d1d2394e48f5b9f992cf9b58a5df78c6a3 and without additional options. The millproject file you sent doesn't work with that version (unknown units and missing parameters).
[gbr.zip](https://github.com/pcb2gcode/pcb2gcode/files/2116729/gbr.zip)
It seems to use the voronoi contours for the inner edges of the copper planes.
Let me know if I can help you with the testing.

eyal0, 2018-06-20:
I just tried it with the files that you've provided but it's working fine for me.  I tried with boost 1.63 and also with boost 1.56.  Both worked fine.  Maybe it's due to the version of boost?

The version that you pointed out was a pretty big change in how voronoi rings are found.  But it should be a much more robust version and it solved problems in the previous one so I'm surprised that it's causing problems.

I'm not sure how to proceed because I can't reproduce your error.  What do you suggest?

christoph1703, 2018-06-22:
I'm using boost 1.67, so that might be the issue. Unfortunately, I have so many unusable installations of boost that I couldn't manage to use a different version. I'll set up a fresh system and try different versions of boost. I'll report back when I have more information.

eyal0, 2018-06-24:
I just tested with boost 1.67 and mine is still working just fine.  I can't figure out how you're getting that problem.  As you can imagine, it's difficult for me to debug that without being able to reproduce it here on my end.

christoph1703, 2018-06-24:
Hmm, the problem is gone on my freshly installed system. I have no idea what went wrong, but I think we can close this until (if ever) the bug returns. Sorry to have caused trouble for nothing.

closed 2018-06-24 by christoph1703

eyal0, 2018-06-24:
No biggie.  It happens to everyone.  Thanks!