pcb2gcode/pcb2gcode #167: Segmentation fault on back cupper

hedefalk, 2018-07-22:
I get a segmentation fault on the back cupper of  the gerbers in this thread 

https://www.muffwiggler.com/forum/viewtopic.php?t=153845&highlight=

`Exporting back... [1]    38815 segmentation fault  pcb2gcode`

Is there any logging I can turn out to nail it down? The front and drills work so it's just something particular with the back cupper layer…

hedefalk, 2018-07-22:
This is on HEAD, aed2d09

eyal0, 2018-07-22:
Can you attach the gerbers?  I don't see them.

hedefalk, 2018-07-23:
Yes, sorry, I guess one could only download files if logged in to that forum. Here it is:

[matcher_215.zip](https://github.com/pcb2gcode/pcb2gcode/files/2218870/matcher_215.zip)

eyal0, 2018-07-25:
It worked okay for me.

In the past, the only time that I had crashes was when I was using the option "vectorial=true" (that is the default) and using an old version of boost.  Can you check your version of boost?  Just run `pcb2gcode -V` and it will print it.

If you can, try to install a newer version and use that.  Or turn off vectorial, though the results are not as good.

hedefalk, 2018-07-25:
```
~/d/p/e/Y/gcode-build ❯❯❯ pcb2gcode -V
1.3.2
Git commit: aed2d09
Boost: 106700
```

How do I up Boost version in this build? Sorry, my experience with C++ build is pretty dated. I'm simply building with homebrew using git head:

```
brew install --HEAD pcb2gcode
```

hedefalk, 2018-07-25:
Anyways, you're right that it was only with verctorial=true. Setting false I got through:

```
~/d/p/e/Y/gcode-build ❯❯❯ pcb2gcode                                                                  ⏎
Importing front side... not specified.
Importing back side... DONE.
Importing outline... not specified.
Processing input files... DONE.
Exporting back...
Warning: pcb2gcode hasn't been able to fulfill all clearance requirements and tried a best effort approach instead. You may want to check the g-code output and possibly use a smaller milling width.
DONE. (Height: 98.2956mm Width: 49.7816mm)
Importing drill... DONE.
Exporting drill... DONE. The board should be drilled from the BACK side.
END.
```

eyal0, 2018-07-25:
1.67 is a very new boost version. It should work. I tested with 1.63. I will try 1.67 with your files and see what happens.

Please also send you millproject file. Maybe it is affecting the result?

hedefalk, 2018-07-25:
My millproject:

```
# options are here: https://github.com/pcb2gcode/pcb2gcode/blob/master/options.cpp#L215

metric=true
metricoutput=true
vectorial=true

# https://github.com/winder/Universal-G-Code-Sender/issues/881#issuecomment-361567185
nog81=true #Maybe grbl does support now, it seemed to work?
nog91-1=true

front=../Downloaded_gerbers/matcher-F.Cu.gtl
back=../Downloaded_gerbers/matcher-B.Cu.gbl
drill=../Downloaded_gerbers/matcher.drl

output-dir=../build

#outline=vco555-Edge_Cuts.gbr
zcut=-3
cutter-diameter=1
cut-speed=30000
cut-feed=100
cut-infeed=1

# This seems to set the origin differently, but I don't understand the docs
#zero-start=true

zwork=-0.1
zsafe=3
zchange=5
zdrill=-3 # drill down this far
offset=1.5 # half of tool diam
# drill speed
drill-feed=150
drill-speed=30000

#onedrill=true # use only one bit

milldrill=true
milldrill-diameter=0.6
mill-feed=60
mill-speed=30000 # don't think this is controllable on my T8?
```

I tried to build with boost 1.57 after reading the README, but I still get the same when building:
```
>brew install boost@1.57
>brew uninstall --force pcb2gcode
>export LDFLAGS=-L/usr/local/opt/boost@1.57/lib
>export CPPFLAGS=-I/usr/local/opt/boost@1.57/include
>brew install --HEAD pcb2gcode
>pcb2gcode -V
1.3.2
Git commit: aed2d09
Boost: 106700
```

eyal0, 2018-07-26:
I am able to reproduce the problem with boost 1.67:

Importing front side... DONE.
Importing back side... DONE.
Importing outline... not specified.
Processing input files... DONE.
Exporting back... zsh: segmentation fault  ~/private/pcb2gcode/pcb2gcode

With other versions it works okay.  I will take a look.

In the meanwhile, if you want to compile with a different version of boost you will need to make pcb2gcode after you run `./configure` with the `--with-boost` option.  The readme has some explanations about this.

hedefalk, 2018-07-26:
@eyal0 Thx! I'll try to manually build then.

hedefalk, 2018-07-26:
Ok, sorry, I now did:

```
autoreconf -fvi
./configure --disable-dependency-tracking --disable-silent-rules --with-boost=/usr/local/opt/boost@1.57
make
```

Got a build error:

```
In file included from autoleveller.cpp:20:
In file included from ./autoleveller.hpp:44:
./geometry.hpp:78:12: error: redefinition of 'coordinate_traits<long long>'
    struct coordinate_traits<int64_t> {
           ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/opt/boost@1.57/include/boost/polygon/isotropy.hpp:180:10: note: previous definition is here
  struct coordinate_traits<polygon_long_long_type> {
         ^
1 error generated.
make[2]: *** [autoleveller.o] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

eyal0, 2018-07-26:
Boost is problematic.  Some versions break. Can you try 1.66?  That's the version that worked for me.

eyal0, 2018-07-26:
By the way, I understand why the code was crashing for you and I think that I know how to fix it.  Hopefully it won't take more than a few days...

goddisignz, 2018-07-26:
I only want to confirm the segfault using "vectorial=true" and boost 1.67 on Arch Linux. Heres my gdb output in case somebody can get some information out of that:
```
(gdb) run --back driver_board-B.Cu.gbr 
Starting program: /usr/local/bin/pcb2gcode --back driver_board-B.Cu.gbr
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Importing front side... not specified.
Importing back side... DONE.
Importing outline... not specified.
Processing input files... DONE.
Exporting back... 
Program received signal SIGSEGV, Segmentation fault.
0x00005555555c1c86 in std::vector<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, std::allocator<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian> > >::operator= (this=this@entry=0x5555559df4b0, __x=<error reading variable: Cannot access memory at address 0x8>) at /usr/include/c++/8.1.1/bits/vector.tcc:186
186	    vector<_Tp, _Alloc>::
```

eyal0, 2018-07-26:
Do it again and then run the command `bt` to generate a backtrace please.  Here is the backtrace that I am debugging:

```
#0  std::vector<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, std::allocator<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian> > >::operator= (this=this@entry=0x555555acf1c0, __x=...) at /usr/include/c++/7/bits/vector.tcc:203
#1  0x000055555564ada5 in operator= (this=<optimized out>) at /usr/local/google/home/eyalsoha/private/boost_1_67_0/include/boost/geometry/geometries/ring.hpp:64
#2  operator= (this=<optimized out>) at /usr/local/google/home/eyalsoha/private/boost_1_67_0/include/boost/geometry/geometries/polygon.hpp:76
#3  Surface_vectorial::offset_polygon (this=this@entry=0x5555559c75b0, input=..., voronoi_polygons=..., toolpath=..., contentions=@0x7fffffffbb76: true, offset=59055.118110236217, index=114, 
    steps=1, do_voronoi=false) at surface_vectorial.cpp:332
#4  0x000055555564c8d6 in Surface_vectorial::get_toolpath (this=0x5555559c75b0, mill=..., mirror=<optimized out>) at surface_vectorial.cpp:139
#5  0x0000555555633a49 in Layer::get_toolpaths (this=<optimized out>) at layer.cpp:44
#6  0x000055555563463f in NGC_Exporter::export_layer (this=this@entry=0x5555559e8d60, layer=..., of_name=...) at ngc_exporter.cpp:147
#7  0x0000555555637115 in NGC_Exporter::export_all (this=0x5555559e8d60, options=...) at ngc_exporter.cpp:127
#8  0x000055555556c33d in main (argc=<optimized out>, argv=<optimized out>) at main.cpp:373

```

The problem is `surface_vectorial.cpp:332`.  There is an access of element 0 in an empty list.

goddisignz, 2018-07-26:
There you go:
```
#0  0x00005555555c1c86 in std::vector<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, std::allocator<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian> > >::operator=(std::vector<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, std::allocator<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian> > > const&) (this=this@entry=0x5555559df4b0, __x=<error reading variable: Cannot access memory at address 0x8>) at /usr/include/c++/8.1.1/bits/vector.tcc:186
#1  0x000055555564ef73 in boost::geometry::model::ring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, true, true, std::vector, std::allocator>::operator=(boost::geometry::model::ring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, true, true, std::vector, std::allocator> const&) (this=0x5555559df4b0)
    at /usr/include/boost/geometry/geometries/ring.hpp:64
#2  0x000055555564ef73 in boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, true, true, std::vector, std::vector, std::allocator, std::allocator>::operator=(boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, true, true, std::vector, std::vector, std::allocator, std::allocator> const&) (this=0x5555559df4b0) at /usr/include/boost/geometry/geometries/polygon.hpp:76
#3  0x000055555564ef73 in Surface_vectorial::offset_polygon(boost::geometry::model::multi_polygon<boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, true, true, std::vector, std::vector, std::allocator, std::allocator>, std::vector, std::allocator> const&, boost::geometry::model::multi_polygon<boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, true, true, std::vector, std::vector, std::allocator, std::allocator>, std::vector, std::allocator> const&, boost::geometry::model::multi_linestring<boost::geometry::model::linestring<boost::geometry::model::d2::point_xy<double, boost::geometry::cs::cartesian>, std::vector, std::allocator>, std::vector, std::allocator>&, bool&, double, unsigned long, unsigned int, bool) (this=0x5555559ddc10, input=..., voronoi_polygons=..., toolpath=..., contentions=@0x7fffffffcfd6: false, offset=5118.110236220472, index=0, steps=1, do_voronoi=false)
    at surface_vectorial.cpp:332
#4  0x0000555555650c5b in Surface_vectorial::get_toolpath(std::shared_ptr<RoutingMill>, bool) (this=0x5555559ddc10, mill=std::shared_ptr<RoutingMill> (use count 6, weak count 0) = {...}, mirror=<optimized out>) at /usr/include/c++/8.1.1/bits/shared_ptr_base.h:1002
#5  0x000055555563f95a in Layer::get_toolpaths() (this=<optimized out>) at /usr/include/c++/8.1.1/ext/atomicity.h:96
#6  0x000055555564045e in NGC_Exporter::export_layer(std::shared_ptr<Layer>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (this=0x555555d12670, layer=std::shared_ptr<Layer> (use count 2, weak count 0) = {...}, of_name="back.ngc") at /usr/include/c++/8.1.1/bits/shared_ptr_base.h:1002
#7  0x00005555556426d9 in NGC_Exporter::export_all(boost::program_options::variables_map&) (this=0x555555d12670, options=...) at /usr/include/c++/8.1.1/bits/basic_string.h:917
#8  0x00005555555713ff in main (argc=<optimized out>, argv=<optimized out>) at main.cpp:373
#9  0x00007ffff38a406b in __libc_start_main () at /usr/lib/libc.so.6
#10 0x000055555557232a in _start () at board.cpp:265
```

eyal0, 2018-07-26:
Yes, you are seeing the same error, on like 332 of surface_vectorial.cpp.  I'm going to do a big clean up that part of the code because it's always been a mess.

eyal0, 2018-07-28:
I've got it working for boost 1.67 now.  The code is much nicer.  However, boost 1.67 broke the `optimise` feature!  I've filed a bug against boost: https://svn.boost.org/trac10/ticket/13645#comment:1

eyal0, 2018-07-28:
Because what I've done makes the code better, even though it doesn't solve the simplify issue with boost 1.67, I'm going to submit it.  However, if you use boost 1.67, you ought to set `optimise=false`.  Hopefully they'll fix it soon!

commit 10324b21 (2018-07-29): Merge pull request #170 from eyal0/new_offset_polygon

Clean up the algorithm for offset polygons

This fixes #167
closed 2018-07-29 by eyal0