pcb2gcode/pcb2gcode #168: Mill oval holes

tomas-raul, 2018-07-25:
Hello,
as I understand, that you will update small holes drilling and biger holes milling (in other thread), there is small problem with oval holes. The reader is not parse good this code (Kicad 5 export):

T5
X083240Y-125400G85X084760Y-125400
G05
X083240Y-131500G85X084760Y-131500
G05
X089080Y-129970G85X089080Y-128450
G05
T0
M30

This is code for milling power connector - oval holes. It would be great, when pcb2gcode will do this (with drilling/milling upgrade as you say in other thread). This time, when I need to mill bigger holes, gcode mill all, without tool change, or drill all holes (for me 0,4 - 3.7mm) in center only.

What about this extnesion about G85 for oval holes ? (I was reading your src, but as I'm delphi and AVR developer, I don't understand it so much).

Thanks

tomas-raul, 2018-07-25:
Hello,
I think something else. Look at my drill file. I would like, that holes < 1mm will be drilled with adequate drill tool, and holes >=1mm (or oval holes) will be milled with 1mm mill tool. Not drilling oval with x drill holes near from it. As my file is not loaded in pcb2gcodeGui good - oval holes is not loaded as oval (or 1,2,3 drill holes.)

This time - when i do - millholes - there ALL holes is milled with somethink (4 circle hole is milled, others is drilled (but with only one tool - milling tool). Oval holes is drilled as normal hole. When I do - without mill holes, then ALL holes is drilled - with adequate tool size (4 big holes with 3.7mm) and oval holes is also drilled as circle normal holes. See attached image of input drill and drl file from zip.

[drill_photo](https://user-images.githubusercontent.com/11463339/43220514-89589618-904a-11e8-9ab6-a6c1ce0c82e4.png)

[Servo4.zip](https://github.com/pcb2gcode/pcb2gcode/files/2229100/Servo4.zip)

eyal0, 2018-07-26:
First, you are using an old version of gerbv.  I can see this because your image has no slots.  Look my my phone and see the slots.

[image]

Do you see the three slots at the bottom?  You don't have those because your version of gerbv is old.  

```
> gerbv --version
gerbv version 2.6A
Copyright (C) 2001 -- 2008 by Stefan Petersen
and the respective original authors listed in the source files.

```

When you run pcb2gcode, did you see warnings output about the unrecognized G85?

First you need to fix this, then we can work on the rest.

tomas-raul, 2018-07-26:
Hello,
thanks - I'm using windows version from https://github.com/pcb2gcode/pcb2gcodeGUI/releases - and this time I see, that the rlease is very old. Will try new release !

Log of my pcb2gcode (from pcb2gcodeGui) :

Importing front side... DONE.
Importing back side... DONE.
Importing outline... DONE.
Processing input files... DONE.
Exporting back... DONE. (Height: 51.2516mm Width: 51.2516mm)
Exporting front... DONE. (Height: 51.2516mm Width: 51.2516mm)
Exporting outline... DONE. (Height: 51.2516mm Width: 51.2516mm) The board should be cut from the FRONT side. 
Importing drill... DONE.
Exporting drill... Warning: 70 holes were bigger than the milling tool.
DONE. The board should be drilled from the FRONT side.
END.

tomas-raul, 2018-07-26:
Ha, don't know how, but I have slot in original_dril.svg
But, pcb2gcodeGUI say me : mirror-absolute is deprecated, it must be true.

tomas-raul, 2018-07-26:
But the result gcode is good form drill, for mill... There are unwanted milling out of pcb. (milling tool is 1.0mm (this time I'm interested only in drill data, not other pcb.) When I change tool to 0.1mm, then small holes is milled ok, but also there is unwanted moves out of pcb (with milling on the way).

How about the other thread - drilling to <max drill size>, then change to milling tool and milling bigger holes and slots ?

[image]

eyal0, 2018-07-26:
I have never used the GUI and also I have never used pcb2gcode on Windows. I do not know how to compile it!

The slots in your output look starange.  I will investigate.

In your png files, do you see slots like I have?

eyal0, 2018-07-26:
I still need to add support for milling big holes and drilling small holes. I hope that it will not.take too long

tomas-raul, 2018-07-26:
Yes, in my original_drill.svg - there is all good :)
Take a look on attached zip (Other files I didn't chesk, only drill file.)
[cnc.zip](https://github.com/pcb2gcode/pcb2gcode/files/2233048/cnc.zip)

eyal0, 2018-07-26:
Yes, mirror absolute must be true. False is broken.

tomas-raul, 2018-07-26:
I checked other files, and back file has some errors near 2 biggest holes. 

This was generated by :

pcb2gcode.exe --back=D:Servo4-B.Cu.gbr --drill=D:Servo4.drl --front=D:Servo4-F.Cu.gbr --outline=D:Servo4-Edge.Cuts.gbr --output-dir=D:cnc --metric=true --metricoutput=true --mirror-absolute=true --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=true --zchange=10.0000 --zchange-absolute=false --zero-start=true --zsafe=5.0000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=0.1000 --voronoi=false --zwork=-0.0500 --drill-feed=1000 --drill-side=auto --drill-speed=10000 --milldrill=false --nog81=false --nog91-1=false --onedrill=false --zdrill=-2.5000 --bridges=0.5000 --bridgesnum=2 --cut-feed=600 --cut-infeed=10.0000 --cut-side=auto --cut-speed=10000 --cutter-diameter=3.0000 --fill-outline=true --zbridges=-0.6000 --zcut=-2.5000

I thing, that this cfg is near default, but this two places seems like voronoi regions :(

eyal0, 2018-07-26:
Can you also run pcb2gcode --version ?

Your zcut is way too big! You are going to break your drill. If you use a
smaller zcut, maybe 0.2, you may see a better path on the output.

tomas-raul, 2018-07-26:
[Export.zip](https://github.com/pcb2gcode/pcb2gcode/files/2233135/Export.zip)

eyal0, 2018-07-30:
I see that you have an error with the two big circles at the top of your output in Export.zip.  But when I run pcb2gcode, I don't have that error.  Are you sure that you made those files with the latest pcb2gcode?  There was an error once in pcb2gcode about that and it was fixed recently.

Please run the latest pcb2gcode and compare:

`gerbv Servo4-B.Cu.gbr`

with:

`outp0_original_back.svg`

Thanks!

eyal0, 2018-07-30:
@tomas-raul I have some code for you to try:

https://github.com/eyal0/pcb2gcode/archive/min_milldrill.zip

If you build that version of pcb2gcode, there is a new option called `min-milldriill-hole-diameter`.  You can write like this:

`min-milldrill-hole-diameter=1.5mm`

If you use that option, all holes that are less than the size you choose will be drilled.  Holes that are the same as the size you choose or bigger will be milled.  For both holes above and below, G85 (slots) are supported.  drilled slots are done with many holes in a line.  milldrilled slots are done with oval milling.

I hope that it works well for you!  Please let me know if you are happy with the results.

tomas-raul, 2018-07-31:
Hello, comparsion of back side seemslike good.

[image]

tomas-raul, 2018-07-31:
I'll try your new source code today later - but,, That's great... (I broke some tools when I was trying to drill oval and bigger holes.) I'm not very familiar with msys and etc.

eyal0, 2018-07-31:
Oh no, that is no good: 

[image]

Look at the red arrow.  On one side, you have a gap. On the other side, no gap!

On my computer it is working fine.  Maybe there is a bug with your pcb2gcode.  Are you using an old pcb2gcode?  For me, it works correctly.

tomas-raul, 2018-07-31:
Hi,
in this moment i compiled your sources and gerbv from git://git.geda-project.org/gerbv.git.

Will try it later, after compilation.

eyal0, 2018-08-03:
By the way, I discovered that boost 1.67 is not good!  It has some problems.  You should use boost 1.66.

tomas-raul, 2018-08-07:
Yes, it it zero point :)

[tmp.zip](https://github.com/pcb2gcode/pcb2gcode/files/2266184/tmp.zip)

tomas-raul, 2018-08-07:
Remove zero point sign is repair this :) Trying to compile sources from here one more time

tomas-raul, 2018-08-07:
So, I have compiled, copy pcb2gcode.exe from .libs directory and then run it (other libraries I dont copy, as it is copied from last time).

d:\projekty\IQtrains.cz\CNC\pcb2gcode>pcb2gcode.exe --back=D:/projekty/IQtrains.cz/CNC/3020-software-spindle-control-master/hw/tmp/3020-cnc-upgrade-B.Cu.gbr --drill=D:/projekty/IQtrains.cz/CNC/3020-software-spindle-control-master/hw/tmp/3020-cnc-upgrade.drl --front=D:/projekty/IQtrains.cz/CNC/3020-software-spindle-control-master/hw/tmp/3020-cnc-upgrade-F.Cu.gbr --outline=D:/projekty/IQtrains.cz/CNC/3020-software-spindle-control-master/hw/tmp/3020-cnc-upgrade-Edge.Cuts.gbr --output-dir=D:/projekty/IQtrains.cz/CNC/3020-software-spindle-control-master/hw/tmp/cnc --metric=true --metricoutput=true --mirror-absolute=true --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=true --zchange=10.0000 --zchange-absolute=false --zero-start=true --zsafe=5.0000 --extra-passes=0 --mill-feed=200 --mill-speed=10000 --offset=0.1000 --voronoi=false --zwork=-0.0100 --drill-feed=200 --drill-side=back --drill-speed=10000 --milldrill=false --nog81=false --nog91-1=false --onedrill=false --zdrill=-2.5000 --bridges=0.5000 --bridgesnum=2 --cut-feed=200 --cut-infeed=10.0000 --cut-side=back --cut-speed=10000 --cutter-diameter=3.0000 --fill-outline=true --zbridges=-0.6000 --zcut=-2.5000 --min-milldriill-hole-diameter=1.0mm
Error: You've supplied an invalid parameter.
Note that spindle speeds are integers!
Details: unrecognised option '--min-milldriill-hole-diameter=1.0mm'

tomas-raul, 2018-08-07:
Ou, I copied this from your post - will try it.
Sorry I dont know which boost is installed. I used your manual for windows compilation - msys. (Only gerbv I downloaded from git)

Kicad has 3 point of zero (x0,y0). I dont know why 3, but when I use 3, one broke outline (see attached zip in previous post). When I removed one of them, all is okay. (When I dont use them, then the gcode is with high offset to somewhere :) )

tomas-raul, 2018-08-07:
Ou, I tested milldrill and it is working very good for first look. Will look deeply in some hours. Very good is that you use 2 files - drill and milldrill :) Many thanks

eyal0, 2018-08-07:
Yes, I used different files so that you can choose if you want to do both or just one. If you one want to mill drill big holes, you can just make both files and ignore the small one!

Here you can read ideas how to use it. Good luck!

https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Drilling

tomas-raul, 2018-08-07:
You are the best one ! 
Many thanks. Todays PCB was did by hand (generate one type, and skip some holes, then generate second type and skipp others), so in a few next days I'll checked this, but it looklikes very good !

tomas-raul, 2018-08-07:
By the way, are you interested in my binaries ? I can send it to you for others !

eyal0, 2018-08-07:
Yes! I want to make a release! But first two things:
1. I need to add to the version number so it will be 2.0 and not 1.3.2. That is easy for me to do.
2. Boost 1.67 is broken! It makes big mistakes. I told them but no one listened yet. We need a release with boost 1.66. That if the newest one that works. I will learn how to do this and we can see if you are able.
3. I might try to update pcb2gcodeGUI, too.

Then, we make make some good binaries.

tomas-raul, 2018-08-07:
Oukej :) 
1,2) As you wish, when you send me adequate sources and give me a way for boost 1.66, then I could create new release for win32/64 in msys.
3) You are the best one ! (As there is lot of parameters, gui is better for beginners - I saw, that there is also offset and offset for both sides pcb (which was not in GUI) (prevously I used mirroring which is deprecated now, so offset for this is good for me).

I wrote some info about your project on facebook, so the new release with last gerbv will be very good for beginners.

tomas-raul, 2018-08-14:
Hello, how about building ? Should I now create a build release ?

tomas-raul, 2018-08-14:
Ok, when you will be prepared, please write me here :)

eyal0, 2019-02-23:
I have made a new release, version 2.0.  It has many new features now, all described in the wiki.  If you can compile a version for Windows, I will add it to the releases page.

eyal0, 2021-01-07:
This is fixed by b88a2fbd and 011336db and 6d15f726

closed 2021-01-07 by eyal0