pcb2gcode/pcb2gcode #178: Missing ground plane and thermal reliefs

goddisignz, 2018-08-01:
pcb2gcode version running arch linux:
```
1.3.2
Git commit: v1.3.2-201-g10324b2
Boost: 106700
```

I tried to produce the gcode for the back side of a pcb (driver_board-B.Cu.gbr).
In the result (back.ngc) you can see, that the thermal reliefs (or the ground plane at all) are not traced. In pcb2gcodeGUIs preview, at least some of the holes for the reliefs are drawn and gerbv shows the layout as it should be. 

[test.zip](https://github.com/pcb2gcode/pcb2gcode/files/2250903/test.zip)

Seems to be a problem in the newest version. In previous versions (with Boost 1.65 under Ubuntu) it worked.

eyal0, 2018-08-01:
Can you try setting "optimise" to false?  Boost 1.67 has a broken optimize function.  Sorry about that.  I've filed a bug for them to fix it.

goddisignz, 2018-08-02:
Hi, my bad! The ngc-file was created without the "optimise" option. I probably only forgot to save the millproject afterwards. With "optimise=true" the result is a bit worse regarding other missing traces"

eyal0, 2018-08-02:
I tried it with boost 1.67 and I saw the same results as you.  :cry: So it's not just optimise that's broken!  It's really sad that boost had such a regression.  I don't believe that pcb2gcode is abusing boost and using undefined behavior.  I think that boost is simply broken from 1.67 oneward.

https://svn.boost.org/trac10/ticket/13645

1.66 worked great for me, though.  I see all the thermals and the outline of the board is there, too.  Can you use that one?

goddisignz, 2018-08-02:
I hoped this might be fixed in boost 1.68 but then looked at your ticket to see it currently isn't.

I now used linuxbrew to install boost 1.60 as @hedefalk suggested in #167. Downgrading boost is no real option because all distro packages are always linked against the newest version. But installing it locally and then linking to the brew version is more than acceptable for me.
Thanks a lot!

PR #181 (2018-08-04): Require Boost < 1.67
This fixes #178
closed 2018-08-02 by goddisignz