pcb2gcode/pcb2gcode #179: Question - rotating gerbers…?

hedefalk, 2018-08-01:
Does anyone know any tool to rotate gerbers+excellon? Problem I have is I have a design as gerbers+excellon but no source kicad/eagle or anything so I need to rotate the gerbers or the gcode. This because very limited CNC area. I tried hacking with G86 in the gcode but couldn't get it to do anything.

Again, sorry posting questions as issues here, don't know any better forum right now. Let me know if it's annoying.

eyal0, 2018-08-02:
I think gerbv can do it.

hedefalk, 2018-08-02:
@eyal0 Yes it could, thx!

Yet, I'm still having issues :)

So I loaded the gerbers+excellon in gerbv, rotated them 90° and saved them out. I then verified in gerbview (part of KiCad) that they are rotated:

[image]

Also, diffing the coordinates are all new so really rotated all points, not by any meta instruction. However, after processing them with pcb2gcode, the resulting gcode is still _not_ rotated but just as the original gerbers, at least not when looking at them from chilipeppr:

[image]

Isn't that weird? I'm pretty sure I've made sure there's no silly mistake:

- [x] I really do point out the rotated gerbers:
```
#front=../Downloaded_gerbers/matcher-F.Cu.gtl
#back=../Downloaded_gerbers/matcher-B.Cu.gbl
#drill=../Downloaded_gerbers/matcher.drl
front=../Downloaded_gerbers/matcher-F.Cu-90.gtl
back=../Downloaded_gerbers/matcher-B.Cu-90.gbl
drill=../Downloaded_gerbers/matcher-90.drl

```
- [x] I've cleared my output and checked timestamps of the newly generated gcode on importing to chillipeppr. It really _is_ newly processed gcode from the rotated gerbers. Still it looks like before…

hedefalk, 2018-08-02:
Actually, the version of gerv I can use on OSX (2.6.0) has a bug that is doesn't export rotation of the drill file:

https://sourceforge.net/p/gerbv/bugs/171/

eyal0, 2018-08-09:
Is there any further action needed from me on this issue?  If so, please reopen.  Good luck!

hedefalk, 2018-08-09:
@eyal0 Sorry, should have closed this. 

No, no actions from you unless you wanna add a rotate feature :)

I'm having no luck actually, spent half a day trying to build a newer version of gerbv from source on OSX which turned out to be really messy. Can't seem to find anything else that's supposed to be able to rotate excellon.

eyal0, 2018-08-10:
I wrote an idea yesterday but maybe it didn't get saved...

Crazy idea how you can rotate gerbers: Rotating my 90 degrees is just swapping the X and Y values and converting one of the Y into negative, right?  So, what if you do a search and replace on your gerber file and change:

1. All "Y-" to "NEW"
2. All "Y" to "NEW-"
3. All "X" to "Y".
4. All "NEW" to "X".

The result will be a gerber that is rotated 90 degrees counterclockwise.  I have to use the "NEW" there in the middle because when I convert the X to Y and the Y to X, they are switching places so one of them has to sit in a temporary name during the process.

Only run on the part of the file that has gerber.  Don't run it on the comments and stuff at the top of the file.

This idea is just crazy enough to work!  Do it for all the geber files and the drill file, too.

I want to hear if you try this and how it works.  Good luck.

hedefalk, 2018-08-10:
@eyal0 Great minds think alike :)

I actually did it with a simple regex replacement:

query: X(.*)Y(.*)
replace: X$2Y$1

BAM:
[image]

Let me make a sed-script if it might be useful for someone else…

hedefalk, 2018-08-10:
Me and sed aren't friends. I get it working with test input:

```
❯ echo "X5.200Y-2.5\nX5.200Y-6.4" | sed -E 's/X(.+)Y(.+)/X\2Y\1/'

X-2.5Y5.200
X-6.4Y5.200
```

But trying on a file it just drops the X-part:

```
cat matcher.drl | sed -E 's/X(.+)Y(.+)/X\2Y\1/'
…
Y5.200
Y5.200
```

hedefalk, 2018-08-10:
Well, it's something with the original drl file because I can get a test file working:

```
~/d/p/e/Y/Downloaded_gerbers ❯❯❯ echo "X5.200Y-2.5\nX5.200Y-6.4" | sed -E 's/X(.+)Y(.+)/X\2Y\1/'

X-2.5Y5.200
X-6.4Y5.200
~/d/p/e/Y/Downloaded_gerbers ❯❯❯ echo "X5.200Y-2.5\nX5.200Y-6.4" > test.drl
~/d/p/e/Y/Downloaded_gerbers ❯❯❯ cat test.drl | sed -E 's/X(.+)Y(.+)/X\2Y\1/'
X-2.5Y5.200
X-6.4Y5.200
```

hedefalk, 2018-08-10:
Hm…

```
~/d/p/e/Y/Downloaded_gerbers ❯❯❯ cat matcher-torotate.drl
X5.200Y-2.5
X5.200Y-6.4%                                                                                           ~/d/p/e/Y/Downloaded_gerbers ❯❯❯ cat matcher-torotate.drl | sed -E 's/^X(.+)Y(.+)$/X\2Y\1/'
Y5.200
X-6.4Y5.200
```

hedefalk, 2018-08-10:
Ok, this works:

`cat matcher-torotate.drl | sed -E 's/^X(.\+)Y(.\+)$/X\2Y\1/'`

I guess the + needed escaping, don't understand how it worked at all without it then though…

hedefalk, 2018-08-10:
Ok, came through by doing more explicit matching for just the numbers:

cat matcher.drl | sed -E 's/X[-]?([0-9]*\.[0-9]*)Y([-]?[0-9]*\.[0-9]*)/X\2Y\1/g'

However, I realize as you already wrote I haven't inverted any axis therefore also mirroring… So yeah, need to fix that.

eyal0, 2018-08-11:
Yes. For one of the axis, you need to convert negative to positive and positive to negative.

eyal0, 2018-10-02:
I don't think that there is anything for me to do here.  If you need to rotate your gerber, you'll have to use gerbv or some other tool.

closed 2018-10-02 by eyal0

hedefalk, 2018-10-02:
👍