pcb2gcode/pcb2gcode #180: Mirror-absolute is deprecated.

kosme, 2018-08-02:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):

1.3.2
Git commit: v1.3.2-237-gc002a45-dirty
Boost: 105800

What did you try (include command-line arguments):

pcb2gcode --back="/home/kosme/Documents/Hormiga Azul/kicad/arch/arquitectura/gerbers/arquitectura-B.Cu.gbr" --drill="/home/kosme/Documents/Hormiga Azul/kicad/arch/arquitectura/gerbers/arquitectura.drl" --front="/home/kosme/Documents/Hormiga Azul/kicad/arch/arquitectura/gerbers/arquitectura-F.Cu.gbr" --outline="/home/kosme/Documents/Hormiga Azul/kicad/arch/arquitectura/gerbers/arquitectura-Edge.Cuts.gbr" --output-dir="/home/kosme/Documents/Hormiga Azul/kicad/arch/arquitectura/gerbers/gcode"  --metric=true --metricoutput=true --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=true --zchange=10.0000 --zchange-absolute=false --zero-start=true --zsafe=5.0000 --mill-feed=600 --mill-speed=10000 --voronoi=true --zwork=-0.0100 --drill-feed=1000 --drill-side=back --drill-speed=10000 --milldrill=false --nog81=false --nog91-1=false --onedrill=false --zdrill=-2.5000 --bridges=0.5000 --bridgesnum=2 --cut-feed=600 --cut-infeed=10.0000 --cut-side=back --cut-speed=10000 --cutter-diameter=1.0000 --fill-outline=true --zbridges=-0.6000 --zcut=-2.5000 --mirror-absolute=false

What happenned:

mirror-absolute is deprecated, it must be true.

What did you expect to happen:

Why was the mirror-absolute option deprecated? It was a vital part of the building process I already had implemented for my 2-sided PCB's.

eyal0, 2018-08-03:
It didn't work consistently and I couldn't find anyone that could explain to me why it's useful! In some cases it was just plain wrong.

In the wiki, I wrote an explanation of how to make two sided boards using the mirror-axis option. Can you read that and see if you can make PCBs using it? Or explain your process to me?

Sorry about that. I hope that you'll find the new process working or maybe even better!

eyal0, 2018-08-09:
@kosme It's been nearly a week.  Any progress on this?  You can reopen the issue if needed.

kosme, 2018-08-10:
My process was quite simple.
1. Drill through the back side
2. Mill the back side.
3. Cut the outline of the PCB.
4. Run a python script that analyses the front face, gets the size, and generates a file for milling a box in which it fits exactly.
5. Place a scrap piece of wood or MDF and cut the previously generated file.
6. Using the same alignment that was used on the precious step, insert the milled PCB on the previously milled box and mill the front face.

eyal0, 2018-08-10:
Okay, that's kind of an interesting way to do it!

Here's a way that you could basically maintain your process: Have your script that analyzes the front face output the middle x coordinate.  Then set `mirror-axis` to that value and run pcb2gcode again.

If you are willing to modify your process, we could maybe streamline it and make it better?  Something like this:

1. Cut a rectangular hole in your scrap wood that is the size of a blank PCB.
2. Run pcb2gcode with --mirror-axis equal to the center x value of the rectangular hole.
3. Put your PCB in there and mill the back.
4. Flip it over and mill the front.

So long as mirror-axis is along the center x-coordinate of the rectangular jig that you mill from the wood, it'll work.

This has the advantage that you can resuse the jig for all PCBs, even if you made bigger ones, so long as you keep the same size raw material PCB.

What do you think?  Could that process work for you?

kosme, 2018-08-10:
I think the first option would work for my process, I'll give it a try. The second one would require some time to learn the new work routine and would probably mess up the first PCB's

eyal0, 2018-08-11:
That's just a matter of mirror-axis. The default is to mirror across zero so you're getting negative values. Try mirror-axis=100cm

eyal0, 2019-02-23:
No progress here in a long time, I'll close this.

closed 2019-02-23 by eyal0