pcb2gcode/pcb2gcode #187: Please add auto Dynamic Tool Length Offset adjusting option with probing for drilling

SmallestLeslie, 2018-08-06:
pcb2gcode generates drill files as the following:
```
( pcb2gcode 1.3.2 )
( Software-independent Gcode )

( This file uses 2 drill bit sizes. )
( Bit sizes: [0.8 mm] [1 mm] )

G94       (Millimeters per minute feed rate.)
G21       (Units == Millimeters.)
G91.1     (Incremental arc distance mode.)
G90       (Absolute coordinates.)
G00 S10000     (RPM spindle speed.)

G00 Z30.00000 (Retract)
T1
M5      (Spindle stop.)
G04 P1.00000
(MSG, Change tool bit to drill size 0.8 mm)
M6      (Tool change.)
M0      (Temporary machine stop.)
M3      (Spindle on clockwise.)
G0 Z2.00000
G04 P1.00000

G81 R2.00000 Z-2.50000 F500.00000 X-81.28980 Y51.06380
X-76.46380 Y56.65180

G00 Z30.00000 (Retract)
T2
M5      (Spindle stop.)
G04 P1.00000
(MSG, Change tool bit to drill size 1 mm)
M6      (Tool change.)
M0      (Temporary machine stop.)
M3      (Spindle on clockwise.)
G0 Z2.00000
G04 P1.00000

G81 R2.00000 Z-2.50000 F500.00000 X-72.14580 Y40.39580
X-62.23980 Y47.50780

G00 Z30.000 ( All done -- retract )

M5      (Spindle off.)
G04 P1.000000
M9      (Coolant off.)
M2      (Program end.)
```

With auto Dynamic Tool Length Offset adjusting option and probing it could generate something like this:
```
( pcb2gcode 1.3.2 )
( Software-independent Gcode )

( This file uses 2 drill bit sizes. )
( Bit sizes: [0.8 mm] [1 mm] )

G94       (Millimeters per minute feed rate.)
G21       (Units == Millimeters.)
G91.1     (Incremental arc distance mode.)
G90       (Absolute coordinates.)
G00 S10000     (RPM spindle speed.)

O1 SUB
G49
G0 Z35
G0 X0 Y0 ; need to be the actual tool sensor position
M6 ; do the change
(MSG, Attach probe)
M0      (Temporary machine stop.)
G38.2 F200 Z -5 ; the speed is critical here, it must be slow enough for the machine to stop inside the probe travel
G0 Z[#5063+2]
G38.2 F80 Z -5
G43.1 Z[#5063]
G0 Z2
(MSG, Detach probe)
M0      (Temporary machine stop.)
O1 endsub

T1
M5      (Spindle stop.)
G04 P1.00000
(MSG, Change tool bit to drill size 0.8 mm)
O1 call
M3      (Spindle on clockwise.)
G04 P1.00000

G81 R2.00000 Z-2.50000 F500.00000 X-81.28980 Y51.06380
X-76.46380 Y56.65180

T2
M5      (Spindle stop.)
G04 P1.00000
(MSG, Change tool bit to drill size 1 mm)
O1 call
M3      (Spindle on clockwise.)
G04 P1.00000

G81 R2.00000 Z-2.50000 F500.00000 X-72.14580 Y40.39580
X-62.23980 Y47.50780

G00 Z30.000 ( All done -- retract )

M5      (Spindle off.)
G04 P1.000000
M9      (Coolant off.)
M2      (Program end.)
```

Variables for the o1 subroutine could be:
```
O1 SUB
G49
G0 Z[#1]
G0 X[#2] Y[#3] ; need to be the actual tool sensor position
M6 ; do the change
(MSG, Attach probe)
M0      (Temporary machine stop.)
G38.2 F200 Z -5 ;the speed is critical here, it must be slow enough for the machine to stop inside the probe travel
G0 Z[#5063+2]
G38.2 F80 Z -5
G43.1 Z[#5063]
G0 Z[#4]
(MSG, Detach probe)
M0      (Temporary machine stop.)
O1 endsub
```
Where:
```
#1: Z for tool change
#2: X for tool change and probing (X=0 was fine for me)
#3: Y for tool change and probing (Y=0 was fine for me)
#4: Z height for movement
```
To be able to the probing finish successfully, the PCB cannot be located lower than -5 in Z.
I assumed that Z for tool change located higher than Z for movement therefore it is safe for moving.

eyal0, 2018-08-08:
This is interesting.

As I understand it, the firmware that runs on the CNC does **not** understand M6.  M6 is read by the program that is sending gcode commands to the printer and then interpreted.  The same is true for subroutines and stuff like "#5063",

I am using bCNC, which is a tool to send gcode to the machine.  It does not support subroutines like LinuxCNC.  In bCNC, I have to put the parameters for the M6 command.

[image]

See in the photo, I can choose the start location, speed, max depth, etc.  When bCNC sees M6, it does a fixed routine.  It doesn't use variables like #5063, it has different names.

What does your software do when M6 is received?  I see that, for LinuxCNC, it's possible to have a routine that gets run whenever there is an M6.

What is the correct value for "#4"?  I don't see why it is necessary.

SmallestLeslie, 2018-08-08:
> As I understand it, the firmware that runs on the CNC does not understand M6.

In my setup there is no firmware.
The gcode interpreter software linuxcnc runs on a PC and calculates all the required driving signals for the mechanic (accelerations, decelerations, etc.) and it outputs only step/dir signals on parallel port (really high tech stuff) for the stepper motor drivers. The motor drivers themselves are not smart, they are only a power stage for the PC.
With manual tool change linuxcnc just pop up a message to change the tool.
By default there is a tooltable in linuxcnc with the needed tool length offsets but as with manual tool change most likely the tool cannot be put into the same place into the chuck.
So this is why the dynamic tool length offset detection is needed.

One way to do this dynamic tool length offset detection is when we mill a PCB that we attach one end of the probing electrical circuit to the copper foil of the PCB and other end to the electrically conductive tool.
[http://2.bp.blogspot.com/-4--cuQBZz28/VmIRcIY2i1I/AAAAAAAASAw/tO1uE2aHzs0/s1600/IMG_20151204_144909.jpg](url)
Then with the probing command the tool approaches the PCB then makes contact with it. This is sensed by the software and gets stored as tool length offset.
"#4" is needed only when the probing surface in not in the same plane as the surface to be milled.
[https://www.picclickimg.com/d/l400/pict/253596523724_/CNC-mill-router-tool-height-Z-setter.jpg](url)
I wanted to add this to make the code as universal as possible but there might better way to do this so this can be omitted I think.

In linuxcnc "#1"-"#15" are volatile variables that can be used for subroutine calls like:
`O1 call [2] [4] [6]`
then the O1 subroutine will have the values 2, 4 and 6 in the "#1", "#2" and "#3" variables.
The variable "#5063" carries the result of the probing process.

chaotix-, 2018-10-15:
Are you sure pcb2gcode is the best place to implement this in? You can implement a tool change routine in LinuxCNC. You just need a touch plate/switch at a fixed location. There is at least one implementation for Axis and it's already part of Gmoccapy by default. (I don't know about the other GUIs as I have never tried them). I'm using an adapted version of https://github.com/koppi/mk/blob/master/linuxcnc/nc_files/tool-change.ngc and it works really well. This way the TLO is set for all tool changes in all G-Code programs and it's not specific to pcb2gcode.

GRBL has the same possibility to measure the TLO on every tool change. So it also does not need a special implementation for pcb2gcode. (And as it does not support G-Code variables it would not be possible to implement it there as part of the pcb2gcode output anyway.)

SmallestLeslie, 2019-02-19:
Actually I did not know there is other gui available for linuxcnc other than axis. I will check it as soon as I have some time for it.
Apart from this I do believe that pcb2gcode would be a good place to implement this feature. As cpb2gcode already has options for control software on the autolevel tab this way it could not cause problems to other softwares.