pcb2gcode/pcb2gcode #188: Suggestion for specifying mirroring axis for double sided PCB

SmallestLeslie, 2018-08-06:
As the surface of the PCB can be considered as a plane and on a plane a line (or in this case an axis) can be defined by two points.
If the user pays attention to place two unique sized holes to the PCB, then export the drill file, it will contain only these two hole with the given unique size.

Hole size specification in KiCad:
[image]

PCB with two unique sized holes:
[image]

Drill file:
M48
;DRILL file {KiCad 4.0.7} date 08/06/18 19:45:21
;FORMAT={3:3/ absolute / metric / suppress leading zeros}
FMAT,2
METRIC,TZ
T1C0.800
T2C1.000
T3C1.100
T4C1.200
T5C1.300
T6C1.320
T7C3.048
**T8C3.737**

**T8
X71120Y-87376
X149352Y-143764**

PCB with defined mirroring axis:
[image]

With this method the mirroring axis could be defined with just specifying the unique hole size with one command line option like:
`--mirroring-holes=3.737` for example

eyal0, 2018-08-09:
Mirroring on the diagonal of a rectangular board would make a weird angle when flipped. It might not fit on the CNC anymore.

In the code, everything assumes that mirroring is along a vertical line. Changing that would be very hard and I'm not sure how useful it would be.

I understand that you want to be able to specify the mirror axis inside kicad but I'm not sure if a fake drill holes is the right way. Maybe the design told have a way to put down a marker of some sort, like on a different layer?

SmallestLeslie, 2018-08-09:
> I understand that you want to be able to specify the mirror axis inside kicad but I'm not sure if a fake drill holes is the right way

We talked about a simple way to specify the mirroring axis and I ended up with the above mentioned. It is not perfect but it is simple.
Those drill holes are not necessary fake ones as neither in the above example are. It's a little bit cheating but neither kicad, nor pcb2gcode know what sized drill will those holes drilled with at the end.

> In the code, everything assumes that mirroring is along a vertical line. Changing that would be very hard and I'm not sure how useful it would be.

I might forgot to mention that I am a little afraid of splitting the process and putting the responsibility of the mirroring accuracy on the user.

> Maybe the design told have a way to put down a marker of some sort, like on a different layer?

I'm not sure how you mean this. Is that marker for manual flipping or for pcb2gcode?

eyal0, 2018-08-09:
> I'm not sure how you mean this. Is that marker for manual flipping or for pcb2gcode?

Yes.  In KiCad, for example, it's possible to draw on any one of multiple layers: front, back, edge cuts, etc.  There is also, for example, a fab layer.  Maybe users could draw a line on the fab layer and export that together with the rest and *that* would be the mirror axis?

I want to get at the heart of the problem here that we're trying to solve.  Is it very difficult for users to specify a mirror-axis?  Does letting a user put a drill hole on the PCB make it easier?

I'm inclined against allow any flipped other than along a line x=CONSTANT.  Because it would really complicate the code, like, a lot, and I just can't imagine the scenario when flipping along anything other than X would be useful.  Blank PCBs are rectangular so flipping either horizontally or vertically makes the most sense.  Anything else requires a larger CNC to do both sides.

SmallestLeslie, 2018-08-09:
> Maybe users could draw a line on the fab layer and export that together with the rest and that would be the mirror axis?

This is absolutely doable I guess. My biggest concern about double sided PCB milling is alignment error.
I don't say that alignment holes are the only option but in my opinion they are the most foolproof.

> I'm inclined against allow any flipped other than along a line x=CONSTANT.

Ok, I do understand your point.
A line drawn on a layer could be usable as well. I just suggest to find a solution to mark the mirroring axis on the PCB with the cnc mill to minimize the possible alignment error.

eyal0, 2018-08-09:
I agree that the alignment holes should be done by the CNC! But that is easy. Just put holes on either side of whatever mirror axis you choose. If you select mirror axis 25 then you can put holes at x, 25±y for any values of x and y. Just drill right through your PCB and wasteboard below. And when you flip, align those holes again.

It's a bit of a manual process to jog the CNC milling tool into place and do it but not too much. The alternative is for the user to put marks on the board for flipping the board and the user would have to make sure that they are far enough away from the board to not infere with milling. And then the user needs to place alignment pegs in place... I would like an elegant solution.

Maybe what you write is best but with adjustment. I don't like limiting the user to just 2 alignment holes. 4 is more sturdy. Maybe we do like you say and the user adds holes but allow for more than just two. pcb2gcode can compute the mirror axis from the holes provided and also check that they are symmetrical across the mirror axis. Then the mirror axis is set automatically and I display on the screen the automatically computed mirror axis along with the locations of the alignment holes. And we put the alignment holes into a new drill file that gets run before the milling (because usually milling must be done before drilling).

That could work okay. I'll think some more about it.

eyal0, 2018-08-09:
Doesn't work with tiling, though. Tiling complicates it. I need to think about it some more.

SmallestLeslie, 2019-02-19:
Can I ask you what you got about this question?
Maybe there is something I can add. I use clamps to hold the PCB on the machine therefore I think two alignment holes would be enough for me.

AlexeyGusev, 2019-08-15:
how about using a «layer alignment target» tool in kicad? It places a target sign on edge.cuts layer and if you put two with the same X coordinates, you have defined a line that the pcb can be flipped around. Maybe kicad's generated gerber code from edge.cuts layer that contains such alignment target can be used do define drills for board flipping?

eyal0, 2019-08-31:
Okay, that's an interesting idea.

I tried to put a layer alignment tool on the edge cuts and plot it and it does show up, looking like a target sign, on the edge cuts.  The problem is that it doesn't look special at all, it just looks like a regular edge cut!  So it'll be hard for pcb2gcode to know what is a real edge cut versus what is a target shape.  I'll have to keep thinking about it...