pcb2gcode/pcb2gcode #192: Thermal Spokes are skipped

a7hybnj2, 2018-08-08:
pcb2gcode version: 1.3.2

What did you try: `pcb2gcode --front Gerber\WORef\wirelessISP-B.Cu.gbr --metric --metricoutput --zsafe 5 --zchange 5 --zwork 0 --offset 0 --mill-feed 1500 --mill-speed 1000 --zero-start`

What happened: It almost worked perfectly! It did not gcode the thermal relief spokes in the copper pour.

What did you expect to happen: I expected complete output of all areas that needed copper removed.

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

[Link to pictures](https://github.com/a7hybnj2/KicadStuffs/tree/master/pcb2gcode_experience)
[Link to Gerber](https://github.com/a7hybnj2/KicadStuffs/blob/master/Gerber/wirelessISP-B.Cu.gbr)
[The items crudely circled in red is what is missing from the generated gcode](https://github.com/a7hybnj2/KicadStuffs/blob/master/pcb2gcode_experience/outp0_original_front%20HIGHLIGHTED.png)

a7hybnj2, 2018-08-09:
I have since found #178 but...

I have:
```
brew unlink pcb2gcode
brew install boost@1.60
cd ......Library/Caches/Homebrew/pcb2gcode-pcb2gcode--git
autoreconf --fvi
/configure --disable-silent-rules --with-boost=/usr/local/opt/boost@1.60
make
make install
```

but when I run pcb2gcode now I get:
```
pcb2gcode --front Gerber/WORef/wirelessISP-B.Cu.gbr --metric --metricoutput --zsafe 5 --zchange 5 --zwork 0 --offset 0 --mill-feed 1500 --mill-speed 1000 --zero-start yes --optimise no
Importing front side... DONE.
Importing back side... not specified.
Importing outline... not specified.
Processing input files... libc++abi.dylib: terminating with uncaught exception of type boost::geometry::overlay_invalid_input_exception: Boost.Geometry Overlay invalid input exception
Abort trap: 6
```

a7hybnj2, 2018-08-09:
Great. I'll give that a shot and will post my results here.

I can try any operating system it doesn't have to be mac...

a7hybnj2, 2018-08-09:
Well, I just tried a quick install of boost 1.66 from a .tar.gz and then a reinstall of pcb2gcode. I bunged it up somewhere and although I think I installed 1.66 correctly and I think I linked pcb2gcode it still isn't working.

I will try again in the morning.

eyal0, 2018-08-09:
@a7hybnj2 I tried it a few days ago and it didn't work.

https://svn.boost.org/trac10/ticket/13645

There has been no response on my bug.  I have limited time now but if you know how, maybe you can find who it was that made the changes to bg::simplify and email him with the bug report, to get his attention?  It's not just simplify that was broken, by the way, other stuff, too.

https://github.com/pcb2gcode/pcb2gcode/blob/master/configure.ac#L38

eyal0, 2018-08-09:
Here are my attempts:

```
pcb2gcode --front Gerber/WORef/wirelessISP-B.Cu.gbr --metric --metricoutput --zsafe 5 --zchange 5 --zwork 0 --offset 0 --mill-feed 1500 --mill-speed 1000 --zero-start
Importing front side... DONE.
Importing back side... DONE.
Importing outline... not specified.
Processing input files... DONE.
Exporting back... DONE. (Height: 78.8416mm Width: 104.242mm)
Exporting front... DONE. (Height: 78.8416mm Width: 104.242mm)
Importing drill... not specified.
END.

pcb2gcode -V
1.3.2
Git commit: v1.3.2-288-g7d402ab-dirty
Boost: 106600

```

and with older boost 1.60:

```
pcb2gcode --front Gerber/WORef/wirelessISP-B.Cu.gbr --metric --metricoutput --zsafe 5 --zchange 5 --zwork 0 --offset 0 --mill-feed 1500 --mill-speed 1000 --zero-start
Importing front side... DONE.
Importing back side... DONE.
Importing outline... not specified.
Processing input files... terminate called after throwing an instance of 'boost::geometry::overlay_invalid_input_exception'
  what():  Boost.Geometry Overlay invalid input exception
zsh: abort      ~/private/pcb2gcode/pcb2gcode --front Gerber/WORef/wirelessISP-B.Cu.gbr    5 

pcb2gcode -V
1.3.2
Git commit: v1.3.2-291-g07dbbd0
Boost: 106000
Gerbv: 2.6A

```

The error is very similar but not exact, probably because I'm using Linux and not OSX.  I feel confident that the version of boost is the issue.

I don't have OSX so I don't know how to add boost 1.66 to homebrew!  Maybe you know how or we could get help?  I asked for help here: https://github.com/Homebrew/homebrew-core/pull/30914/comment#issuecomment-411662760

a7hybnj2, 2018-08-09:
@eyal0 thanks for the updates. I look forward to the day boost is straightened out but until then I guess I will try an install on a fresh linux-mint box.

I am currently trying to install boost from the downloaded 'boost_1_66_0.tar.gz' and am running into issues that I can hopefully figure out. something about 'sys/types.h' now being available.

a7hybnj2, 2018-08-09:
Needed those 'build-essential's...doh

Now I am configuring pcb2gcode and I am getting an error after: `./configure --disable-silent-rules --with-boost=/usr/local/boost_1_66_0 --enable-static-boost` that reads ` configure: error: cannot find the flags to link with boost program_options`

a7hybnj2, 2018-08-09:
@eyal0 DOH!

So, I was able to install pcb2gcode 1.3.1 with boost 1.66 and the gcode output still did not have the thermal reliefs/spokes that I was expecting. When you ran my .grb through did you get gcode that included these features?

I was unable to install 1.3.2 because of the errors listed above but 1.3.1 installed without issue... Was thermal reliefs added in 1.3.2?

eyal0, 2018-08-09:
Thermal reliefs were added recently, yes.

I've nearly got pcb2gcode fixed for boost 1.68. If you can wait a day, you'll be able to use the latest boost with the latest pcb2gcode.

a7hybnj2, 2018-08-09:
@eyal0 What an adventure! Yea, I think I will just wait...Thanks for all the help!

Now irrelevant: https://github.com/a7hybnj2/KicadStuffs/blob/master/pcb2gcode_experience/From1.3.1w1.66.nc

a7hybnj2, 2018-08-10:
@eyal0 Hey, I got impatient and tried to load boost 1.68 with your pcb2gcode repo and I got error:
`checking for BOOST_VERSION >= 105600 && BOOST_VERSION < 106700... no
configure: error: not true: BOOST_VERSION >= 105600 && BOOST_VERSION < 106700`

eyal0, 2018-08-10:
You could have just checked out the simplify_mls branch.

Anyway, it's in now. Try it out!

closed 2018-08-10 by eyal0