pcb2gcode/pcb2gcode #195: Make error

a7hybnj2, 2018-08-10:
I am getting this error trying to make pcb2gcode with boost 1.68

[link to log file](https://github.com/a7hybnj2/KicadStuffs/blob/master/pcb2gcode_experience/log.log)

eyal0, 2018-08-10:
Looks like it was supposed to be "std::isinf" in place of "isinf" in all places. Can you check if that substitution works? I didn't need it but maybe my complication platform is more lenient or something.

a7hybnj2, 2018-08-10:
@eyal0 I can check it...if you let me know how... I know just enough to get into trouble.

The current complication is taking place on a clean spin-up of linuxmint 18.? if that matters.

a7hybnj2, 2018-08-10:
Let me know if I should make a new issue for this one.

[Link to output](https://github.com/a7hybnj2/KicadStuffs/tree/master/pcb2gcode_experience/With%20boost%201.68%20and%20pcb2gcode%209ff6a3b%20)

The output is so close to being perfect. It is still missing parts but has more then it did.

eyal0, 2018-08-10:
Okay, off the top of my head:  What is your setting for offset?  Maybe the offset is bigger than the size of the thermal?  If the milling bit is bigger than the thermal relief, the thermal relief will go away.  To test this, try setting your your offset to 0 temporarily.

eyal0, 2018-08-10:
As in, `--offset=0`

closed 2018-08-10 by eyal0