pcb2gcode/pcb2gcode #215: configure fails at librsvg-2.0 > 2.0

staviq, 2018-09-23:
Attempting to compile on osx from master, via homebrew --HEAD fails at the configure step:

checking for librsvg-2.0 > 2.0... no
configure: error: Package requirements (librsvg-2.0 > 2.0) were not met:

No package 'librsvg-2.0' found

Shouldn't configure.ac say ">=" instead of ">" ?

closed 2018-09-26 by eyal0

hedefalk, 2019-01-22:
I have a weird deja vu here, but I think I might have had this problem before. For some reason I can't build within homebrew, but just checking out master and doing:

```
autoreconf
./configure
make install
```
 and I'm fine.

xunker, 2019-02-01:
@hedefalk I ran in to the same issue, and after a lot of work manually setting paths, etc, I found [this post](https://discourse.brew.sh/t/pkg-config-does-not-respect-pkg-config-path/895/4) that prompted me to try:

`brew install --env=std --HEAD pcb2gcode`

..which worked for me on MacOS 10.14.2. That link says `--env=std` is not recommended but it did work, YMMV.

PR #220 (2018-09-26): Allow librsvg >= 2.0
This fixes #215
commit adabd12e (2018-09-26): Allow librsvg >= 2.0

This fixes #215