pcb2gcode/pcb2gcode #216: SVG files wrong dimensional size

cyber-murmel, 2018-09-23:
I want to use a laser cutter to engrave into an PCB covered in acrylic spray paint to then etch it. Inkscape was unable to convert the strokes to paths properly. This tools seems to work, since it produces SVG as intermediate files. The only problem is that they are not to scale.

## pcb2gcode version (run pcb2gcode --version to see this):
```
marble@archlinux ~ % uname -a
Linux archlinux 4.18.9-arch1-1-ARCH #1 SMP PREEMPT Wed Sep 19 21:19:17 UTC 2018 x86_64 GNU/Linux
marble@archlinux ~ % pcb2gcode --version
1.3.2
Git commit: v1.3.2-434-g0f24bdb
Boost: 106800
Gerbv: 2.6A
```

## What did you try (include command-line arguments):
`pcb2gcode --front=/tmp/test-F.Cu.gbr --output-dir=/tmp --metric=false --metricoutput=false --zchange=1 --zsafe=1  --mill-feed=1 --mill-speed=0 --zwork=0`

`pcb2gcode --front=/tmp/test-F.Cu.gbr --output-dir=/tmp --metric=true --metricoutput=false --zchange=1 --zsafe=1  --mill-feed=1 --mill-speed=0 --zwork=0`

`pcb2gcode --front=/tmp/test-F.Cu.gbr --output-dir=/tmp --metric=false --metricoutput=true --zchange=1 --zsafe=1  --mill-feed=1 --mill-speed=0 --zwork=0`

`pcb2gcode --front=/tmp/test-F.Cu.gbr --output-dir=/tmp --metric=true --metricoutput=true --zchange=1 --zsafe=1  --mill-feed=1 --mill-speed=0 --zwork=0`

## What happenned:
The SVGs that are generated are scaled up by a factor of `20.833`. A trace that is supposed to be 1mm wide is 20.833mm wide.

## What did you expect to happen:
SVGs are scale (1mm wide trace in pcbnew is 1mm wide trace in SVG)

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

cyber-murmel, 2018-09-24:
I managed to "fix" it by changing altering this line in `geometry.hpp`:
```
#define SVG_PIX_PER_IN 96.15384615384615
```

eyal0, 2018-09-24:
It's actually not 96.15..., it's just 96.  That's because 96 pixels per inch is the CSS standard.

I'll change the value to 96.  Can you let me know why you didn't see exactly 96?  Maybe it's due to a rounding error?

cyber-murmel, 2018-09-24:
You are right. Yeah, I think rounding error. I was using Inkscape to measure, which only shows 3 decimal numbers

commit 34788773 (2018-09-26): Change the width and height but not the viewbox parameters.

This fixes #216
closed 2018-09-26 by eyal0