pcb2gcode/pcb2gcode #222: Weird interaction between min-milldrill-hole-diameter and drills-available

chaotix-, 2018-10-15:
pcb2gcode version (run pcb2gcode --version to see this):
1.3.2
Git commit: v1.3.2-458-gab34a3c
Boost: 106200
Gerbv: 2.6.1

When using min-milldrill-hole-diameter and drills-available the bits are rounded as expected, but the hole diameter for milldrill use the rounded values and not the actual hole diameter.

Example:
```
min-milldrill-hole-diameter=1.2mm
milldrill-diameter=1.0
drills-available=0.4mm,0.6mm,0.8mm,1.0mm,1.2mm
```
This will drill all holes > 1.2mm with the 1.2mm drill. No milldrill.ngc file will be generated.

Appending 1.3mm to drills-available will use milldrill for holes bigger than 1.2mm, but all bigger holes will be milled to 1.3mm.

A less sensible example that should work with all boards to demonstrate the problem:
```
min-milldrill-hole-diameter=1.2mm
milldrill-diameter=1.0
drills-available=20mm
```

All holes (even the ones < 1.2mm) will be milldrilled to 20mm with a 1mm bit. (No drill.ngc will be generated.)

I think milldrill should use the actual hole sizes and not the rounded drill bit size for milling.

eyal0, 2018-10-15:
> I think milldrill should use the actual hole sizes and not the rounded drill bit size for milling.

Oh yeah, that was definitely the intent.  If it doesn't work that way, I got it wrong!

If you have a sample configuration and Excellon drill file for me try, I can add it to the pcb2gcode integration tests both to help fix this issue and make sure that it doesn't break again in the future.

Meanwhile, I'll try to work up my own example based on your report.  Thanks!

chaotix-, 2018-10-15:
OK. I prepared a very simple drill and configuration file. It's just 10 holes with diameters from 0,2mm to 2,0mm in steps of 0,2mm. At the moment all holes >= 1,2mm will be milldrilled to 1,2mm.

[milldrilldiatest.zip](https://github.com/pcb2gcode/pcb2gcode/files/2479273/milldrilldiatest.zip)

The comment "( Bit sizes: [1.2 mm] )" in milldrill.ngc should probably also be removed.

eyal0, 2018-10-22:
Let me know if more work is needed.

closed 2018-10-22 by eyal0

chaotix-, 2018-10-23:
Looks great. Thanks a lot.

PR #231 (2018-10-22): Min milldrill fix
This fixes #222