pcb2gcode/pcb2gcode #223: Use helix for milldrill holes

chaotix-, 2018-10-15:
pcb2gcode version (run pcb2gcode --version to see this):
1.3.2
Git commit: v1.3.2-458-gab34a3c
Boost: 106200
Gerbv: 2.6.1

Milldrilling uses several circles at different depth to cut a hole. Wouldn't it be better for the mill and the endmills to use a helix? It even would slim down the resulting G-Code.

Example milldrill.ngc hole (from the test files in issue #222):
```
G0 X130.26000 Y-90.00000
G1 Z-0.29167
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000
G1 Z-0.58333
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000
G1 Z-0.87500
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000
G1 Z-1.16667
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000
G1 Z-1.45833
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000
G1 Z-1.75000
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000
G0 Z1.50000
```

An optimized version would look like this:
```
G0 X130.26000 Y-90.00000
G1 Z0
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000 Z-1.75000 P7
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000
G0 Z1.50000
```

In case GRBL does not know about the P-parameter (at least the preview in bCNC does not) this would be a variant without P, but still a nice helix:
```
G0 X130.26000 Y-90.00000
G1 Z0
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000 Z-0.29167
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000 Z-0.58333
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000 Z-0.87500
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000 Z-1.16667
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000 Z-1.45833
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000 Z-1.75000
G2  X130.26000 Y-90.00000 I-0.10000 J0.00000
G0 Z1.50000
```

eyal0, 2018-10-15:
I have thought about this, yes!  I will look into it.

I think that support for P might be limited.  I couldn't find support for it in the grbl source code.  Is it there?  Probably safest to just write out the loops because that will have the widest support and wide support is more important than filesize, I believe.

The helix solution would need to also support G85 slots ([read more about slots](https://gist.github.com/katyo/5692b935abc085b1037e)).  How do you suggest to do helix mode for this?

Also, you noticed that the final pass must not be a helix but a flat circle.  This is true!  This also needs to be checked.

I'll look into this.  If you want to help with coding and you know some c++, I can point you to the location to fix and I am happy to have help.

chaotix-, 2018-10-15:
Just glanced over the GRBL code and also could not find support for P with G2/G3. I think support for Z is there. Maybe you could check at your machine to be sure? Just position it somewhere in the middle, zero Z and issue a "G2 Z-10 I-2 F500" (change Z-10 if you are using inches!). Does the spindle move down 10mm?

I can see if I can make a pull request at least for the holes. Should be easy enough. (Is there a reason for the X and Y parameters when using arcs? Do some controllers need those when doing a full circle?)

Slots are a lot more complicated... You need to calculate the traveled distance for one pass and then divide the Z moves evenly between the two half circles and the straight lines. Maybe I'll start with the holes first. :-)

chaotix-, 2018-10-18:
Implemented in #224

closed 2018-10-18 by chaotix-