pcb2gcode/pcb2gcode #232: pcb2gcode won't run.

Hactar42, 2018-11-08:
Problem with pcb2gcode-1.3.2-win32   Installed in Windows 10 Home. Ver:	10.0.17134 Build 17134

When I try to run pcb2gcode.exe: A small screen flashes quickly (I cannot read what it says) then the program quits.  Windows task manager does not show it running.

When I run pcb2gcode.exe as an administrator:  The screen turns totally black for a few seconds then the program quits.

I expected the program to run.  My system seems to be running perfectly with no problems with any  other applications.

I have a PCB design completed in Kicad pcbnew and need to convert it to grbl CNC router.

Can't attach any related files because the program never ran.

 Any advice or help would be much appreciated.

eyal0, 2019-02-23:
No progress here in a long time.  I'll close this.

closed 2019-02-23 by eyal0

eyal0, 2021-01-07:
This is fixed by b88a2fbd and 011336db and 6d15f726