pcb2gcode/pcb2gcode #233: armhf Raspiam Stretch error: Exporting back... Internal Error: vector::_M_range_insert

Joel-Mckay, 2018-11-12:
pcb2gcode version: git master Nov 12
 expected: kicad gerber to linuxcnc gcode with probing

These were plotted from kicad 5.0.0, and used a very simple pcb layout to test.

It seems to happen during the top/bottom layer output stages. but does not seem related to the drill or auto leveler subroutines. the test command is included.

[pcb2gcode.zip](https://github.com/pcb2gcode/pcb2gcode/files/2571084/pcb2gcode.zip)

Joel-Mckay, 2018-11-16:
2016 v1.3.2 snap appears to build and work with static boost_1_68_0
https://sourceforge.net/projects/pcb2gcode/files/pcb2gcode/pcb2gcode-1.3.2.tar.gz/download

git master compiles, but throws the same vector::_M_range_insert error during export
#boost_1_68_0
wget https://dl.bintray.com/boostorg/release/1.68.0/source/boost_1_68_0.tar.gz
tar xvf boost_1_68_0.tar.gz
rm boost_1_68_0.tar.gz
cd SRC/boost_1_68_0
./bootstrap.sh --with-libraries=program_options --prefix=/opt/boost_1_68_0
./b2 -a variant=release link=static
./b2 install
sudo ./b2 install

Does not compile pcb2gcode with legacy boost
#boost_1_58_0
wget https://sourceforge.net/projects/boost/files/boost/1.58.0/boost_1_58_0.tar.gz/download
mv download boost_1_58_0.tar.gz
tar xvf boost_1_58_0.tar.gz
rm boost_1_58_0.tar.gz
cd boost_1_58_0
./bootstrap.sh --with-libraries=program_options --prefix=/opt/boost_1_58
./b2 -a variant=release link=static
sudo ./b2 install

make clean
 autoreconf -fvi
 ./configure --with-boost=/opt/boost_1_58 --enable-static-boost
 make

make  all-recursive
make[1]: Entering directory '/home/pi/SRC/pcb2gcode'
Making all in man
make[2]: Entering directory '/home/pi/SRC/pcb2gcode/man'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/pi/SRC/pcb2gcode/man'
make[2]: Entering directory '/home/pi/SRC/pcb2gcode'
  CXX      autoleveller.o
In file included from /opt/boost_1_58/include/boost/bind/mem_fn.hpp:25:0,
                 from /opt/boost_1_58/include/boost/mem_fn.hpp:22,
                 from /opt/boost_1_58/include/boost/function/detail/prologue.hpp:18,
                 from /opt/boost_1_58/include/boost/function/function_template.hpp:13,
                 from /opt/boost_1_58/include/boost/function/detail/maybe_include.hpp:18,
                 from /opt/boost_1_58/include/boost/function/function1.hpp:11,
                 from /opt/boost_1_58/include/boost/program_options/value_semantic.hpp:13,
                 from /opt/boost_1_58/include/boost/program_options/options_description.hpp:13,
                 from /opt/boost_1_58/include/boost/program_options.hpp:15,
                 from autoleveller.hpp:42,
                 from autoleveller.cpp:20:
/opt/boost_1_58/include/boost/get_pointer.hpp:27:40: warning: 'template<class> class std::auto_ptr' is deprecated [-Wdeprecated-declarations]
 template<class T> T * get_pointer(std::auto_ptr<T> const& p)
                                        ^~~~~~~~
In file included from /usr/include/c++/6/memory:81:0,
                 from autoleveller.hpp:39,
                 from autoleveller.cpp:20:
/usr/include/c++/6/bits/unique_ptr.h:49:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
In file included from /opt/boost_1_58/include/boost/smart_ptr/shared_ptr.hpp:28:0,
                 from /opt/boost_1_58/include/boost/shared_ptr.hpp:17,
                 from /opt/boost_1_58/include/boost/program_options/options_description.hpp:16,
                 from /opt/boost_1_58/include/boost/program_options.hpp:15,
                 from autoleveller.hpp:42,
                 from autoleveller.cpp:20:
/opt/boost_1_58/include/boost/smart_ptr/detail/shared_count.hpp:395:33: warning: 'template<class> class std::auto_ptr' is deprecated [-Wdeprecated-declarations]
     explicit shared_count( std::auto_ptr<Y> & r ): pi_( new sp_counted_impl_p<Y>( r.get() ) )
                                 ^~~~~~~~
In file included from /usr/include/c++/6/memory:81:0,
                 from autoleveller.hpp:39,
                 from autoleveller.cpp:20:
[... 119 lines cut ...]
                 from autoleveller.hpp:44,
                 from autoleveller.cpp:20:
/opt/boost_1_58/include/boost/smart_ptr/scoped_ptr.hpp: At global scope:
/opt/boost_1_58/include/boost/smart_ptr/scoped_ptr.hpp:68:31: warning: 'template<class> class std::auto_ptr' is deprecated [-Wdeprecated-declarations]
     explicit scoped_ptr( std::auto_ptr<T> p ) BOOST_NOEXCEPT : px( p.release() )
                               ^~~~~~~~
In file included from /usr/include/c++/6/memory:81:0,
                 from autoleveller.hpp:39,
                 from autoleveller.cpp:20:
/usr/include/c++/6/bits/unique_ptr.h:49:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
In file included from autoleveller.hpp:44:0,
                 from autoleveller.cpp:20:
geometry.hpp:81:12: error: redefinition of 'struct boost::polygon::coordinate_traits<long long int>'
     struct coordinate_traits<int64_t> {
            ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/boost_1_58/include/boost/polygon/polygon.hpp:12:0,
                 from geometry.hpp:29,
                 from autoleveller.hpp:44,
                 from autoleveller.cpp:20:
/opt/boost_1_58/include/boost/polygon/isotropy.hpp:180:10: error: previous definition of 'struct boost::polygon::coordinate_traits<long long int>'
   struct coordinate_traits<polygon_long_long_type> {
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Makefile:901: recipe for target 'autoleveller.o' failed

jacen92, 2018-12-16:
I got the same problem, after investigations it seems the commit dd7a96d1 introduce a regression on arm.
In the voronoi.cpp file in the function build_voronoi() the ring is always empty and the call to ring.insert() raise this exception.
In the loop there is no safeguard for ring.push_back(ring.front());  causing a core dump when the exception is catched.
I am working on a patch to fix these two issues. I let this comment because I don't know when I will be ready to do a pull request.

eyal0, 2018-12-17:
It should work with latest boost. I will try it.

jacen92, 2018-12-17:
I made a pull request: https://github.com/pcb2gcode/pcb2gcode/pull/238

eyal0, 2018-12-18:
@jacen92 Thanks for your work!  I don't know why it's failing on arm when it works on my desktop, even with latest boost.  Maybe that new boost wasn't actually installed correctly?  I've had many issues when I try to install boost and somehow the old files got used.

I merged, thanks!

closed 2018-12-18 by eyal0