pcb2gcode/pcb2gcode #234: msys2 tries to run pcb2gcode but gives an error:  "Safety height not specified"  Help please

Hactar42, 2018-11-16:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):

What did you try (include command-line arguments):

What happenned:

What did you expect to happen:

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

Hactar42, 2018-11-17:
I have designed some PCBs in Kicad Ver. (5.0.1)-3, release build and have not yet found out how I can generate CNC gcode for my grbl based CNC.

I have tried MSYS2 to launch pcb2gcode but have been unsuccessful
Using:

MSYS2.EXE Version:  GNU bash, version 4.4.19(3)-release (x86_64-pc-msys)
PCB2GCODE Version:  pcb2gcode-1.3.2-win32

When I try to run Pcb2gcode.exe using the MSYS2 command ./  I get an error “safety height not specified”

I have never gotten to the point where I could load any gerbers.

eyal0, 2018-12-18:
That error means that you didn't configure a variable.  Did you set z-safe?

eyal0, 2020-06-28:
Haven't heard back in too long.

closed 2020-06-28 by eyal0