pcb2gcode/pcb2gcode #237: when outline cutting, do not mill around pcb

AlexeyGusev, 2018-12-17:
```
$ pcb2gcode --version
1.3.2
Git commit: v1.3.2-491-g9af2eac
Boost: 106200
Gerbv: 2.6.1
```

When --outline is given, pcb is about to be cut out from FR4. This makes milling around the often present ground plane useless, because CNC will cut along (or in close proximity of) the milled track around ground polygon. This effect is especially pronounced if high extra-passes parameter is used (exaggerated for clarity on this pcb, but might still be needed if you want thick isolation inside pcb in certain places):

[image]

Generated outline:

[image]

It should be easy to detect and apply the no-mill rule on the outmost milling route, even if it doesn't fit into the cutting track.

I propose implementing this feature with `mill-around=false` or similar parameter, since this feature might not always be wanted.

eyal0, 2018-12-18:
How come the outline has so many passes but the traces don't?  Is it because of a ground pour?

Options:
1. Leave it as is: lots of wasted time and milling
2. Never `mill-around`: This is a problem for board where there is no outline cut, like if you want to put multiple PCBs on a single FR4.
3. Only mill-around if there is no outline.  If there is an outline, never mill-around.
4. Add `mill-around` option.

I think that 1 and 2 are bad.  Either 3 or 4 could be fine.  I like 3 because there is no extra parameter.  That is nice.  Can you think of a reason why someone might want `mill-around` true even if there is an outline?  By the way, when no outline is provided, the code makes a rectangle around all the traces and pretends that this is the outline, and that is what gets milled-around.  Maybe that affects the decision?

AlexeyGusev, 2018-12-19:
> How come the outline has so many passes but the traces don't? Is it because of a ground pour?

Yeah, because the board has a ground plane. Picture of original kicad board:
[image]

> Can you think of a reason why someone might want mill-around true even if there is an outline? 

if the outline is far away from the ground plane and elements, and you want isolation around ground polygon. Or if the board has irregular shape. In other words, when the mill-around track does not match the cutter tool path.

eyal0, 2018-12-19:
> mill-around track does not match the cutter tool path.

But those two are always the same. The outer most path for milling around is always taken from the outline. So they will always be the same.

Like say you have a design with just one trace and no edge cuts provided. Then the mill around path is going to be a rectangle around that trace. But if you do have edge cuts set then the mill around will be the same as the cutter path.

Basically, I think that the way that pcb2gcode is written, it is impossible to have the cutter path be different from the mill around path. If you don't specify a cutter than no cutting file is generated. If you don't specify edge cuts then that path is just the bounding rectangle.

Maybe you can provide a concrete example where it would be different?

AlexeyGusev, 2018-12-19:
Hmm, I am not sure what you mean by «mill around path», in my case it's a path around outermost polygon (copper fill), and that doesn't necessarily matches the cutter tool path. Example (examine kicad board and gerbers here: [test.zip](https://github.com/pcb2gcode/pcb2gcode/files/2696318/test.zip), screenshots provided below for convenience):

Kicad board with a copper fill and cutter tool path. Notice that copper fill area has bevels in top left and bottom right corners. Copper fill min isolation is set to 0.254 mm, and this is what defines isolation between two traces and the copper fill.

[image]
 
gerbv visualization:

[image]

and g-code files generated with `extra-passes=10`, command:
`$ pcb2gcode --front test-F.Cu.gbr --outline test-Edge.Cuts.gbr --drill test-PTH.drl`

for milling:

[image]

and cutting (tilted for clarity):

[image]

eyal0, 2018-12-20:
Okay, I understand my confusion now.  I was thinking about `voronoi` mode.  With voronoi, the edge cuts are used as the outer boundary and if the edge cuts don't exist, the bounding rectangle is used.  So let's put voronoi aside and only look at `voronoi=false`.

For your first example, the blue and yellow one, the mill around isn't needed because there is only a single net between the edge cuts and the ground plane.

But in your second example, it isn't obvious.

[image]

Green traces 1 and 2 are obviously both part of the ground plane.  I know that, of course, and I know that it doesn't matter if those two are connected.  But pcb2gcode doesn't know that those are two actually the same net and don't need to be isolated.  So it might be hard to fix that in software.

One solution is for you to not do that pour.  pcb2gcode does isolation milling, so it's going to leave all that copper in place whether or not you put that pour in kicad.  So maybe you can just draw your traces without the pour and it'll do the right thing.  Can you try that?

Especially check that the thermal relief isn't ruined.

Thanks!

AlexeyGusev, 2018-12-21:
> For your first example, the blue and yellow one, the mill around isn't needed because there is only a single net between the edge cuts and the ground plane.

Agree.

> One solution is for you to not do that pour. 

Not always. Example with two traces and a copper pour around them is completely artificial, but reflects the idea of having mount pads – imagine there are triangular brackets in two opposite corners of the PCB above – isolated from the copper pour.

> So maybe you can just draw your traces without the pour and it'll do the right thing. Can you try that?

Yeah. Here:
[image]
[image]

But the software knows precisely if the mill-around path, fully or partially, matches the cutter path, and can avoid milling there, right? Perhaps with some proximity limit, like if the outmost milling path comes as close as, say, 2mm to cutter path, don't mill there. And of cause, don't mill outside cutter path. Like here, reddish rectangles show areas where milling should be avoided:

[image]

eyal0, 2018-12-22:
Yeah, it doesn't seem to make sense that the milling should go where the cutter is going to work!  I'll clip all the milling paths to within the cutting path.  And if there's no cutting path then I'll leave it as it was.

eyal0, 2019-01-02:
I think that this should fix it: https://github.com/eyal0/pcb2gcode/tree/mill_within_outline

Want to try it out and let me know?  If it looks good to you, I'll merge it.  Thanks!

AlexeyGusev, 2019-01-03:
tested 456c7a1b660b69d9bab64b624153f36ecbc0bbc9

parameters:
offset=0.1mm
extra-passes=20
cutter-diameter=1.5mm

In all examples below, center line of each cut trace exactly matches each side of the copper polygon, in other words, they have the same coordinates. Let's look at three test cases below, first two without taking _actual_ cut trace width into consideration:

- **all elements/traces of your board are inside a polygon**
With polygon isolation = 0 (no gap between cutter traces and polygon):
[image]
pcb2gcode still generates one mill-around path:
[image]
More mill-arounds added if polygon has an isolation gap:
[image]
[image]
In my opinion, in first case above there should be no mill-around paths generated at all. Second example is probably okay.

- **some or all tracks are not contained within a polygon**
pcb2gcode does as much extra passes around _outermost_ traces as possible, limits itself at cutter trace. 
[image]
[image]
Seems to be an extreme variation of previous test case. Correct behaviour.

- **cut traces are wide enough to cover significant part of the polygon**
Now let's consider cut traces that are of proper width, being equal to width of cutter tool, 1.5mm in our case. This seems to confuse pcb2gcode, it generates 16, 18 or all 20 mill-arounds in various places:
[image]
[image]
It seems that mill-arounds are generated up to the centerline of each cut trace. Shouldn't pcb2gcode take cutter tool width ( cutter-diameter parameter) into account and get rid of all milling paths that fit into whole cutter trace, not just centerline?

eyal0, 2019-01-04:
I see. Yes, you are right, I only took the centerline of the cutter into account and not the width of the cutter. I will do that and then you can try again. Thanks!

eyal0, 2019-01-04:
Actually, maybe that isn't the centerline. Are you sure that the very outermost path is the centerline of the cutter and not the inside edge of the cutter's path? I will also check.

AlexeyGusev, 2019-01-04:
> Actually, maybe that isn't the centerline. Are you sure that the very outermost path is the centerline of the cutter and not the inside edge of the cutter's path? I will also check.

Yes, it definitely is not the inside edge, see my exaggerated cutter trace example above.

> But what if we're wrong about the radius of the cutter?

Okay, I see your concern, but g-code generation stage is a wrong place to take care of that, IMHO. One needs to double check his dimensions and tool sizes before this final step. On the other hand, if just one milling path is generated around the board, that's fine.

> That happens a lot for me because I use V-shaped bits and I don't really know the diameter as it depends on depth.

It's actually tan(bit_angle/2)×mill_depth×2 + tip_width. Will be pretty consistent everywhere if you use height maps.

How do you generate the cutter path? Aligned to the center of the cut trace (center of the cutter tool passes along the centerline of the cut trace) or outside the cut trace (inward most cutter tool point passes along the centerline of the cut trace)?

eyal0, 2019-01-04:
> How do you generate the cutter path? Aligned to the center of the cut trace (center of the cutter tool passes along the centerline of the cut trace) or outside the cut trace (inward most cutter tool point passes along the centerline of the cut trace)?

Good question.  I just checked the code and I am doing it wrong.  I have the inward most cutter tool point passes along the **outermost** edge of the cut trace.  I will fix that.  I want the inward most cutter tool point to pass along the **centerline** of the cut trace.  This was a mistake that I made and I filed report #243 .

I already have an idea how to fix that.  After I do, then I'll get back into this issue because that one affects this one.

eyal0, 2019-01-08:
I'm happy with the solution that I found.  Can you check how it looks on your examples above?  Thanks!

AlexeyGusev, 2019-01-08:
Yeah, looks like milling file is generated according to discussions above, thanks! Haven't checked how precisely the outline matches the milling though. 

One issue seems to arise when the board edge (for the cutter tool) is drawn with a thick line (1mm): 

```
pcb2gcode --front test-F.Cu.gbr --outline test-Edge.Cuts.gbr --drill test-PTH.drl
…
Processing input files... Found an unconnected loop while parsing a gerber file while expecting only loops
Found an unconnected loop while parsing a gerber file while expecting only loops
DONE.
```

An empty milling g-code file is generated:

```
( pcb2gcode 1.3.2 )
( Software-independent Gcode )

G94 ( Millimeters per minute feed rate. )
G21 ( Units == Millimeters. )

G90 ( Absolute coordinates. )
G00 S7000 ( RPM spindle speed. )
G64 P0.01016 ( set maximum deviation from commanded toolpath )
G01 F180.00000 ( Feedrate. )

G01 F180.00000 ( Feedrate. )
M3 ( Spindle on clockwise. )
G04 P1.00000

G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G53 G00 Z-1.000 ( retract )

M5 ( Spindle off. )
G04 P1.000000
M9 ( Coolant off. )
M2 ( Program end. )
```

eyal0, 2019-01-08:
That error means that the outline is not a continuous line so pcb2gcode couldn't figure out the area of the board.  Like maybe your gerber file has this:

    +-------+
    |       |
    |       |
    +-----  +

The loop isn't connected.

According to everything that I've read online, the thickness of the edge cuts line is ignored and the fab will cut along the center of the line.  So your board won't even disconnect from the source material!

eyal0, 2019-01-08:
If you send me the outline gerber file, I can look in it to see if it's really true that it's not a loop.  Or you can shrink the width of the edge cut really small to see it yourself.

AlexeyGusev, 2019-01-08:
okay, wrong diagnosis probably.

actually, the loop is connected, but one of the lines is thicker than the others. I didn't move it, just increased the width ×10. Note that endpoints are exactly where they should be:
[image]

if I change the width back and replot, the error goes away. If I increase the width of all edge lines, the error goes away. And since this case is somewhat exotic, it is probably just worth disregarding.

eyal0, 2019-01-08:
Ah yes, okay, I treat different width lines separately when I'm looking for closed loops.  I guess that I could treat them as all the same.  It would be easy to fix but it's such an odd case that I don't know if there is any point!

cyber-murmel, 2019-02-01:
I think this is a problem with tackling. I just spent several hours trying figure out this behavior :smile:

cyber-murmel, 2019-02-01:
Oh, just found out, that also KiCAD has some trouble exporting arcs or something.
Sometimes, when I export, pcb2gcodeGUI also throws the `Processing input files... Found an unconnected loop while parsing a gerber file while expecting only loops ` error. When I then fiddle with the arc end export again, it suddenly works.

eyal0, 2019-02-01:
Is it possible that your edge cuts didn't create a closed loop and that's why it failed? pcb2gcode needs to work with shapes that have area so it uses the edge cuts as if they are egress and looks for closed loops. Otherwise we can tell what is inside the PCB and what isn't.

Because of how pcb2gcode is written, it doesn't try to form loops with lines of different width. I could change that easily enough. Would that solve your issue?

If you send me the problematic input files, I can take a look.

BTW, I don't support pcb2gcodeGUI because I don't have the time for it. Are you able to use the command line interface?

cyber-murmel, 2019-02-02:
## Zip Files
[not_ok.zip](https://github.com/pcb2gcode/pcb2gcode/files/2824461/not_ok.zip)
[ok.zip](https://github.com/pcb2gcode/pcb2gcode/files/2824462/ok.zip)

## pcb3gcode Command Line
```
pcb2gcode \
--front=/tmp/laser_cutter_z_gbr/laser_cutter_z-F.Cu.gbr \
--outline=/tmp/laser_cutter_z_gbr/laser_cutter_z-Edge.Cuts.gbr \
--output-dir=/tmp/laser_cutter_z_ngc \
--metric=true \
--metricoutput=true \
--nog64=false \
--optimise=true \
--tile-x=1 \
--tile-y=1 \
--tolerance=0.0100 \
--vectorial=true \
--zchange=10.0000 \
--zchange-absolute=false \
--zero-start=true \
--zsafe=5.0000 \
--mill-feed=600 \
--mill-speed=10000 \
--voronoi=true \
--zwork=-0.1400 \
--bridges=0.5000 \
--bridgesnum=3 \
--cut-feed=30 \
--cut-infeed=10.0000 \
--cut-side=auto \
--cut-speed=10000 \
--cutter-diameter=3.1750 \
--fill-outline=true \
--zbridges=-0.6000 \
--zcut=-2.0000
```

## KiCAD Version Info
```
Application: kicad
Version: 5.0.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.63.0 OpenSSL/1.1.1a zlib/1.2.11 libidn2/2.1.0 libpsl/0.20.2 (+libidn2/2.1.0) libssh2/1.8.0 nghttp2/1.35.1
Platform: Linux 4.20.5-arch1-1-ARCH x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 2.24
    Boost: 1.68.0
    OpenCASCADE Technology: 7.3.0
    Curl: 7.61.1
    Compiler: GCC 8.2.1 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

## Plot Settings
[image]
[image]

eyal0, 2019-02-04:
Thanks for the files.

In the not_ok version, the lines almost but don't quite make a loop.  This is what is causing the problem.

If you look in the "ok" Edge.Cuts, you will see a line like this:

`X63500000Y-114300000D02*`

That a straight line ending at coordinate 63.5mm and -114.3mm.

In the not-ok file, there is a line like this:

`X63500001Y-114299999D02*`

That's a straight line ending at coordinate 63.500001mm and -114.299999mm.  It's close but not the same!  That little gap caused the loop to appear to be not closed, and that caused the problem.  :sob: 

How to fix it?  I could round all the numbers to the nearest 0.00001, for example.  I'm not sure what the right thing to do it.  This is kind of a KiCad problem because it should not be making numbers that are off by that small amount!  Yours is a modern version of KiCad.  Maybe they already know about this bug?

eyal0, 2019-02-20:
This is a problem with the input.  I'm not sure how to better handle it.  KiCad shouldn't be making things like this!

PR #253 (2019-01-08): Mill within outline
Confine milling to within the outline of the board so that we don't unnecessarily mill beyond the cut edge.  However, don't let that confinement keep us from milling outside the cut edge if the alternative is that we impinge on a trace that we're milling.  Finally, keep all that milling within the voronoi region because we don't want to impinge on other traces either.

This fixes #237
closed 2019-02-20 by eyal0