pcb2gcode/pcb2gcode #287: EasyEDA solid copper issue

Pmarva, 2019-02-13:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
2.0.0
Git commit: v2.0.0-5-gf55dc52
Boost: 106400
Gerbv: 2.6.2

What did you try (include command-line arguments):

pcb2gcode \--software linuxcnc --zero-start \--front *.GTL --front-output front.ngc \--back *.GBL --back-output back.ngc \--drill *.DRL --drill-output drill.ngc --drill-side back \--outline *.GKO --outline-output cutout.ngc --fill-outline --outline-width 0.1 \--metric --metricoutput --noconfigfile \--zsafe 3 --zchange 50 \--cut-side back --cut-feed 150 --cut-speed 6000 --cut-infeed 0.6 --zcut -1.8 \--zbridges -1 --bridges 3 --bridgesnum 4 --cutter-diameter 2 \--mill-feed 500 --mill-speed 6000 --zwork -0.15 --offset 0.15 \--drill-feed 500 --drill-speed 6000 --zdrill -2 \--spindown-time 2 --spinup-time 2 \--tile-x 5 --tile-y 1

What happenned:
Exported Gerber from EasyEDA the copper ground is correct but in G-code there is something wrong. I'm not specialist and do not know much about cnc and G-code so i may do something wrong here.

Correct PCB view from EasyEDA:
[image]

gerbV view also correct, based on gerber files from EasyEDA
[image]

G-code svg and camotics  based on G-code
[image]
[image]

GND copper is not correct there

Also source files:
[PCB.svg.txt](https://github.com/pcb2gcode/pcb2gcode/files/2860921/PCB.svg.txt)
[gerber_test.zip](https://github.com/pcb2gcode/pcb2gcode/files/2860922/gerber_test.zip)
[PCB.json.txt](https://github.com/pcb2gcode/pcb2gcode/files/2860923/PCB.json.txt)

Am I doing something wrong in EasyEDA or with pcb2gcode?

eyal0, 2019-02-13:
That looks very wrong!  I will try it out and let you know.  Thanks.

eyal0, 2019-02-14:
First, thanks for your good bug report.  I appreciate it when I can just copy and paste the provided command-line and run it to make it work.

Second, yes, I see similar issues.  It looks like the code that reads the gerber file and tries to convert it into shapes is failing.  I'm not yet sure why but I'll start looking into it.  EasyEDA seems to have made a sort of complicated shape so it's probably related to that.  I don't usually see KiCad make such weird shapes.  But the shape is legal so it has to work!

eyal0, 2019-02-14:
Okay, I think that I've found the problem and I have a possible solution.  Would it be okay with you if I add your gerber files into the pcb2gcode testing?  It would make it so that your situation is tested for each new version to make sure that it doesn't crop up again.  Let me know if you're okay with that.

Pmarva, 2019-02-14:
Hey EasyEDA just released updates and now the output is much better from pcb2gcode.
[image]

eyal0, 2019-02-14:
Oh hey, that's great!  I wonder what they fixed.

eyal0, 2019-02-21:
I added some code in 299438bd94efb8da8b5f47749e88b5e6903a4775 to do a better job on cases like these.  It's still not perfect but I found it helpful for one of my unit tests so I'm leaving it in.

Nice to hear that EasyEDA is generating better outputs now.

closed 2019-02-21 by eyal0