pcb2gcode/pcb2gcode #291: Mill the area of a stencil entirely away (not just contour)

chiefenne, 2019-02-17:
pcb2gcode version (run pcb2gcode --version to see this): 1.3.2

I would like to mill the area of a stencil entirely away not just the contour. I like to test an approach as shown [here](https://wegstr.com/). They apply UV solder paste on the whole PCB and after curing they mill away the paste on the complete area of the pads (while trying not to touch the copper). I just managed to get the outline but not the "internal" area of the pads milled away (not yet on the router but just looking at the G-code with a viewer).

So is there an option existing for this that I didn'd see?
If not, any workaround or trick to achieve this?

I am not sure if this is the right place to ask. If not forgive me and indicated howto.

Thank you and best regards,
Andreas

[image]

eyal0, 2019-02-17:
This is the right place to ask.

pcb2gcode is for milling around shapes, not inside them.  So if the input were the opposite of the graphic above (black and white swapped) then it would work fine with a settings like:

`--mill-diameter=0.1mm --isolation-width=1000mm"

That would use a 0.1mm bi and make successive passes to remove material.  With 1000mm set, it would remove the maximum, all of it.

The issue is that the input gerber is of the opposite polarity.  I could fix this in code almost trivially, I think.

To proceed, how about I make a version of pcb2gcode that has a `--invert-gerbers` option which inverts the polarity of the front and back input gerbers?  Then, when you want to make a PCB, you can use it without the invert option for regular milling and with the invert option for the mask layers.

I could probably whip it up pretty quickly.  Do you have a sample gerber mask file that I could use for testing?  Like the one above, that would be fine.

chiefenne, 2019-02-17:
That sounds cool. So if the polarity can be changed in the input gerber file (I am using Eagle) the task can already be solved. I'll have a look if I can do that somehow.

I attached the example solder mask.

[soldermask_top.zip](https://github.com/pcb2gcode/pcb2gcode/files/2873425/soldermask_top.zip)

Thank you.

chiefenne, 2019-02-20:
Thanks a lot for implementing this. I will give it a try.

PR #292 (2019-02-20): Add invert-gerbers flag to invert polarity of gerber input files
This fixes #291
closed 2019-02-20 by eyal0