pcb2gcode/pcb2gcode #295: gcode generation failure after commit 02b6f811a54c for KiCAD-generated gerbers

plaes, 2019-02-21:
pcb2gcode version (run pcb2gcode --version to see this):
```
1.3.2
Git commit: v1.3.2-496-g02b6f81
Boost: 106501
Gerbv: 2.6.1
```

Current master is also broken. Checking out one commit before that works properly.

What did you try (include command-line arguments):
```
~/code/pcb2gcode/pcb2gcode --vectorial \
   --software linuxcnc --zero-start \
   --front *-F.Cu.g* --front-output front.ngc \
   --back *-B.Cu.g* --back-output back.ngc \
   --drill *.drl --drill-output drill.ngc --drill-side back \
   --outline *-Edge.Cuts.g* --outline-output cutout.ngc \
   --metric --metricoutput --noconfigfile \
   --zsafe 3 --zchange 50 \
   --cut-side back --cut-feed 500 --cut-speed 6000 --cut-infeed 0.6 --zcut -1.8 \
   --zbridges -1 --bridges 3 --bridgesnum 4 --cutter-diameter 2 \
   --mill-feed 500 --mill-speed 6000 --zwork -0.2 --offset 0.2 \
   --drill-feed 500 --drill-speed 6000 --zdrill -4 \
   --spindown-time 2 --spinup-time 2 \
   --tile-x 1 --tile-y 1
```

What happened:
Front.ngc and back.ngc are mostly empty. Also some `svg` files are broken. 
What did you expect to happen:
Proper gcode as it is from version before :)

Please find the files attached:
[olimex-v0-shield-bug.tar.gz](https://github.com/pcb2gcode/pcb2gcode/files/2889059/olimex-v0-shield-bug.tar.gz)

plaes, 2019-02-21:
Oops.. tarball contains the generated g-code and svg files from the revision before the bug. The `command.sh` in the tarball also contains the arguments required to run pcb2gcode. 

PS. It would be nice if the g-code and/or svg files also include the pcb2gcode --version information.

eyal0, 2019-02-21:
I'll look into it.  Sorry about that!

eyal0, 2019-02-21:
Okay, I see an error already when I run it:

`Found an unconnected loop while parsing a gerber file while expecting only loops`

This is because the Edge Cuts layer is almost but not quite making a loop.  If you open the Edge.Cuts file that you sent, search for the numbers "-87579200" and "-87579000".  The difference between those is 0.0002mm.  It was probably not your intention to have these cuts be separated like that!

This is the second time that I have seen this issue from KiCad, both in the last few weeks.  Something might be wrong with KiCad and they should know about it.

The other instance was here: https://github.com/pcb2gcode/pcb2gcode/issues/237#issuecomment-460399450

plaes, 2019-02-21:
OK, thanks for heads up. I fixed issue in the end coordinates of the cut lines and now it works properly.

Also, if this distance issue is so critical, as user I would expect the program to report an error (mentioning at least the file where this is coming from) and then exit.

eyal0, 2019-02-21:
I will make the warning more visible.  I will also have pcb2gcode correct these automatically.

Sorry about the trouble.  Good luck!

eyal0, 2019-02-21:
I'll reopen this issue and use it to track the better warnings and automatic fixing of KiCad gerbers.

eyal0, 2019-02-21:
Alright, I got it merged.  pcb2gcode can now process the files that you have provided even without edits.  Enjoy!

closed 2019-02-21 by eyal0