pcb2gcode/pcb2gcode #297: make autoleveller aware of already drilled holes

QuantumKittens, 2019-02-21:
There should be an option to not probe points where holes have already been drilled. One way to achieve this would be to check whether the probe point coincides with a hole and either skip it or slighly move it away from the hole.

There are a some cases where this is required with double sided PCBs:

- The back side is milled first and drilling is done from the back side. This allows to use the holes for alignment but limits autolevelling to one side.
- Through hole plating. The board surface needs to be conductive, so all holes need to be drilled before any kind of milling. This prevents using autolevelling altogether. (Which is a particular issue in this case because the plating process messes up the board flatness.)

eyal0, 2019-02-22:
Oh, that seems like it would be very complicated!  I admit, I have never done any two-sided boards.

The autolevelling routine knows nothing about the drilled holes.  It would be difficult to do.  Also, the gcode that does all the compensating assumes that the probe points are in a regular grid.

In what order do you usually expect to work.  My process was like this:

1. Autolevel
2. Mill the back
3. Paint with solder resist.
4. Drilling
5. Cutting.

Paint must come after milling, of course, but I do it before drilling so that the paint doesn't get into all the litlte holes.  Cutting comes last because then I can no longer align after the PCB disconnects from the FR4.

How do you do it for two sides?

eyal0, 2019-02-22:
The way that I described in the [wiki](https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Alignment-for-two-sided-PCB-and-offsets) is what I expected would work.  First attach the FR4.  Then drill the alignment holes *outside* the area of the PCB.  Because you're using the CNC to jog over to the holes, you know their exact location.  Take the average x coordinate and use that with the `--mirror-axis` option.  So if the alignment holes have an average X of 75mm, use that.  pcb2gcode will generate code that flips the board along the x=75mm line.  And since those holes aren't in the PCB's surface, autolevelling won't be affected.

QuantumKittens, 2019-03-03:
This is more or less what I'm using right now and it works okay for 'normal' double sided boards. I haven't yet found a way to do chemically plated boards larger than about 3x3cm though. The plating process appears to make the FR4 swell unevenly, so autoleveling ist a must.
I tried a larger test board where all the holes are aligned to not coincide with the autoleveling grid. This worked perfectly, but is mostly impractical with real boards.

jschneider1207, 2023-07-14:
This just bit me in the ass. I drill first as I dont want the drilling to potentially screw up thin traces. Haven't had a problem until now. My board has a 4.5mm hole I drilled into it and autolevel keeps failing as its kinda difficult to try and avoid a probe spot being right in the hole, tried a dozen bin sizes. I dont know how to fix this though. Maybe if you allow specific probe points to be toggled off by clicking on the intersections of the grid that is drawn?

eyal0, 2023-07-16:
The autoleveller just does an even grid over the board, it doesn't know about all the milling and drilling that is going to occur.  It would be pretty hard to fix this!  :-(

jschneider1207, 2023-07-16:
Thats why I was thinking maybe you could just click on one of the grid
intersections and “turn it off”, then the autoleveler would just skip over
that probe point. Idk if thats easier to do, just a thought I had.

eyal0, 2023-07-19:
The issue here is the math.

See, there are a few ways to do leveling (technically it's tramming but PCB people usually call it leveling).  One way is to do with with your software.  In this method, pcb2gcode isn't even involved.  You just tell it to check the level of the board and it will record the height at every location.  And then when you actually do your milling, the software that you are using will automatically adjust your drill.  For instance, the output from pcb2gcode might say to drill to a depth of 0.5 but your software is going to tweak it and tell it to drill to 0.56 or something like that.

The other way is to have pcb2gocde output gcode that includes leveling.  In that case, pcb2gcode is going to emit gcode that does leveling.  It's going to probe a grid of points and store the height of each one in a variable.  And then, each time it does milling, it will adjust the height to be a math equation based on the stored variables.

The math to find the height at each point is not too complex if you know the height of the grid points around that point.  However, if there is a grid point missing then the math gets more complex.  We could do it, but it's complex.

Here's the formula that we are already doing, for a rectangle: https://en.wikipedia.org/wiki/Bilinear_interpolation#Repeated_linear_interpolation

Now imagine that we need to do it for a grid with missing points.  That would be a trapezoid!  It would be a lot of work.

Honestly, I've never had great luck with leveling.  I found that the reason that my boards were skewed was because the boards didn't lie flush on the CNC.  And when I tried to compensate for that, the milling tool would just flex the board instead of milling it.  It's very important that the PCB has no flex to it when you mill so that the milling bit will have a contra force so that it can really dig in.  Once I got the PCB affixed better then the autoleveling didn't become so important.  I never did anything all that fine, though, so maybe for you it is more important.