pcb2gcode/pcb2gcode #302: Have gerberimporter return lines for thin lines

eyal0, 2019-03-01:
Currently, pcb2gcode converts the gerber files into a 2d surface of polygons. Milling is done around those shapes.

For outline layers, pcb2gcode can treat the drawn lines as 1D lines and not 2D shapes, and then join the lines together to make shapes. This is the default because the edge cuts layer is usually drawn as lines with thickness to be ignored.

Users may want to also use pcb2gcode to draw a the silk layer on to the PCB. The silk layer may be a bunch of thin lines that should be drawn as if zero thickness lines, but not joined together into shapes like with the edge cuts layer.

So add an option to return as lines and not as shapes any drawn lines. I think that we can do it for all lines no matter their thickness. If later we see that we only want certain lines to stay as lines and not shapes, that can be done.

c-, 2019-03-03:
For context (since I was the one who asked for this), FlatCam has a technique for [tracing the silkscreen layers in a Gerber](http://caram.cl/software/flatcam/tracing-the-silkscreen-with-flatcam/). I use(d) this approach to either engrave or pen plot the silk screen on the non-copper side of a PCB.

I've since migrated to pcb2gcode for my PCB milling needs (the workflow in FlatCam is far too tedious and repetitive), and for my silk I've hacked together a script to generate g-code for just those layers. However, this script isn't nearly as robust/complete as pcb2gcode at Gerber handling, the silk doesn't align perfectly with the pcb2gcode output (especially with multiple tiles), nor can it optimize the gcode output in any fashion. Etc.

eyal0, 2019-03-04:
Okay, I think that I've got it nearly done now.  For the record, here's the old way that you could have done that already works:

[image]

That's with these settings:

```
mill-diameters=0
milling-overlap=-0.2mm
invert-gerbers=true
extra-passes=50
```

I've set invert-gerbers because I don't want to mill *around* the shapes, I want to mill *inside* of them.  Inverting tricks pcb2gcode into think that it's milling a big shape with some "C1" shaped holes in it.  Then I set mill-diameters=0.  That means it'll mill with an offset of zero from the edges of those shapes, so it'll mill right around the edges.  Now, I want it to fill in the shapes so I need to tell it to do extra passes that overlap by **negative** 0.2mm.  Usually mills have a positive thickness and you overlap a little so that you get all the copper removed but here I'm telling it that there is **zero** diameter bit so I need instead a *gap* of 0.2mm, which is negative overlap.  Finally, I tell it to do 50 passes, so it'll do 50 passes each 0.2mm closer to the inside of the shape.  You can set this number as high as you like because once it whittles a shape down to nothing, it quits making more passes.

So that's what I end up with.  It very much resembles the input's shape but it's not what you want.  But it would pass.

Now I've added the new `--draw-gerber-lines` option.  When set, it'll treat shapes the same as usual but it will treat lines like you expect.  Here's the result:

[image]

Should look just like the original.  Also, compared to the [example](http://caram.cl/wp-content/uploads/2014/10/tracing_example3.png) that you provided, I think that the paths with pcb2gcode might be better.  pcb2gcode works to minimize travel time.  You can see how the dotted-lines are tiny.  I experimented with having the pen not even lift between lines that are really close to each other but it sometimes caused ugly added lines.  Let me know if you think that this would be something that I should look into further.

Regarding the alignment, I ran pcb2gcode twice, once with the B.Cu layer and appropriate settings for milling copper and then again with --draw-gerber-lines with the F.SilkS and those settings put the files together:

[image]

I think that it's perfectly aligned.  If you see that alignment doesn't work, let me know.  You should set an outline in both runs because the outline is used for alignment.  There are some options to mess with alignment, too.

Let me know how that looks and if anything is missing.  The code is here: https://github.com/pcb2gcode/pcb2gcode/pull/303

c-, 2019-03-04:
Oh, that looks perfect! I'll run some tests tonight, and I might even be able to mill a board tomorrow.

As far as the alignment goes, I don't see problems until I tile a board. --tile-x=4 usually shows maybe 1mm of drift towards the last time. I'm quite certain that's just my code script missing something subtle in the tile size calculation.

c-, 2019-03-17:
*Finally* had a chance to run it on a real board. Overall it's quite lovely.

The only glitch I can see is with tool changing. That is, it doesn't appear to pay any attention to --mill-diameters=0.4mm (pen width, in my case) and keeps prompting for a tool change for each of the tool widths in the input Gerber.

I'm not sure if I'm holding it wrong or there's a code path missing somewhere.

eyal0, 2019-03-18:
Glad to hear it!

The tool change is due to lines being drawn at different widths, like you noticed.  In fact, the mill-diameters option is ignored because it's only ever really used for knowing how far outside the shape to mill (the mill radius).

So I see three options:

1. **Leave it as it is**
    * *Pros:* Good if you want to switch pens.
    * *Cons:* Annoying if you only have one pen.
2. **Always condense all the different pen widths into one**
    * *Pros:* No annoying tool changes.
    * *Cons:* Problem if you want to have more than one pen.
3. **Obey mill-diameters if specified, rounding to nearest.  If no mill-diameters, do 1.**
    * *Pros:* Most flexible, solves @c- problem with current configuration.
    * *Cons:* More coding.  Not clear if the default should be 1 or 2 above.
4. **Something else?**

I'll think about it some.  Maybe 2 is good enough because no one will every have more than 1 pen?

If I want to do number 3, I would do it like the drill-diameters: You specify which pens you have and we round each line to the nearest diameter size.  And if you only specify 1 pen, all will use that same pen.  If you specify no pens, there is no rounding and we'll do all the tool changes.

eyal0, 2019-03-18:
I think that I like 3 best because it's the most fun to implement.  I'm open to being convinced of a different choice.

c-, 2019-03-18:
I'd be fine with option 2, but option 3 is functionally better.

I'd be more inclined to switch pen *colors* rather than pen widths (using a different Gerber/layer for each color), but if I'm engraving the silk (which I prefer where the substrate allows) then multiple diameters could be a time saver.

I wouldn't do anything like mill-drill by trying to build up thicker lines with multiple passes of a smaller diameter. I think nearest diameter is more than adequate.

c-, 2020-12-31:
> I experimented with having the pen not even lift between lines that are really close to each other but it sometimes caused ugly added lines.

As a data point, the backtracking optimization has "broken" this behaviour, and it's drawing the pen back over existing lines. On the other hand, backtracking can be disabled, so it's probably not worth pursuing. I'm not really noticing a serious quality problem using a marker, and the speedup is nice.

PR #303 (2019-03-04): Add option to mill lines that have no thickness
This fixes #302