pcb2gcode/pcb2gcode #317: PCB2Gcode crashes with std::bad_alloc

SmallestLeslie, 2019-06-16:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
2.0.0
Git commit: v2.0.0-102-g6b71101
Boost: 106900
Gerbv: 2.6.0

What did you try (include command-line arguments):
pcb2gcode --back=/home/laci/gyújtás/gyújtás-B.Cu.gbr --drill=/home/laci/gyújtás/gyújtás.drl --outline=/home/laci/gyújtás/gyújtás-Edge.Cuts.gbr --output-dir=/home/laci/gyújtás/kimenet --metric=true --metricoutput=true --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=true --zchange=15.0000 --zchange-absolute=false --zero-start=true --zsafe=2.0000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=0.2500 --voronoi=false --zwork=-0.0150 --drill-feed=500 --drill-side=back --drill-speed=10000 --milldrill=false --nog81=false --nog91-1=false --onedrill=false --zdrill=-1.8000 --zcut=-1.6 --cutter-diameter=3.0 --cut-feed=450 --cut-speed=10000 --cut-infeed=0.8 --al-back=true --software=linuxcnc --al-x=20.0 --al-y=20.0 --al-probefeed=100

What happenned:
With autolevel option enabled PCB2Gcode crashes with throwing:
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

The computer has 4 GiB RAM installed (3 GiB available as it runs a 32b OS). It has 1.4GiB free at the time of issuing the command to start PCB2Gcode.

An older version of PCB2Gcode from February worked on this system.

What did you expect to happen:
?

[GH.tar.gz](https://github.com/pcb2gcode/pcb2gcode/files/3294145/GH.tar.gz)

eyal0, 2019-06-20:
I tried it on my computer and it worked fine.

Can you try to run this with `catchsegv ` in front of the command?  It will output a backtrace of the error that could help me narrow down the cause.

SmallestLeslie, 2019-06-20:
It works now. I don't understand. I copied the command from the terminal when it did not work and now back and it works now.
I am sorry.

SmallestLeslie, 2019-06-20:
I simply forgot that I downgraded PCB2Gcode last time and it seemed that it works.
Now I upgraded again and tried `catchsegv` but it did not return anything more than last time.
Or maybe I used it a wrong way?

catchsegv pcb2gcode --back=/home/laci/gyújtás/gyújtás-B.Cu.gbr --drill=/home/laci/gyújtás/gyújtás.drl --outline=/home/laci/gyújtás/gyújtás-Edge.Cuts.gbr --output-dir=/home/laci/gyújtás/kimenet --metric=true --metricoutput=true --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=true --zchange=15.0000 --zchange-absolute=false --zero-start=true --zsafe=2.0000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=0.2500 --voronoi=false --zwork=-0.0150 --drill-feed=500 --drill-side=back --drill-speed=10000 --milldrill=false --nog81=false --nog91-1=false --onedrill=false --zdrill=-1.8000 --zcut=-1.6 --cutter-diameter=3.0 --cut-feed=450 --cut-speed=10000 --cut-infeed=0.8 --al-back=true --software=linuxcnc --al-x=20.0 --al-y=20.0 --al-probefeed=100

The result was:
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted

eyal0, 2019-06-23:
It is hard for me to know where the error is happening!  The only thing that I can suggest is to use git bisect between the working version and the broken one.  I have a hard time debugging this because I am unable to see the problem on my own computer...

If you want to try git bisect and need help with it, I can let you know how to use it.

I see that usually the first thing printed is `Importing front side` but it is not printing that.  So the error must occur before then.

eyal0, 2019-06-23:
Maybe it has to do with the filenames, that have accents and non-ASCII characters?  It worked fine on my computer with accents but you could try renaming folders and see if it makes a difference.

You could also try `valgrind`.  `sudo apt-get install valgrind` to install it and then run the same command with `valgrind` instead of `catchsegv`.  Maybe it will find a memory error?

SmallestLeslie, 2019-07-09:
Thank you for your help.
Sorry, I did not have much time lately, I have not tried valgrind yet.
The PCB design is just an one side design. The front side is just for a few jumper wires to be installed by hand. So I did not imported the front into PCB2Gcode.
The crash occurs during PCB2Gcode working on the back side.

eyal0, 2019-11-13:
Still waiting on this.  Try removing the non-ASCII characters and replacing them with ASCII and try again?  Re-open if you want to continue work here.

SmallestLeslie, 2019-11-16:
Sorry for the delay.
Using all ascii compatible characters did not help.
With valgrind it generates the gcode without error even with non-ascii filenames also.

SmallestLeslie, 2019-11-16:
Is it worth to mention that during making it gives warnings as the following?

  CXX      autoleveller.o
  CXX      board.o
In file included from gerberimporter.hpp:32:0,
                 from surface_vectorial.hpp:38,
                 from board.hpp:35,
                 from board.cpp:21:
/usr/include/gerbv-2.6.0/gerbv.h:99:27: warning: ISO C does not permit named variadic macros [-Wvariadic-macros]
 #define GERB_FATAL_ERROR(t...) g_log(NULL, G_LOG_LEVEL_ERROR, ##t);
                           ^
/usr/include/gerbv-2.6.0/gerbv.h:100:29: warning: ISO C does not permit named variadic macros [-Wvariadic-macros]
 #define GERB_COMPILE_ERROR(t...)  g_log(NULL, G_LOG_LEVEL_CRITICAL, ##t);
                             ^
/usr/include/gerbv-2.6.0/gerbv.h:101:31: warning: ISO C does not permit named variadic macros [-Wvariadic-macros]
 #define GERB_COMPILE_WARNING(t...)  g_log(NULL, G_LOG_LEVEL_WARNING, ##t);
                               ^
/usr/include/gerbv-2.6.0/gerbv.h:102:23: warning: ISO C does not permit named variadic macros [-Wvariadic-macros]
 #define GERB_MESSAGE(t...)  g_log(NULL, G_LOG_LEVEL_MESSAGE, ##t);
                       ^
  CXX      common.o

SmallestLeslie, 2019-11-16:
I tried to bisecting the problematic commit but I could not get bisect work.
No matter how I tried to specify the bad commit it told me:
fatal: Needed a single revision
Bad rev input:

I tried to specify with the following ways:
git bisect bad 76ff1a307841d5c25a4ef38db8895c2069857373
git bisect bad 76ff1a3
git bisect bad v2.0.0-102
git bisect bad v2.0.0-102-g6b71101

None of the above worked.
What did I wrong?

eyal0, 2019-11-22:
The warnings I know about.  You can ignore them.

Looks like you are doing git bisect correctly.  Maybe the versions that you are mentioning do not exist on your computer?  You can run `git pull` to get the versions that you need.  Make sure that the version that you specify exists on your computer.  This works for me:

```git bisect bad 76ff1a30```

SmallestLeslie, 2020-03-01:
After a lot of struggle, I got the following conclusion:

8b5c2d38152cd3eaebdcc29a3d622f98534ce04c is the first bad commit
commit 8b5c2d38152cd3eaebdcc29a3d622f98534ce04c
Author: Eyal Soha <eyalsoha@gmail.com>
Date:   Mon Mar 4 22:07:50 2019 -0500

    partition_segment for better autoleveller segments and unit tests

:100644 100644 3fd1f0e01fa3b0fa28f0a80e763865de8d51659b 39230e25451b979b0588d6b99ebca5ce8a0635f9 M	Makefile.am
:100644 100644 e4454630b3aec3b1371c2fca95481e75dbd4aa2a b29a7dc0a6b7c4cc2c97ed1f29f2bbbffd982060 M	autoleveller.cpp
:100644 100644 5f790ba0f39422dd62e730296125832880ec35ae f95e6027dfe7272ca0b806d00e7d33619af76993 M	autoleveller.hpp
:000000 100644 0000000000000000000000000000000000000000 6872c79420f9154c0bf59f553c96761dd4fb2693 A	autoleveller_tests.cpp
:040000 040000 436cda8f3b5fd2703344e7b24a565a5b214c933d c38757d4c5f2f51945e08e23d9e2ff738ead6b72 M	testing

The last known good version is: v2.0.0-82-g1d7f51c
The first known bad is: v2.0.0-86-g734d06e
Between this two versions there was a bisecting try where there was no change in the version.
The message I got was:
Bisecting: 1 revision left to test after this (roughly 1 step)
[8b5c2d38152cd3eaebdcc29a3d622f98534ce04c] partition_segment for better autoleveller segments and unit tests
Then:
make[2]: Nothing to be done for `install-data-am'.
I guess it was v2.0.0-84.

I hope this helps some.

eyal0, 2020-03-09:
Interesting.  Okay, so the problem is with the new autoleveller code.  I wrote nicer code for figuring out the autoleveller points.  std:alloc happens when we try to allocate and we fail.  Maybe because the allocation is too big?  But it isn't failing for me and I can't figure out where the allocation would fail.

If there were a print statement in the beginning of `partition_segment` and then at the end of that function, maybe that would be useful.  I would be interested to see how many times line 371 that adds a new point on the end, to see how many times that happens.  Maybe there's where all the memory is getting used up by doing too many of those?

eyal0, 2021-01-10:
I don't know how to fix this and I may never know.  I hope that it just went away.

closed 2021-01-10 by eyal0