pcb2gcode/pcb2gcode #318: cutting of non outlines breaks copper milling

johnqube, 2019-06-23:
Hello

running 1.3.2
I am facing something maybe related or duplicate of #7 and #237 
I am trying to cut shapes inside a PCB without cutting the PCB outline (the shape of my PCB is already at the correct dimensions, no need to cut the outlines)
when adding the outline layer, this breaks the milling of the copper layers
it looks like the computation of the outlines makes a very small bounding box cropping all the other layers.

you may look a the produced files attached
[plot.zip](https://github.com/pcb2gcode/pcb2gcode/files/3317776/plot.zip)

thank you

eyal0, 2019-06-24:
Please send the command-line that you used to run this.  Thanks.

eyal0, 2019-06-24:
Also, 1.3.2 is very old.  Can you run a newer version?  Please output your version number by typing `pcb2gcode --version`.

eyal0, 2021-01-10:
Long time without a comment and the old code was old.

closed 2021-01-10 by eyal0