pcb2gcode/pcb2gcode #322: zchange-absolute=true is ignored

AlexeyGusev, 2019-08-13:
$ pcb2gcode --version
2.0.0
Git commit: v2.0.0-102-g6b71101
Boost: 106700
Gerbv: 2.7.0

with these options:
```
zchange-absolute=true
mill-diameters=1.5mm,0.8mm,0.6mm,0.1mm
zchange=-1mm
```
pcb2gcode generates g-code for front/back milling that doesn't always use G53 to retract Z axis to safe position. As a result, instead of being retracted to topmost position - 1mm, the tool is inserted 1 mm (placed 1 mm below surface) into pcb. 

generated g-code example that demonstrates the problem:
```
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G53 G00 Z-1.000 ( retract )

M5 ( Spindle off. )
G04 P1.000000

G00 Z-1.00000 (Retract to tool change height)
T1
M5      (Spindle stop.)
G04 P1.00000 (Wait for spindle to stop)
(MSG, Change tool bit to mill diameter 0.80000mm)
M6      (Tool change.)
M0      (Temporary machine stop.)
M3 ( Spindle on clockwise. )
G04 P1.00000 (Wait for spindle to get up to speed)
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
```
the first Z-1.000 line is correct and the second one (has more zeros after comma) will break the router bit.

eyal0, 2019-08-14:
If you set zchange to be -1, then the tool change height will be -1.  If you want higher, you need to set zchange to be, like 10mm or something, right?

I am not an expert at `G53`.  Can you send me an example of which line is bad and what you expect it to be?  Did you require that G53 be included on the line that says `Retract to tool change height`?

eyal0, 2019-08-14:
Is this the same as #315?

AlexeyGusev, 2019-08-14:
yes, this is basically the same as #315, but I would say that mill files actually include G53 for tool change, but not always.

If you say `zchange-absolute=true`, you want the tool change to happen in machine coordinates, right? So when `zchange=-1mm`, Z axis should be retracted to machine Z axis zero, which is Z topmost position, minus 1 mm. You can do that with `G53 G00 Z-1.000`. And this is what the code snippet above does, on line two.

On line five, however, the code does the same move in work coordinates: `G00 Z-1.00000`, and in work coordinates Z axis zero is pcb surface! So you actually try to punch the material that you work on by moving Z axis 1 mm below surface in order to change tools.

Why the G00 Z-1 command is repeated twice, once with G53 and once without, is a question on it's own:
```
G53 G00 Z-1.000 ( retract )

M5 ( Spindle off. )
G04 P1.000000

G00 Z-1.00000 (Retract to tool change height)
```
I see no obvious reason to do so - one command with G53 is enough. Anyway, G00 Z-1 without G53 is dangerous, especially if your zchange parameter is -2 or -5mm.

AlexeyGusev, 2019-08-14:
Speaking in terms of machine coordinates (G53):

> If you set zchange to be -1, then the tool change height will be -1. 

Yes, that is 1 mm below topmost position, because Z=0 is the upper physical limit of Z axis.

> If you want higher, you need to set zchange to be, like 10mm or something, right?

No, higher than -1 is only 0. These numbers cannot be positive, at least in a common case scenario.

zchange can, however, be positive, like 10mm or similar, if you use work coordinates instead of machine coordinates. In this case zchange=10mm would mean 10mm above pcb surface, because pcb surface is 0 in work coordinates.

eyal0, 2019-11-13:
I think that this is fixed now.

closed 2019-11-13 by eyal0