pcb2gcode/pcb2gcode #328: FR: separate zsafe for drilling

QuantumKittens, 2019-09-16:
It would be helpful if there is an option to specify a separate zsafe for drilling (and for completeness also for outline milling).
Drilling involves tool changes and I've ruined a few boards because a drill was sticking out too far and then milled away the tracks during a rapid move. One could just set zsafe higher, but this would slow down the isolation milling considerably as (in my case) the vertical feed rate for milling is much lower than for drilling.

eyal0, 2019-09-29:
How come the CNC isn't compensating for the difference in length of the tools.

Between each tool change, you should be running a tool change procedure that sets your machine's tool offset. pcb2gcode supports this and puts tool change procedures into the output

When the offset it set correctly, the zero height of the tool is always right at the height of the PCB surface, no matter the length of the tool. So then, the same zsafe is correct for all work.

How is your procedure different?

eyal0, 2019-11-07:
I think that you should be using the tool change offset (TCO) for compensating for the different lengths of tools instead of trying to modify zsafe.  zsafe should simply be the height at which all horizontal moves are safe and wont' crash into something.

closed 2019-11-07 by eyal0

QuantumKittens, 2019-11-10:
Sorry, I somehow missed you reply.
I dont actually use any tool length compensation as the PCB drill bits have rings to set the height, so its possible to just touch off the the tip of the first drill to the PCB surface. The issue is that a manual tool change with a collet may be sometimes off by a bit. With a larger zsafe (i.e. 3mm) this isn't a problem, but using the same zsafe for milling increases milling time due to the much lower feed rate for z movements. 
For now, I just run pcb2gcode twice, with different values for zsafe (3mm for drilling/routing, 0.75mm for milling).

eyal0, 2019-11-11:
Your CNC doesn't have a tool change measure?  The CNC that I used has a little button for tool change.  When the tool change command comes up, a canned routine moves the bit into a certain position and then slowly lowers the tool until it depressed a button.  Then the machine sets the TCO based on when it detected the button getting pressed.  It work pretty great.

If you don't have that built in to your machine, it might be worth making it.  Mostly software should support it already so you'd just have to put a button in and wire it to the control board.