pcb2gcode/pcb2gcode #329: pcb2gcode v2.0.0 ignores a track in the gerber file

jboecker, 2019-09-20:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
````
2.0.0
Git commit: v2.0.0-119-gf5ee31c
Boost: 106200
Gerbv: 2.6.1
````

What did you try (**include command-line arguments**):
I designed a PCB in KiCad 5.1.4 on Windows and exported the gerber files, then ran the `pcb2gcode` command (all parameters were supplied using the attached millfile).

What happened:
One of the tracks was not included in pcb2gcode output (look at the track highlighted in gerbv in the following screenshot):
[image]

What did you expect to happen:
All tracks should be imported.

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

The millfile, gerber files and the resulting g-code and SVG files are included in the following ZIP file:

[pcb2gcode-ignored-track-input-and-output.zip](https://github.com/pcb2gcode/pcb2gcode/files/3636755/pcb2gcode-ignored-track-input-and-output.zip)

Notes:
* I first compiled a 32-bit version of pcb2gcode v2.0.0 using MSYS2 on Windows and ran into this issue. Then I compiled a 64-bit Linux executable using the Windows Subsystem for Linux running Debian 9.6 and reproduced the issue. The version info and files attached to this bug report are from the Linux version.
* pcb2gcode v1.3.2 correctly processes the input file (all tracks are included). Unfortunately, in the v1.3.2 output the drill.ngc and back.ngc files are out of alignment, which is a bit annoying to work around so I want to switch to v2.0.0, which fixes that bug.

Thanks a lot for this amazing piece of software! After the first-time set up of the millfile, the process to go from a KiCad layout to a finished prototype board is quick, easy and repeatable.

eyal0, 2019-09-24:
That's a nice bug report!  I appreciate that you've provided everything that I need in order to debug this.

I'm able to see, yes, that trace is missing when I run pcb2gcode, too.  I'll look into it.

eyal0, 2019-09-24:
The first step is to trim down the example so that I can debug it more easily.  Here's a much smaller obu-pcb-B_Cu.gbr that causes a similar mismatch between the gerbv output and the svg file.  (Just the lines from `%MOMM*%` onward:

```
%MOMM*%
%LPD*%
G04 APERTURE LIST*
%ADD17C,0.400000*%
G04 APERTURE END LIST*
D17*
X111760000Y-60960000D02*
X111760000Y-61510000D01*
X111760000Y-60960000D02*
X111760000Y-62560000D01*
X111760000Y-60960000D02*
X111760000Y-66294000D01*
M02*
```

Just three short lines that all start at the same point and extend various lengths in the same direction.

eyal0, 2019-09-24:
I have marked the place where you might have drawn two overlapping traces in KiCad.

[image]

eyal0, 2019-09-25:
Thanks for your kind words earlier.  Glad you enjoy `pcb2gcode`!

closed 2019-09-25 by eyal0

jboecker, 2019-10-04:
Hi, thanks a lot for the quick fix!

You are right, at the place you marked there are three tracks overlapping in KiCad like this:

[image]

Interestingly, two overlapping tracks happen naturally in other places and were already handled fine before the bugfix (possibly because one of them is entirely covered by the pad, so is either not exported by KiCad or ignored by pcb2gcode):

[image]

Sorry that it took me a while to get back to you, I spent the last few days tearing my hair out with another board that came off the mill with way too thin traces. Tried new endmills, going back to old pcb2gcode versions, etc. only to find out that I had specified the tracks that thin in my board layout in the first place... so now I can report that pcb2gcode v2.0.0-119-gf5ee31c works flawlessly for me.

eyal0, 2019-10-29:
Awesome news!  Nice to hear about success.

I already had overlapping tracks handled in many cases.  The only problems were when the overlapping tracks started or ended at the same spot.  Other types of overlap worked fine, so it's probably not related to the pad.

Anyway, `pcb2gcode` tried to be as generous as possible with gerber files so even if they aren't perfect, it tries to guess the intention and do it right.

PR #330 (2019-09-25): Fix overlapping tracks in gerberimporter
This fixes #329