pcb2gcode/pcb2gcode #331: Mill whole trace (like paint on flatcam)

dzervas, 2019-10-09:
I want to use a pen plotter to paint the traces with a marker and then etch the copper, but I can't find a way to paint the whole trace and not just its outline. Am I missing something?

eyal0, 2019-10-11:
Sure, you want the [`invert-gerber`](https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Milling#--invert-gerbers) option.

The output looks like this: https://github.com/pcb2gcode/pcb2gcode/blob/bd545f36c2e94dd9dcce0c7752265a72b9ee7f6f/testing/gerbv_example/invert_gerbers_fill/expected/traced_front.svg

Will that work?  The options that you need to fiddle with:

```
invert-gerbers=true  # So that we fill in the middle.
mill-diameters=0.5mm  # This is how wide a mark your marker makes.
milling-overlap=10%  # This is how much overlap you want on those marker lines.
isolation-width=2000mm  # This will fill in up to 2 meters in width, which is plenty for all your traces.
```

Try that, look at the generated gcode and .svg file, and let me know if that'll work.

dzervas, 2019-10-12:
I can't believe that I missed that... Thank you for your time!

closed 2019-10-12 by dzervas