pcb2gcode/pcb2gcode #333: Pen polotting (inverted milling) results in traces twice wide

dzervas, 2019-10-13:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
```
2.0.0
Git commit: v2.0.0-125-gdcefc52
Boost: 106700
Gerbv: 2.7.0
```

What did you try (**include command-line arguments**):
`./pcb2gcode --back ~/Lab/grafos/export/grafos-F_Cu.gbr --output-dir ~/Lab/grafos/export/` and then plot with pen using bCNC
All traces are 0.35mm wide with 0.2mm clearance (That's what KiCad says)
My pen is 0.3mm diameter

What happenned:
All traces where ~0.7mm

What did you expect to happen:
All the traces to be ~0.35mm (+- 0.1mm at worst)

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**
[millproject.txt](https://github.com/pcb2gcode/pcb2gcode/files/3722486/millproject.txt)
[grafos-F_Cu.gbr.txt](https://github.com/pcb2gcode/pcb2gcode/files/3722489/grafos-F_Cu.gbr.txt)

eyal0, 2019-10-14:
Okay, I have done a zoom and screen capture of the output svg file.

[image]

The original is on the left.  That trace is 0.35mm wide.  On the right is the path in the same place: two thin black lines.  I added to those lines 4 magenta colored lines, each one is 1.5mm.  Notice that the bottom-most magenta line does not reach the top of the trace.  That's why the second path is also needed, so that it will add enough width.

I checked in bCNC and I see that the pairs of paths as 0.05 mm apart.  That seems right.

So I think that pcb2gcode is doing the right thing.  Can you give an explanation of what you expected?  I think that pcb2gcode can do what you want if you explain to me the path that you want the pen to travel.

dzervas, 2019-10-14:
Wow, fantastic debugging, thank you for your time!

Then it’s probably the pen (ink spreading wide to 0.7mm). Can I make it do 1 pass?

eyal0, 2019-10-14:
Okay, I think that I understand.  You ran it and it turned out that your pen is making marks that are actually wider than the trace itself.  Okay, no problem.  You need to put the true width of the mark of your pen into the `millproject` file.  One way is for you to use the CNC to draw a mark with the pen and then measure that.  If you find that the mark is 0.7mm, then you can fix this line in `millproject` from:
```
mill-diameters=0.3
```

to:

```
mill-diameters=0.7mm
```

That will let pcb2gcode know the width of your pen, so that it can make an accurate output.  (I put the `mm` in there because you want millimeters but you could leave it off because you already set metric=1, so millimeters is the default.  Up to you.)

closed 2019-10-16 by dzervas