pcb2gcode/pcb2gcode #336: Possibly an issue with me - not with pcb2gcode

Soldering-Stan, 2019-10-20:
Is there any way of milling isolation slots using pcb2gcode ?  I am using kicad and can produce the plots etc.  I can cut the borads out no problem but can't get isolation slots to cut.  I have tried moving the slots to one o f the user layers and I can get the slots cut BUT I can't find any way of getting them cut without the outline.
If I have a 0.08" cutter and want an 0.08" width slot, how do I do it.  All my attempts at various cutter-dia, vector settings, fill-outline and outline just seem to run the cutter path round the outside edge.
Any ideas ?

Soldering-Stan, 2019-10-21:
I've worked round the problem.
Create a graphic line of the path required for the cut (Use one of the User layers - e.g Eco2.User below).  Make the line 0.001" wide.  Then "pcb2gcode --outline-output isolation.ngc --outline projectName-Eco2.User.gbr --cutter-diameter 0. --zcut -0.08 --zsafe 0.04 --zchange 1.6 --cut-feed 15 --cut-speed 24000 --cut-infeed 0.04 --cut-side back --fill-outline=true"
Hope this helps somebody :-)

S

eyal0, 2019-10-21:
It ought to be easier than that! Can you send me the Gerber file as an example?

eyal0, 2019-10-22:
There are a few ways that slots can work.  One is for the slot to be in the drill file as a series of holes.  The other way is for the slot to be drawn on the Edge Cuts layer.  `pcb2gcode` supports both.  If you can send a gerber file and the settings that you tried, I can take a look.

Soldering-Stan, 2019-10-22:
Hi,
I have attached three gerber files in a zip archive  (it wouldn't let me upload a tar file).
[dsr.zip](https://github.com/pcb2gcode/pcb2gcode/files/3757180/dsr.zip)
 Edge.Cuts.gbr is tha standard edge cuts layer.  This I have just measured the board and it is oversize so it looks as if I need to do some more checks, I thought that --fill-outline would have made it the correct size...  (it should have been 2.5 x 2.25) pcb2gcode --outline-output 05_outline.ngc --spinup-time 10 --spindown-time 0 --outline Edge.Cuts.gbr --cutter-diameter 0.125 --zcut -${DEPTH} --zsafe 0.04 --zchange 1.6 --cut-feed 15 --cut-speed 24000 --cut-infeed 0.04 --cut-side back --bridges 0.02 --optimise --bridgesnum 4 --fill-outline=true

I have also attached eco1-08.gbr which has an 0.08 trace and eco2-001.gbr which uses a 0.001" trace.  I have also attached the script I used to generate the files from the Kicad plots.

Also FYI, I have attached a bit of perl script that I wrote for splitting the drl file into one file per drill.  LinuxCNC is fiddly to change the drills without terminating the program :-)
Regards,

D

eyal0, 2019-10-26:
I took a look at the files.  What I see is this:

There's an Edge.Cuts.gbr file.  That one looks fine.  It is a square that isn't filled in.  That is the normal way to make an outline in KiCad.  If you give something like that to pcb2gcode or to a fab, they know that what you want is a PCB that cuts *around* that.  You want a filled-in rectangle for edge cuts.  You probably don't want your PCB to look like a picture frame, which is a really thin rectangle border that isn't filled in.

For traces, pcb2gcode and the fab do something different.  If you draw a rectangle that isn't filled in, what you get is a copper trace shaped like a picture frame with nothing in it.  If you wanted a filled-in rectangle, you'd have to fill it in on a copper layer.

Your Edge.Cuts.gbr file is a non-filled rectangle, so it'll work fine.  But your slot *is* filled-in.  So it's confusing to pcb2gcode.  pcb2gcode is assuming that "slot" is a line to trace around, like with the edge cuts file.  But you want it all removed.

I can try to make pcb2gcode more forgiving and treat lines that don't make loops as if they are slots.  That might solve your problem and then you'd be able to put the slot right into the edge cuts file.  Ideally, you can process all your gerber files with one run of pcb2gcode.  This is actually the best way because if you do multiple runs, pcb2gcode is going to need to figure out how to align all the different runs.  It doesn't necessarily know that each run needs to use the same origin.

I hope that all made sense!  Let me know how you'd *prefer* for things to work and we'll see if we can accommodate!

Soldering-Stan, 2019-10-28:
Hi Eyal,

Thanks for the reply.

That describes what I thought it was doing - following the outer edge of a wide cutout.

From a brain hurting point of view, it seems much harder to work out where the outline is and follow it rather than just follow the cutter path.  Respect for being able to determine closed edges as well :-)

As for preferences, I'm not an expert on this sort of thing so my preference for how it would need to work would probably not be the way most people would expect it to work.  Certainly what you propose makes sense to me and would be how I would expect it to work.

The board has been sent to fabrication now and they coped OK with the slot, it's only a problem with my cnc milling which I can work round easily enough (now I understand what it is doing).

KiCad allows me to set an origin and pcb2gode seems to honour that so simply putting a the slot in the edge cuts layer fro the fab house and having the same slot but only 0.001" wide in a custom layer works fine for me.  I don't include the custom layer in the files sent to the fab house.

Regards,

D

eyal0, 2019-11-04:
Is it okay if I put your Edge Cuts and the eco1 file into one of the test cases?  It would become one of the directories here: https://github.com/pcb2gcode/pcb2gcode/tree/master/testing/gerbv_example

Soldering-Stan, 2019-11-05:
Hi Eyal
Certainly, please feel free to use them.  Is there anything more required ?
You can also add the perl program for splitting the drill files if you think it will help people :-)
Regards,

D

commit 41968f8a (2019-11-07): Merge pull request #361 from eyal0/easier_slots

Make easier slots

This fixes #336
closed 2019-11-07 by eyal0