pcb2gcode/pcb2gcode #337: Internal cutouts mess with bridges

c-, 2019-10-26:
#336 reminded me of a weird glitch I ran into using cutouts on the Edge Cuts layer.

2.0.0
Git commit: v2.0.0-125-gdcefc52
Boost: 106501
Gerbv: 2.6.1

pcb2gcode --mirror-absolute --metric=true --metricoutput --zchange 10 --nog81 --nog64 --optimise --tsp-2opt --vectorial=true --eulerian-paths=true --tile-x 1 --tile-y 1 --mill-diameters 0.64,0.24 --isolation-width 0.9mm --tolerance 0.01 --zsafe 0.5 --milling-overlap 25% --outline Edge.Cuts.gbr --cut-side back --zcut -2 --cut-infeed 0.7 --cut-feed 25 --cut-speed 12000 --cutter-diameter 1.2 --bridgesnum 2 --bridges 0.5 --zbridges -0.3 --outline-output Edge.Cuts-back-1.2mm.nc --back B.Cu.gbr --back-output B.Cu-back-0.1mm.nc --zwork -0.125 --mill-feed 75 --mill-vertfeed 25 --mill-speed 12000

Bridges on the external cutout aren't aligned correctly. Two bridges should be opposite of each other on the perimeter, and they obviously aren't. I'm assuming that pcb2gcode is doing something like including the internal cutouts when calculating perimeter length and then placing bridges accordingly rather than just the external perimeter. I'm also not sure what would happen if I made the internal ring larger and added more bridges; I wonder it it might place a bridge internally as well (which actually might be nice).

[image]

[Edge.Cuts.gbr.txt](https://github.com/pcb2gcode/pcb2gcode/files/3774913/Edge.Cuts.gbr.txt)

eyal0, 2019-10-26:
pcb2gcode selects the largest segments for making bridges.  Because your outline is a circle, it's probably a whole bunch of segments that are almost identical in line, making a polygon that approximates a circle.

So it picked the biggest two and even though they are all pretty much the same size, maybe those two were a hair longer or maybe it was a multi-way tie and it just picked two at random.

It would be better to pick in a different way, I agree.  I'm not sure what the best algorithm for this would be!  It should be something that is general enough to work on any outline, whatever the shape.  Ideas?  Maybe I could add a requirement that the bridges are as far apart as possible from each other?

c-, 2019-10-26:
That explains it; I thought it was entirely a side-effect of the cutout.

An ideal circle is made up of same-length segments, so there's never just two "largest". In theory. That the bridges aren't right next to each other is likely due to some "slop" in calculating the segment lengths (vectorial, tolerance, etc). I'm guessing a perfect square could show the same effect... Hm. Yep:

[image]

Yeah, if all the segments are the same length then pcb2gcode needs a secondary rule to spread the bridge locations. Evenly spaced on the outline around the centroid of the board (i.e. if bridges=*n*, then intersect with the outline every 360/*n* degrees) would be another solution, although some care might be needed for non-convex outlines (maybe that's where "furthest" adds additional filtering).

eyal0, 2019-10-28:
I think that you're exactly right.  The circle that you use was like a bunch of segments that were almost identical in size and some rounding caused those two segments to be picked.

One thought that I had is to just take all the segments that are at least long enough to put a bridge. 
 Then find the subset of those that are furthest apart from each other and use that.  That would solve the above two problems, right?

c-, 2019-10-29:
It would solve that specific problem, but will "furthest apart" be a manageable constraint when $bridgesnum>2 ?

That's why I'd lean towards a strategy of dividing the board into $bridgesnum quadrants and then finding the bridge in each quadrant. Other constraints could be added on top of that... furthest from the board center, furthest from the edges of the quadrant, furthest from previously placed bridges, etc, which would progressively put the bridges closer to opposite locations, but still give enough support to more pathological shapes (consider a boomerang-shaped board, for example).

eyal0, 2019-10-29:
I'm trying to figure out an algorithm that will work well enough without being too complicated to implement.  "Furthest apart" is pretty easy to implement but the run time could be long.  The good news is that I could probably just run it for a short time and get a decent answer that is good enough.

The suggestion that you offer might work, too, except that it would be complicated to do.  Dividing the edge into sections, sure, easy.  Then then I need to add constraints on top of that and it's not clear how to put all those together.  What if furthest from center is different from furthest from edge of quadrant, which do I pick?  How do I weigh all those.

The good news is that for most board, there aren't that many edges and the shapes are usually pretty regular.  So I think that I'll just stick with the furthest apart strategy.  If it turns out that it's not working well in practice then send me the practical example of the problem and we'll go from there.

c-, 2019-10-29:
My gut feeling is that dividing into sections would handle 99% of boards (i.e. regular shapes like rectangles and circles), and the other ideas wouldn't be necessary unless someone got really weird. But furthest apart should solve my problem with that particular shape, so I can't complain either way.

eyal0, 2019-10-30:
Here's a round PCB that has a round hole in the center.  Below I have images of the path both before and after #347 .  I think that it looks pretty good!  You can see how the bridges used to be placed kind of strangely and now they're are spaced very evenly around the edges.

[image]
[image]

The algorithm runs fast, so fast that I can't hardly notice any delay.  What do you think about this?

c-, 2019-10-30:
I like it. I'm sure it's possible to find pathological shapes that'll mess with it, but it's 100% better than what it was doing before.

closed 2019-11-04 by eyal0

c-, 2019-11-04:
I ran into this glitch yesterday with another design... I guess bridgesnum=2 and more complex shapes is living dangerously:

[image]

It looks good now:

[image]

eyal0, 2019-11-04:
Great!  I added testing examples with round PCBs using 3, 4, or 5 bridges and the bridges get spread around really nicely.  The algorithm runs so fast that it isn't noticeable and the results seem solid.

In your example, notice how those two spots are the furthest apart that you could possibly place bridges.  Really nice result on this one and lots of new testing added to the code along the way.

c-, 2019-11-04:
Ironically, this is actually one of those situations where placing the bridge at the furthest points can actually be a problem. Or maybe I should just take more care when removing the board... ;)

[image]

commit c0d6ac29 (2019-11-04): Merge pull request #347 from eyal0/better_bridges

Use a better bridges algorithm

This fixes #337