pcb2gcode/pcb2gcode #362: compile problems with gerbv 2.6.1

c-, 2019-11-09:
Ubuntu 18.04 LTS ships with gerbv 2.6.1, so I doubt I'd be the only person to run into this.

```
In file included from gerberimporter.hpp:31:0,
                 from gerberimporter.cpp:54:
/usr/include/gerbv-2.6.1/gerbv.h:99:27: error: ISO C++ does not permit named variadic macros [-Werror=variadic-macros]
 #define GERB_FATAL_ERROR(t...) g_log(NULL, G_LOG_LEVEL_ERROR, ##t);
                           ^~~
/usr/include/gerbv-2.6.1/gerbv.h:100:29: error: ISO C++ does not permit named variadic macros [-Werror=variadic-macros]
 #define GERB_COMPILE_ERROR(t...)  g_log(NULL, G_LOG_LEVEL_CRITICAL, ##t);
                             ^~~
/usr/include/gerbv-2.6.1/gerbv.h:101:31: error: ISO C++ does not permit named variadic macros [-Werror=variadic-macros]
 #define GERB_COMPILE_WARNING(t...)  g_log(NULL, G_LOG_LEVEL_WARNING, ##t);
                               ^~~
/usr/include/gerbv-2.6.1/gerbv.h:102:23: error: ISO C++ does not permit named variadic macros [-Werror=variadic-macros]
 #define GERB_MESSAGE(t...)  g_log(NULL, G_LOG_LEVEL_MESSAGE, ##t);
                       ^~~
cc1plus: all warnings being treated as errors
Makefile:980: recipe for target 'gerberimporter.o' failed
```

Can be easily enough fixed with:

```
diff --git a/Makefile.am b/Makefile.am
index c919097..393abc6 100644
--- a/Makefile.am
+++ b/Makefile.am
@@ -58,7 +58,7 @@ ACLOCAL_AMFLAGS = -I m4
 GIT_VERSION = `git describe --dirty --always --tags`
 GERBV_VERSION = `pkg-config --modversion libgerbv`
 
-AM_CPPFLAGS = $(BOOST_CPPFLAGS_SYSTEM) $(glibmm_CFLAGS) $(gdkmm_CFLAGS_SYSTEM) $(gerbv_CFLAGS) $(CODE_COVERAGE_CPPFLAGS) -DGIT_VERSION=\"$(GIT_VERSION)\" -Wall -Wpedantic -Wextra -Werror
+AM_CPPFLAGS = $(BOOST_CPPFLAGS_SYSTEM) $(glibmm_CFLAGS) $(gdkmm_CFLAGS_SYSTEM) $(gerbv_CFLAGS) $(CODE_COVERAGE_CPPFLAGS) -DGIT_VERSION=\"$(GIT_VERSION)\" -Wall -Wpedantic -Wextra -Werror -Wno-variadic-macros
 AM_CXXFLAGS = $(CODE_COVERAGE_CXXFLAGS) -DGIT_VERSION=\"$(GIT_VERSION)\" -DGERBV_VERSION=\"$(GERBV_VERSION)\"
 AM_LDFLAGS = $(BOOST_PROGRAM_OPTIONS_LDFLAGS)
 LIBS = $(gerbv_LIBS) $(BOOST_PROGRAM_OPTIONS_LIBS) $(CODE_COVERAGE_LIBS)
```

Once you get past that, we have:

```
  CXX      gerberimporter.o
gerberimporter.cpp: In member function ‘virtual std::pair<boost::geometry::model::multi_polygon<boost::geometry::model::polygon<boost::geometry::model::d2::point_xy<double> > >, std::map<double, boost::geometry::model::multi_linestring<boost::geometry::model::linestring<boost::geometry::model::d2::point_xy<double> > > > > GerberImporter::render(bool, bool, unsigned int) const’:
gerberimporter.cpp:887:45: error: ‘GERBV_INTERPOLATION_LINEARx10’ was not declared in this scope
     } else if (currentNet->interpolation == GERBV_INTERPOLATION_LINEARx10 ||
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gerberimporter.cpp:887:45: note: suggested alternative: ‘GERBV_INTERPOLATION_LINEARx1’
     } else if (currentNet->interpolation == GERBV_INTERPOLATION_LINEARx10 ||
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                             GERBV_INTERPOLATION_LINEARx1
Makefile:980: recipe for target 'gerberimporter.o' failed
```

Unfortunately, it's an enum, so it can't be handled with a simple #ifndef. YYMV, but this is the sort of thing I usually work around using autoconf tomfoolery:

```
diff --git a/configure.ac b/configure.ac
index ccaf3be..eae95a6 100644
--- a/configure.ac
+++ b/configure.ac
@@ -63,6 +63,8 @@ AC_SUBST([gdkmm_CFLAGS_SYSTEM], ['$(subst -I,-isystem ,$(gdkmm_CFLAGS))'])
 AC_SUBST(gerbv_LIBS)
 AC_SUBST(gerbv_CFLAGS)
 
+AX_C_COMPILE_VALUE(GERBV_INTERPOLATION_LINEARx10,LINEARx10,gerbv.h)
+
 # Checks for header files.
 AC_HEADER_STDC
 
diff --git a/gerberimporter.hpp b/gerberimporter.hpp
index edbcb2c..9372289 100644
--- a/gerberimporter.hpp
+++ b/gerberimporter.hpp
@@ -29,6 +29,10 @@
 
 extern "C" {
 #include <gerbv.h>
+
+#if COMPILE_VALUE_LINEARX10==0
+#      define GERBV_INTERPOLATION_LINEARx10 GERBV_INTERPOLATION_x10
+#endif
 }
 
 class gerber_exception: public std::exception {};
```

If you're okay with those approaches I can do up a pull request, but they're trivial enough that I'm not sure it's necessary.

eyal0, 2019-11-11:
Ah, I'll tell you what I did, I made all those warnings into error and then I fixed them in my private branch of gerbv.  I also fixed a few other things because gerbv has some real problems and I got tired of waiting for the gerbv people to get around to it.

I'll make it so that compiling with -Werror is off and only turn it on as part of my own development and TravisCI.  For now, just remove the -Werror line from here: https://github.com/pcb2gcode/pcb2gcode/blob/aa71eaf94a35bb13365580891f45f7b45026eab1/Makefile.am#L61

Sorry about the inconvenience!

c-, 2019-11-11:
Sure, removing -Werror fixes the one problem...

But the problem with the GERBV_INTERPOLATION_LINEARx10 not being in 2.6.1 isn't going to be fixed with a compiler flag.

eyal0, 2019-11-11:
I see.  Yes, that change is more recent.  It was made in 9b768ea3284b792e303ecf10352b0a728ac66af9 on October 3rd.

I'll look into fixing that, too.

eyal0, 2019-11-11:
Sounds like you have a better idea of how to fix this than I!  Please send a PR.

I tried to put -Werror as a flag: https://github.com/pcb2gcode/pcb2gcode/pull/364

Did I do it right?  If you have a better solution, I'm happy to try it!

c-, 2019-11-11:
> Did I do it right? If you have a better solution, I'm happy to try it!

That looks about right.

Philosophically, just using -Werror -Wno-variadic-macros gives pedantry with minimum compromise, but I have a sneaking feeling that -Werror may cause problems on other platforms/distros/versions so a more liberal approach might be best outside of a test environment.

I'll do up a PR for the other problem.

PR #365 (2019-11-15): Allow us to use the GERBV_INTERPOLATION_LINEARx10 symbol with 2.6.1
Interpolation types are enums, so the usual #ifdef magic doesn't work to allow backwards compatibility. Use automagic to detect whether the symbol is defined and make it available if not. Should fix #362.
closed 2019-11-15 by eyal0