pcb2gcode/pcb2gcode #397: Kicad generated drill file ==> bad_alloc

quartane, 2020-01-30:
Hellow everyone, minor bug here 

I'm using 
> 2.0.0
> Git commit: v2.0.0-294-g513cdb4-dirty
> Boost: 106700
> Gerbv: 2.7.0

Using a drill file generated from KiCad, the drill file creation process crash with a bad_alloc.
the crash occur at the line 

> curBit.unit = string(currentDrill->drill_unit);
of the file drill.cpp

What happend?
For some reason, gerbv  return me the default tool.  The default tool of the gerbv has "drill_unit" = NULL
So the string function crash ;)

I used gerbv version
> gerbv version 2.7.0
> Copyright (C) 2001—2008 by Stefan Petersen
> and the respective original authors listed in the source files.

What did you expect to happen:
I thinks that gerbv should give a non null unit tools by defaukt but I thinks too that the software should manage the exception better.

either by ==> throwing an identified error telling the user that his drill file containe any or default tool.

or  by not crashing and not adding the default tool to the list of tools available (that's how I patched mine but I think the first way is cleaner)

In any Case, a big printf that warn the user that for any reason gerbv returned his default tool is welcome

I hope I've helped.  

By the way thanks for the hard work given :)  that's great.   And have a good day

eyal0, 2020-02-03:
Do you have a sample drill file and command line that I can try so that I can see the error, too?

quartane, 2020-02-03:
I do,

here the command used to call pcb2gcode :
`pcb2gcode --output-dir  "./generatedGCODE/" --zchange  20 --metric --drill-speed  10000 --bridgesnum  4 --bridges  0.5 --cut-feed  250 --zdrill  -1.6 --extra-passes  0 --cutter-diameter  3 --zcut  -1.56 --fill-outline --zbridges  1 --mill-feed  300 --offset  0.3 --front  "/home/jag/test/testCNC-F_Cu.gbr" --mill-speed  800 --cut-speed  12000 --drill-feed  1000 --outline  "/home/jag/test/testCNC-Edge_Cuts.gbr" --optimise --zsafe  1 --cut-infeed  0.5 --zwork  -0.26 --drill  "/home/jag/test/testCNC-NPTH.drl" --nog64 `

Here's my "project" test
[testCNC.zip](https://github.com/pcb2gcode/pcb2gcode/files/4148068/testCNC.zip)

eyal0, 2020-06-26:
Ah yeah, I see that, too!  I'll take a look, sorry to keep you waiting!

eyal0, 2020-06-26:
How did KiCad select no tool?  Strange.  I suppose that I can work around it.

PR #449 (2020-06-26): Check for null drill_unit from gerbv.
This can happen if there are no holes to drill.
This fixes #397
commit 10f8e054 (2020-06-26): Check for null drill_unit from gerbv.

This can happen if there are no holes to drill.
This fixes #397
closed 2020-06-26 by eyal0