pcb2gcode/pcb2gcode #399: pcb2gcode ignores narrow traces

skorokithakis, 2020-02-07:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):

```
2.0.0
Git commit: v2.0.0-294-g513cdb4
Boost: 106700
Gerbv: 2.7.0
```

What did you try (**include command-line arguments**):

millproject:

```
# Common options
metric=true
metricoutput=true
optimise=true
zero-start=true
zsafe=4
zchange=5
zwork=0.1

invert-gerbers=true
mill-diameters=0.3mm
milling-overlap=10%
isolation-width=2000mm
mill-feed=500
mill-speed=0
output-dir=output/
```

I tried to process the following file:

[image]

What happened:

As you can see, pcb2gcode ignored some of the traces because they were too narrow. It looks like it was trying to make two lines for them, which would have been too thick, so it just didn't try.

What did you expect to happen:

I would have expected it to either only make one line, or overlap the two lines.

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

[output.zip](https://github.com/pcb2gcode/pcb2gcode/files/4168494/output.zip)

eyal0, 2020-02-07:
That is surprising to me!  I expected that it would go into those tight spaces and output a warning letting you know about that.  I will take a look at this within the next 3-5 days, maybe sooner.

skorokithakis, 2020-02-07:
Thank you! It seems that the correct behavior would be to draw a single line, rather than two.

eyal0, 2020-02-07:
I agree.  I am still thinking about how to do this!

The way that most shapes are drawn is that we start with the shape, then we enlarge it, then again, then again, and for each one, draw an outline.  Those outlines are the path that the milling tool will take.  So if you start with a rectangle, you get a bigger rectangle with rounded corners, then even bigger, etc.

Now, these shapes are on the **inside**, so when I enlarge, the holes start to *disappear*.  Like if you take a doughnut shape and you enlarge, the hole inside the doughtnut will get smaller and smaller until it vanishes!

That is what happened here.  I enlarged the shape and the narrow strip disappeared.

This doesn't happen if I have two unconnected shapes because I use the voronoi algorithm to make sure that the traces are never escaping their area.  So two shapes that are very close to each other are still in different regions.  But in your case, it's just one big shape: a big, inverted PCB.

I will need to think about an algorithm that can enlarge a shape and turn very narrow spaces into a line, instead of making it disappear.  The only tools that I have are the normal shape operations, like enlarge, shrink, union, intersect, etc.

I will think about this.

skorokithakis, 2020-02-08:
I see, thank you for the explanation. Would it be helpful if you ran the algorithm until two lines (or one) were left, and then never ran it again? It might fail with bridging things that shouldn't be bridged instead of disappearing things that shouldn't disappear, which feels like a better failure mode to me, at least.

eyal0, 2020-02-10:
I thought about that but how would I do it?  Like, first I would do it and see it disappear.  Then I would widen it a bit and try again and this time it would be too wide.  So then I'd have to narrow it a bit...

I'd have to go back and forth a bunch of times to narrow down on the right number and that would potentially take a long time.

Sometimes there are things that can be done with the shape functions that solve this.  Like instead of shrink a shape, I can invert the shape, grow it, and then invert it again.

Voronoi is able to find the absolute middle between different shapes.  So, for example, if the two sides of the trace weren't connected, we already have algorithms to bisect the space in between.  That would draw a line right down the middle of your trace, which would be perfect!  But your inverted PCB is all one big shape so it doesn't know where to separate.  Maybe there is a way to fake it.  I'll keep thinking about this.

skorokithakis, 2020-02-10:
Hmm, yeah. I meant that you'd shrink down to 2 lines and then leave those there, even if they'd be too wide (you wouldn't shrink past 2 lines). Basically, err on the side of too-wide, rather than "disappeared".

I don't have a good grasp of the problem, though, so I may be way off.

eyal0, 2020-02-10:
I'll explain some more.

The geometry library that I'm using is boost.  Boost has a function called "buffer" that will take a 2 area and grow it or shrink it.  If you start with a doughtnut shape and grow it, it'll have a bigger outer circumference and a smaller inner "hole".  If you grow it by a lot, the inner hole will disappear and merge.

The way that pcb2gcode works, it starts with the shape and grows it by half the diameter of the tool.  The outline of that shape is the path to mill.  If you want extra passes, you can just keep growing it by the diameter of the tool minus the overlap.

In your case, the tool is wider than the trace, so when I grow the shape, the narrow trace disappeared.  I could grow the shape by less but then the question is: how much less.  Ideally I would grow it just enough to make the trace turn into a very very thin line, and then just run the tool along that.  But how can I know how thin to go?

One way is to shrink the space a little at a time until the trace disappears and then go back a bit.  But I would be doing a lot of guess-and-check and it would be really slow.

Another way might be to measure the width of the trace and divide that by two.  But traces have complicated shapes.  And anyway, you don't want that behavior at the ends of the trace where it's a big pad.  So it needs to be a mix.

It's complicated but the grow and shrink functions can be customized.  What I'd like is a grow function that would make a doughnut hole disappear.  The doughtnut holes should become points and the doughnut slots should become lines.  Does that make sense?

That'll take some time, though.  Unless I think of something smarter.  Maybe it's already built-in to boost geometry?  I need to read about boost geometry buffer.  https://www.boost.org/doc/libs/1_72_0/libs/geometry/doc/html/geometry/reference/algorithms/buffer/buffer_7_with_strategies.html

skorokithakis, 2020-02-11:
Ahh, I see, so the shape you're growing is the non-copper instead of the copper region? I think that's what was tripping me up before. It seems to me that you need a grow function that will preserve shapes and not disappear them, yes.

To do that, you need to have a way to check that the area that was there is still there, and hasn't completely disappeared, but that's hard. The other thing that's happening that you might be able to check for is that one contiguous area gets turned into two disconnected areas (the pad and the rest of the circuit). That might be an easier condition to check for.

mjml, 2020-04-16:
Version 2.0.0?

Where do I get this? I am working off of master and I get 1.3.2.

eyal0, 2020-04-17:
@skorokithakis Yes, that would work, I could check if the shape becomes more than one shape.  But then what?  Like, say I want to grow by 30mm and it becomes two shapes.  Then I try 15mm, now it's just 1 shape.  So now I try 22.5mm.  It's still one.  So now I try 26mm.  Now it's 2, then I try less....  I would be narrowing in down on to the right answer but each step is taking a lot of computing times!

Maybe a better way is to look for places where the shape will shrink and become to small?  If I examine the whole shape, is there a way for me to determine it?  Each wall of the shape moves in a perpendicular motion, right?  So if you have a square, the top and bottom move up-down and the sides move left right.  Whichever direction the boundary stretches, that boundary will move sideways.

So I could look at each boundary and try to find how far it can be moved until it hits a different boundary.  In the case of a thin trace, the side of the trace can be moved just the width of the trace.  I do that for each side, find the more restricted one, and then I shrink the whole shape by half of that value.  Maybe that would work?  I'm not sure how complex it is to compute that.

skorokithakis, 2020-04-17:
@eyal0 I don't think you need to bisect it like that, can't you grow until it disappears and then back off the width of the trace? That should leave you with exactly the one trace, no?

eyal0, 2020-06-02:
@skorokithakis bisecting would actually be faster because it would narrow down on the right size.

How would I do the grow and back off?  Like, what is the first grow amount?  I could try, say, 0.00001 each time, but that would be many, many times to grow to reach whatever I'm trying to reach.  And each time that I try the grow takes a few milliseconds.  If I have to do many of them, it'll really add up.

skorokithakis, 2020-06-02:
I don't think you need to go that thin, half the tool width seems like a good idea, since you're looking for a line that will be `tool_width` mm thick.

eyal0, 2020-07-06:
Some of the changes that I did around overlap width accidentally fixed this problem a little bit.  Maybe try with the latest version?

skorokithakis, 2020-07-06:
I will, thanks!