pcb2gcode/pcb2gcode #401: milldrill gcode doesn't include any toolchange directives

c-, 2020-02-19:
2.0.0
Git commit: v2.0.0-294-g513cdb4
Boost: 106501
Gerbv: 2.6.1

I have a separate PTH and NPTH drill config, and after running the PTH drills with a tool change and tool offset probe, I ran the NPTH and it... spun up and drove itself through the PCB.

The milldrilldiatest test case shows the situation clearly: https://github.com/pcb2gcode/pcb2gcode/blob/master/testing/gerbv_example/milldrilldiatest/expected/drill.ngc contains a series of tool changes. https://github.com/pcb2gcode/pcb2gcode/blob/master/testing/gerbv_example/milldrilldiatest/expected/milldrill.ngc has no tool change.

The tool change cycle seems to be necessary for the machine/software (I'm a bit fuzzy where the calc is done) to know about the tool length offset, since it doesn't seem to carry over from previous runs. Starting a job without the tool change causes it to assume the original tool length that the last Z zero probe was at; in my case, that was the engraving v-bit, which is considerably shorter than the drill. GRBL acted like zsafe was approximately -9.0mm, and while I was able to save the PCB, the drill didn't survive.

So it would be good if a milldrill file included the tool change operations like everything else seems to use.

It doesn't look like a difficult fix, just copy the tool setup from the drill logic to the milldrill?

eyal0, 2020-02-21:
Hmm.

Currently, when people make a PCB, there can be many steps.  Like you might have milling, and drilling, and outline.  If you only use one bit for each then `pcb2gcode` assumes that the user will always adjust the tool offset before each file.  So there is no tool change in any of those files.

But, if you use multiple drills or multiple milling bits, then we need a tool change for each one because all that drilling is in the same file.

So maybe the best and most consistent thing would be to begin every file with a tool change.  The disadvantage would be that if you already set the tool offset correctly when you start the file, it'll ask you again to do it, which is a very small waste of time.

The advantage, however, is that you could now take all the files and concatenate them into a single file and run that and it would have toolchanges in all the right places.  This seems the most consistent to me.

Does that sound reasonable to you?  I'll make sure that every file starts with a tool change.  And each file will continue to have tool changes between different tools.  How does that sound to you?

I cannot offer you money for your broken drillbit so I only have an apology to offer you.

c-, 2020-02-22:
>  If you only use one bit for each then pcb2gcode assumes that the user will always adjust the tool offset before each file.

That's definitely not the case with outline; it generates tool change even for one bit. And that's also the case for drill files where there's only one bit.

See attached example:

[issue401.zip](https://github.com/pcb2gcode/pcb2gcode/files/4238726/issue401.zip)

Run it as is, and you get a drill.ngc file that uses one bit (1.0mm) and *includes a tool change section*.

Now, here's where we get the problem. *Change* the drill diameter in the .drl file from T1C1.000 to T1C1.200 and run it again, and you get... a milldrill.ngc file that uses one drill (again, 1.0mm) and *doesn't include a tool change section*.

So it looks like it's specifically that the milldrill code path isn't doing the tool change output everything else is doing.

> So maybe the best and most consistent thing would be to begin every file with a tool change.

That's what I'm thinking... consistent tool handling across all the g-code output. That being said, if you were to implement #307 then an option to disable tool changes entirely could make sense... --nom6 maybe?

> I cannot offer you money for your broken drillbit so I only have an apology to offer you.

Not a problem; I buy my most-used drills in cheap ten packs. Even the minor damage to the PCB turned out to be a non-issue as I'd messed up the position of some pads and had to bin it.

eyal0, 2020-03-18:
Fixed in #405

c-, 2020-03-18:
Confirmed, thanks, that seems to do it.

closed 2020-03-18 by c-