pcb2gcode/pcb2gcode #409: Can't create bridges that span multiple edge cut segments (eg: with a circle outline)

mjml, 2020-04-07:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
  1.3.2

What did you try (**include command-line arguments**):
  pcb2gcode --front bikelight-F_Cu.gbr --front-output bikelight-F_cu.ngc \
            --back bikelight-B_Cu.gbr --back-output bikelight-B_cu.ngc \
            --outline bikelight-Edge_Cuts.gbr --outline-output bikelight-Edge_Cuts.ngc \
            --drill bikelight-PTH.drl --drill-output bikelight-PTH.ngc \
            --milldrill --milldrill-diameter 0.6 --zdrill -1.5 --drill-feed 80 --drill-speed 10000  \
            --mill-speed 10000 --mill-feed 80 --mill-vertfeed 80 \
            --metric 1 --metricoutput 1 --optimise \
            --zwork -0.2 --zsafe 2 --zchange 12 \
            --offset 0.12 --extra-passes 2 \
            --cut-feed 120 --cut-vertfeed 120 --cut-speed 10000 --zcut -1.5 --cut-infeed 0.305  --cutter-diameter 1.52 \
            --bridges 3 --bridgesnum 3 --zbridges -0.05

What happenned:
Importing front side... DONE.
Importing back side... DONE.
Importing outline... DONE.
Processing input files... DONE.
Exporting back... DONE. (Height: 35.7216mm Width: 35.7216mm)
Exporting front... 
Warning: pcb2gcode hasn't been able to fulfill all clearance requirements and tried a best effort approach instead. You may want to check the g-code output and possibly use a smaller milling width.
DONE. (Height: 35.7216mm Width: 35.7216mm)
Exporting outline... Can't fit any bridge in the specified outline. Are the bridges are too wide for this outline? Are you sure you've selected the correct outline file?
DONE. (Height: 35.7216mm Width: 35.7216mm) The board should be cut from the FRONT side. 
Importing drill... DONE.
Exporting drill... Warning: 7 holes were bigger than the milling tool.
DONE. The board should be drilled from the FRONT side.
END.

What did you expect to happen:
Expected the utility to generate Edge.Cuts such that there are tabs/bridges in the outline suitable for milling from larger copper blanks. This doesn't happen because the outline is a circle, and no segment is longer than the width of a bridge. You'd need the bridge logic to be able to span multiple (small) segments.

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**
[bike-light.zip](https://github.com/pcb2gcode/pcb2gcode/files/4441453/bike-light.zip)

eyal0, 2020-04-07:
Yes it's true. This is a drawback with how the outlines work. I'd like to make a better way. Do you have a suggestion? Given just some random shape, how do you choose where to put the support bridges?

What if they were instead placed like a dashed line? So the perimeter would be like 5cm no bridge then 1cm bridge then 5cm without then 1cm bridge, etc. And it would follow the curvature of the shape. That would fix your problem?

Does it seem like a good solution in the general case?

mjml, 2020-04-07:
At the moment I hand-edit the file and put G0 Z2 then G0 Z<df> around a couple of the known segments using search and replace. 

I think the best solution would be to start at a given point, then subtract out partial or whole line segments until you have the length of the bridge.

However, it doesn't avoid the other problems associated with placement of the bridges themselves. Any automated solution can accidentally placie the bridges near troublesome places. For example in the finger areas of a keyed connector. A quick fix would be to include a "bridge-phase-offset", which would be just a simple offset before the placement of the first bridge. Should the utility put a bridge in an inconvenient place, you could run again with a different bridge-phase-offset value and it would "rotate" the bridges along the outline contour by a certain distance.

eyal0, 2020-04-10:
What you describe is the same as what I describe: A dashed line around the perimeter, where it's X cm no bridge, then Y with bridge, etc.  Like:

```
........___........___........___........___
```

Making the way around the edge.  And the offset in case you see, experimentally, that it's putting the bridge in a lousy place.  Yeah, I guess that could work...

I'll think about it.  I kind of like the algorithm that it has now, where it tries to make the bridges as far apart from each other as possible.  I think that gives a good layout.  Maybe I'll see if I can make the bridges bend around the corners yet still keep that feature?

c-, 2020-04-28:
Just looking at this... what if the user had the option to directly specify the number and distance (absolute, relative?) of bridges around the perimeter from where the edge cuts start? i.e. bridge_at=0mm,5mm,12mm,18mm,25mm would give 5 bridges centered on those distances.

It could even work for internal cutouts if bridges were calculated as a continuation of the main outline. It'd require some experimentation for the user to figure out where everything goes, but you get precise, controllable, and repeatable bridge positioning out of it.

eyal0, 2020-07-06:
I fixed this.  I didn't implement the idea about specifying where bridges go.  I just made it so that bridges can go around curves.  Bridges continue to be placed such that the center of the bridges are far apart from one another.

eyal0, 2020-07-06:
If you see this not performing well enough in practice, let me know.

closed 2020-07-06 by eyal0

mjml, 2020-07-24:
Thank you, I will have a look soon.