pcb2gcode/pcb2gcode #413: Gcode optimization, or lack of.

terrylm, 2020-06-03:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
2.0.0
Git commit: 
Boost: 106501
Gerbv: 2.6.1

What did you try (**include command-line arguments**):
Used GUI interface under Ubuntu 18.04.4

What happenned: <-- Too many 'n's

All worked well, save that the gcode is very non-optimal.

What did you expect to happen:

I would expect that when isolation milling, once the tool is cutting, it would stay down and cutting as long as possible, like a doodle. What actually happens though, is that the tool will pull up, when it could have keep cutting, then move to some nearby location, then drop back in and start cutting again.

Some times the "new location" is very near the "old location", then the cutter cuts to the "old location", reverses and proceeds to cut what is should have just continued to cut in the first place.

Some times when it leaves a cut, there is only a tiny amount left to finish it, but it goes else where, then comes back later to finish it.

Some time it will reverse a cut. That is fine for very short cuts, but some times it will cut many mm, only to reverse and redo the whole thing in the opposite direction.

At this time I have only a low speed spindle to engrave with, which means a very slow feed rate. And all the unneeded up/down and rapid traverses, plus the occasional long reverse cuts, literally add hours to the run time.

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

Humm... it seems you don't accept gerber files? .gbr
"We don't support that file type. Try again with a GIF, JPEG, JPG, PNG, DOCX, GZ, LOG, PDF, PPTX, TXT, XLSX or ZIP."

No .gbr or .gcode, .ngc or .cnc allowed?

Oh well, they are big files anyway, would add almost 10k lines for the gcode file.

No .svg either? OK, I made a .png of both sides of the board:
[image]

terrylm, 2020-06-03:
I am willing to work on optimizing the g-code output. Some developer docs or other guidance as to the inner workings of the program would be helpful.

eyal0, 2020-06-04:
pcb2gcodeGUI is based on an old version of pcb2gcode.  The latest version has new features and works a good deal better.  However, it doesn't have a GUI because I have never used or worked on the GUI.

Would you be able to try using the latest version of pcb2gcode?  Usually users build it themselves.

If you're able to send the gerbers and the command-line that you used to run it, I can run it on my computer and look at the output gcode and SVG files myself.  If not, then maybe send the portions of the SVG file that don't look good and we can work with that.

Any file that github won't let you upload, just zip it.  Thanks!

terrylm, 2020-06-04:
Everything was built from current sources, pulled as zip files from github. However, I removed that version and instead cloned pcb2gcode and built/installed it again. The only difference seems to be a slightly newer gerbv, there seems no substantial difference in the output files.

I am looking at the source, still wrapping my head around it. My programming skills are a bit rusty, and C++ is not something I have used a lot. Looks like I need to learn about std::map ...

While I do not know what command line the GUI created originally, here is the command line I used this time:

pcb2gcode --front LatheControler.top.gbr --back LatheControler.bottom.gbr --drill LatheControler.drill.cnc --nog64 --zwork -0.5 --zsafe 1.0 --mill-feed 10.0 --mill-speed 1000 --vectorial --voronoi --zdrill -2.5 --zchange 100 --drill-feed 20.0 --drill-speed 1000 --nog91-1 --software LinuxCNC --metric --metricoutput --zero-start

The output was essentially the same.
Here is the zip you asked for. I did change the tool number and feedrates of the copy on the mill, but  the files in the old directory are the originals.

[example.zip](https://github.com/pcb2gcode/pcb2gcode/files/4731915/example.zip)

eyal0, 2020-06-04:
Sure, I'll take a look!  So that I'm clear on what is and isn't working: You're happy with the final PCB, like, it milled and drilled in all the right places?  You'd just like it to take a better path to save time?

eyal0, 2020-06-04:
I think that I see the issue.  I'll describe it.  Sounds like you know some programming so I'll get technical.

pcb2gcode takes the path that we're trying to mill and converts it into a bunch of curves that don't overlap.  Then it creates a Eulerian path on those segments so that it it will traverse each path exactly once, in the direction specified as needed, but only once.  And it creates an optimal number of paths, connecting paths where possible.

However, that constraint that each path is traversed just once slowing you down because the time to lift the tool, move, and set the tool back down is longer than the time to just re-mill a path.

If you set --eulerian-paths=false then you'll see a different output but it's still not great for your case.

Ideally, we keep eulerian paths working but relax the constraint that an old path can't be re-milled.  I think that would give you the output that you want: like a doodle which gets all the paths but sometimes retraces itself.  Does that sound right to you?

If you want to take a look at it, that specific function is pretty well documented.  It's `eulerian_paths.cpp`.  The input is a vector (a list) of pairs.  And the pairs are of a linestring and a boolean.  The linestring is a list of points that describe the curve.  The boolean says whether or not the path is reversible (the comment there is wrong, I'll fix it.)

The output is a vector of linestrings, which is a multi_linestring.  If you fix that, it should fix your problem.  let me know how much effort you are willing and able to put in here.

eyal0, 2020-06-04:
One idea that I have is post-processing the eulerian path results.  We can do all the eulerian paths and then loop through all the paths created and join paths where we find that the start of one path matches a point along another path, like in a T shape.  In those cases, do one path, then go out and back along one of them, then continue for the rest of the path.  Like tracing a capital T letter without lifting your pencil.

I haven't been able to figure out if there is an optimal ordering for doing this.  It seems to me like you want to double-back on the shortest leg of it.  But it's not clear to me how to pair up all the different line segments to find which one to do.  I'd probably start with a greedy algorithm that just does the shortest first and goes from there.  Maybe it's fast enough that it doesn't matter.

Seems easy enough that we can just try it and look at the output.

I'm also wondering if there is some sort of back-tracking cut off.  I have that for the path-finding: pcb2gcode measures the time that it would take to mill versus the time to lift the tool, run over to the location, and drop it back down.  And it only uses path-finding if that is a faster solution.  So similarly for doubling-back, it only makes sense if it is actually faster.  And that is only known after the travelling salesman problem solver (TSP).  So maybe we need to do TSP first and then see where it is possible to double-back.

Of course, all this would be controlled by a flag so that people who really really don't want to double-back on a path don't have to do it.  For example, people are using pcb2gcode with lasers to use a chemical process.  So they might be sensitive to the doubling-back.

That's my thinking so far.  What do you think?

terrylm, 2020-06-05:
OK, I was already looking at some of what you mentioned, but was not sure about what I was looking at. Will look again now that you have confirmed that is where to look.

Doubling back was not what I was wanting so much, although for very short distances it might be faster, so is OK.

By doodle I had in mind not so much that it might backup, as that it would just keep moving forward as long as there was a possible forward that had not been milled already. So at the end of a path, if there is another path that keep going from that location, in any direction except back, then take it. If there are multiple choices, then just take the first one found that has not already been milled, or the one that makes for the straightest extension of the previous path... or some such. At a four way intersection, it would be OK to cross an already milled path.

I will get back to it maybe tomorrow, if not, then the next day. I have some other commitments tomorrow.

Good night

eyal0, 2020-06-05:
> By doodle I had in mind not so much that it might backup, as that it would just keep moving forward as long as there was a possible forward that had not been milled already. So at the end of a path, [...]

What you have described is how it already works. Except instead of randomly selecting at each intersection, it does it ahead of time and then looks for loops to stitch into the milling. It actually arrives at an optimal solution that lifts the milling bit the fewest times possible (which is half the number of vertices with odd degree).

It's just that your design had a lot of vertices with odd degree. If you look closely at the drilling path, you'll find that it always starts and ends at a vertex with odd degree. It's optimal.

terrylm, 2020-06-05:
In Surface_vectorial::post_process_toolpath() the call:
bg::simplify(combined_toolpath, temp_mls, mill->tolerance);

I added a cout to see what mill->tolerance was ... 0.0004, so then just to see what it would do, I changed it to 0.004... WOW!!! File size for front.ngc went from 352kB to 150kB! Viewing it on the LinuxCNC screen, it looks the same. So I tried changing tolerance to 0.01, not as dramatic of a change, but still noticable, file size of front.ngc dropped to 112kB with still no noticeable change on the screen of the mill.

I did not chase down where tolerance comes from. There is nothing obvious in the man page. Is this a calculated value? or does it come directly from a command line argument that I missed? I also have no idea what units it is in?

It should be noted that I have not tried to actually mill with the new files, and it looks like they still have a lot of probably unneeded pull outs.

terrylm, 2020-06-05:
eyal0, I am not sure what you mean by "odd degree"? Like X^3?

eyal0, 2020-06-05:
Simplify removes the number of points.  It is a trade off between the accuracy of curves and the file size.  So if you increase that tolerance, you'll find that your round corners are more bumpy.  Might be fine for you.  The algorithm is this one:

https://en.wikipedia.org/wiki/Ramer%E2%80%93Douglas%E2%80%93Peucker_algorithm

As for odd degree, I mean the number of vertices with an odd number of edges out of them.  1, 3, 5, etc.  Like this in image:

[image]

There are 8 vertices with degree 3.  All the rest have degree 2 or 4.

Try to doodle that shape without retracing any paths and without lifting your pencil (crossing paths is allowed).  It's impossible without lifting your pencil because you need at least 4 lines to do it.  You can try it yourself.

pcb2gcode computes exactly 4.  Between each of the four, it lifts the milling tool and moves to the next one.  There is no way to do less than 4 without backtracking over a previously drawn trace.

c-, 2020-06-05:
> There is no way to do less than 4 without backtracking over a previously drawn trace.

For sake of argument, is backtracking over a previously drawn trace always bad? If the cutter ultimatelly has to travel the same distance, there's only a gain in milling time if a z-lift+rapid move+z lower takes longer than just retracing the path, which seems unlikely for very short moves. An obvious risk would be in double-cutting the part (i.e. an accidental spring pass), but that might be worth a performance gain.

Another potential issue I've noticed with the optimized g-code is that is seems to have a predilection for sharp direction changes (i.e. backtrack down a trace rather than continuing on as far as possible). In terms of pure path optimization that's probably correct, but acceleration/deceleration being what it is, I wonder if it's really that optimal? And then there's the fact that sharp reversals introduce backlash error. I can't say I've identified this as the source of any errors on my machine, but it's been low-grade bothering me for a while.

eyal0, 2020-06-05:
Does the unoptimized g-code have this problem?

Optimization just decreases the number of gcode lines, sacrificing a bit of the roundness of some curves.  But it ought to be very minor, almost not noticeable!  Shouldn't have much affect on the output.

When you use the word backtrack, let's be clear.  Above when I wrote backtrack, I was referring to the process of re-milling the exact same path.  What you're saying sounds more like whether or not there is a curve in the path.  Am I getting that right?

What you say sounds interesting.  Right now, the eulerian path algorithm just randomly chooses a direction.  For example, in the grid above, it will start a path at an intersection that is a T and from there, travel around randomly until it hits a dead-end.  Maybe instead of randomly selecting a vertex, it would be better to select the vertex that leads to the straightest path?  We could keep it simple and just, at each opportunity, greedily pick whichever gives us the straightest path.  It might be that picking a smooth path now causes us an even bumpier path down the road but it might still be good enough.

What do you think?  Seems like it would be a piece of cake for me to implement so I can give it a shot and then maybe someone can eye-ball the results and see if it looks nicer?

c-, 2020-06-06:
>  Above when I wrote backtrack, I was referring to the process of re-milling the exact same path. What you're saying sounds more like whether or not there is a curve in the path. Am I getting that right?

No, I do mean remilling the same path. There are a lot of situations where I see the CNC doing a z lift, and *tiny* (like, maybe a mill diameter or two), then a z lower, where it *could* probably have just followed a section of milled path more directly (or otherwise passed through a previously milled space). I find that this is most noticeable when a smaller milling bit is "cleaning up" areas a larger mill can't cut.

> What do you think?

Yeah, that's exactly what I had in mind... give smaller direction changes a higher weight whenever the optimizer has to resolve a fork.

I'll grant that there will be times where it might lead the optimizer down a long track where it has to do a longer rapid to get to the next trace, but I'd think that's an acceptable tradeoff for less accel/decel and backlash.

I'd be very interested in this, because it's something that really bugs me when I'm watching the CNC work; that sort of error accumulation can't be healthy.

eyal0, 2020-06-06:
Okay, so we have two ideas here and this bug has veered from it's original proposal, which was Euler paths (aka doodles). Pcb2gcode already does an optimal Euler path, even respecting partially directed graphs. So I'll close this and open two new issues:

First, what I'll call backtracking. We want to allow for milling backwards along undirected segments of doing that will be faster than lifting the tool, moving, and lowering.

Second, we want to make paths have fewer sharp turns. When the milling bit reaches an intersection, rather than select a random direction, select a direction that continues straight on.

closed 2020-06-06 by eyal0