pcb2gcode/pcb2gcode #414: --path-finding-limit shows up with --help, but not in man page and has no effect.

terrylm, 2020-06-04:
Please fill this out:

pcb2gcode version (run pcb2gcode --version to see this):
2.0.0
Git commit: v2.0.0-317-g4f60273-dirty
Boost: 106501
Gerbv: 2.7.0

Dirty... I added a blank line in board.cpp to make it easier to read. No real changes.

What did you try (**include command-line arguments**):
pcb2gcode --help vs man pcb2gcode

What happenned:
--help lists --path-finding-limit under Optimization options
The man page has no mention of it.

Comparing output file sizes, it has no effect.

What did you expect to happen:

I was hoping for the output file to only be half as big...

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**

Not relevant, as this option seem to have been removed. It just needs to be removed from the --help listing.

eyal0, 2020-06-04:
path finding has an effect if you use it with multiple passes.  By default it is turned off because it makes pcb2gcode sometimes run much much slower and the benefit wasn't all that big.  You can try it on some of the examples in the examples directory.

eyal0, 2020-06-06:
Path finding does an optimization but it may not have been helpful for you.

closed 2020-06-06 by eyal0