pcb2gcode/pcb2gcode #415: --zero-start option gives negative x coordinates; wrong corner

Patronics, 2020-06-05:
pcb2gcode version (run pcb2gcode --version to see this):
2.0.0
Git commit: 
Boost: 106600
Gerbv: 2.6.0

What did you try (**include command-line arguments**):
ran `pcb2gcode` with attached millproject. Also tried explicitly running `pcb2gcode --zero-start`
What happenned:
Output gcode goes from (-X_size, 0) to (0,Y _size), Positive Y coordinates but negative X coordinates.
What did you expect to happen:
Output gcode goes from (0,0) to (X_size, Y_size), all positive coordinates.
relevant files: 
[pcb2gcodeZeroStartBugReport.zip](https://github.com/pcb2gcode/pcb2gcode/files/4738700/pcb2gcodeZeroStartBugReport.zip)

eyal0, 2020-06-06:
Were you milling the front or the back? The front will be in first quadrant. The back is in the second quadrant. That way, the front and back will match if you "fold" the PCB over the x mirroring axis, which is the line x=0.

If you want a different mirroring axis, you can set that.

Read this: https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Alignment-for-two-sided-PCB-and-offsets

If you still find that pcb2gcode is misbehaving, reopen this issue and I'll take another look.

closed 2020-06-06 by eyal0

Patronics, 2020-06-06:
Ok, thanks, I bet that was the reason! I'm milling single sided boards, only the back, so assumed the alignment options for 2 sided boards wasn't a relevant section of the manual to read.

Patronics, 2020-06-06:
@eyal0 Looking at those sections of the manual, it looks like I can definitely do what I need with --y-offset or --mirror-axis, but I'm curious why the --mirror-absolute option is deprecated, as that would seem to be a much easier option (when set to false) to have it automatically stay centered in the 1st quadrant, rather than needing to manually adjust the offset for each board?

eyal0, 2020-06-06:
In theory, the `--mirror-absolute` option is supposed to find the center of the board and mirror the board so that the mirror axis is in the side-to-side center of the board.

In practice, the math that it has was all screwy and it gave unpredictable results because it wouldn't account for the outline of the board nor tiling.  I was worried that there are people depending on its behavior would try it and get all confused so I just made it deprecated.

If your board is one-sided then I guess that you could just pick a mirror axis is that is about half way across the board.  Alternatively, most CNCs will let you jog the mill to the side and set a new (0,0) center so you could just jog to the bottom right corner and set that as your 0 coordinate.

Hope that helps!