pcb2gcode/pcb2gcode #416: Backtrack on Euler paths

eyal0, 2020-06-06:
We want to allow for milling backwards along undirected segments if doing that will be faster than lifting the tool, moving, and lowering.

_Originally posted by @eyal0 in https://github.com/pcb2gcode/pcb2gcode/issues/413#issuecomment-640084732_

eyal0, 2020-06-06:
This is called the (Chinese postman problem)[https://en.wikipedia.org/wiki/Route_inspection_problem]. The solution is basically to add some extra paths into the graph so that there will be fewer vertices with odd degree. Then just run the Euler algorithm like it already exists.

The hard part is knowing where to add the new edges in the graph. Each new edge is a path to be remilled so we want them to be short. The solution to this problem is a (maximum matching)[https://en.wikipedia.org/wiki/Maximum_weight_matching] with minimum weight. Looks like there is an (algorithm)[http://pub.ist.ac.at/~vnk/papers/BLOSSOM5.html] but that seems overly complex.

For now, we can do a simple greedy algorithm:

1.Collect all vertices that must be the start of a Euler path, because of odd-degree or unmatched directional edges.
2. For each one, start a BFS using (Dijkstra's algorithm)[https://en.wikipedia.org/wiki/Dijkstra%27s_algorithm]. End the algorithm as soon as *any* vertex that must be the end of a Euler path is found. Also end the search if the current search is already longer than the previous search or if the found path has exceeded the limit.
3. Add a reverse path along the best found.
4. Start over.

The limit in step 2 is reached when the time to lift the tool, race over to the other end, and lower again, is faster than remilling. All those values are known up front.

This is a greedy algorithm so it ought to be quick but it might not be optimal. Making it optimal will probably not be worthwhile and this seems like a good and easy first step.

c-, 2020-06-06:
Something like this seems like a great example of the basic problem:

[image]

[test-backtrack.zip](https://github.com/pcb2gcode/pcb2gcode/files/4740922/test-backtrack.zip)

At least four of those tool lifts seem like they shouldn't be necessary.

eyal0, 2020-06-06:
That's a good example.  I'll use it in my test code.  Could you also send the command that you run?

c-, 2020-06-07:
`pcb2gcode --mirror-absolute --metric=true --metricoutput --zchange 10 --nog81 --nog64 --optimise --tsp-2opt --vectorial=true --eulerian-paths=true --tile-x 1 --tile-y 1 --mill-diameters 0.45 --isolation-width 0.9mm --tolerance 0.05 --zsafe 0.5 --milling-overlap 50% --outline test-Edge.Cuts.gbr --cut-side front --zcut -2 --cut-infeed 0.7 --cut-feed 25 --cut-speed 12000 --cutter-diameter 1.2 --bridgesnum 4 --bridges 0.5 --zbridges -0.5 --outline-output test-Edge.Cuts-front-1.2mm.nc --front test-F.Cu.gbr --front-output test-F.Cu-front-0.3mm.nc --zwork -0.15 --mill-feed 100 --mill-vertfeed 25 --mill-speed 12000`

It also occurs to me that the zip file I posted is missing the edge cuts. This is more complete:

[test-backtrack.zip](https://github.com/pcb2gcode/pcb2gcode/files/4741215/test-backtrack.zip)

eyal0, 2020-06-07:
Okay if I add that to the unit tests?  It's just a circle, not exactly your top-secret design, right?  :-)

c-, 2020-06-07:
Oh, go ahead. It's actually a via... you don't need the drill files, do you? ;)

eyal0, 2020-06-12:
I tried another thing which showed up in my reading about the Chinese Postman Problem:

The whole thing of finding a milling plan can be stated as an optimization problem:

The path is a bunch of edges that need to be drawn, some potentially directional.

Also, we can "take flight" from any vertex to any other vertex by lifting the tool, zooming over to another spot, and putting it back down.

We know that a eulerian circuit, that is, a path that covers the whole thing with each edge taken just once, exists if each vertex has an even number of edges.

So I can make a bunch of variables, one for each edge, representing how many times the edge is taken.  And for each vertex, I can make a rule that the number of edges it has is the number of times that each edge that includes it is taken.  Also, I make a rule that each edge must be taken at least once.

Then I add a bunch of "flights" where we lift the tool, move to a spot, and lower it again.  That can be done for every pair of vertices.  For those, the rule is that they can be done 0 or more times.

Finally, a cost function where the cost to mill is the G1 speed and the cost to fly is the time to retract plus the time to move at G0 speed along the longer of x or y plus the time to set the tool back down.  I multiple each edge count by the cost and aim to minimize that.

It works great on small examples!  And it only took a short time to code it, like barely a day.  *However*, it's super duper slow.  On the example with the via, which is pretty simple, it's been running for 6 minutes so far and I don't see it ending.  It's an NP-complete problem.

I would settle for an approximation or whatever can be done within 5 seconds.  I'm not sure if that's something that this ILP solver provides (OR-tools with COIN-OR).

I'll go back to my greedy solution, which is doing pretty well, maybe good enough to submit it as-is.  And then I'll look into a hybrid solution that does most of the work with the greedy solution and then maybe uses ILP for the strongly-connected components?

c-, 2020-06-12:
Oh wow. Even 30 seconds for that sort of trivial example is too much.

In general, yeah, trying to optimize the whole path at that level is a NP-complete rabbit hole. I'd stick with the greedy solution. Maybe there'd also be some value to thinking of it from the perspective of #417 in the sense that it could treat a z-lift-rapid-z-lower action as an "extreme direction change" and see what allowing backtracking does to that approach?

closed 2020-06-16 by eyal0

c-, 2020-06-17:
Oh, my. A quick check suggests that this and #417 knock about 30% milling time of post #425 g-code, which was already near 30% faster. Sweeeeeettttt!

commit c31a3382 (2020-06-16): Merge pull request #427 from eyal0/backtrack

Add backtracking feature

this fixes #416