pcb2gcode/pcb2gcode #425: Only optimize at the end

eyal0, 2020-06-12:
> No, I do mean remilling the same path. There are a lot of situations where I see the CNC doing a z lift, and *tiny* (like, maybe a mill diameter or two), then a z lower, where it *could* probably have just followed a section of milled path more directly (or otherwise passed through a previously milled space). I find that this is most noticeable when a smaller milling bit is "cleaning up" areas a larger mill can't cut.

> What do you think?

_Originally posted by @c- in https://github.com/pcb2gcode/pcb2gcode/issues/413#issuecomment-640053841_

eyal0, 2020-06-12:
Uh-oh, I see something bad in the gcode.  Looks like the bg::simplify is running too many times.  Because of how the [simplify algorithm](https://en.wikipedia.org/wiki/Ramer%E2%80%93Douglas%E2%80%93Peucker_algorithm) works, if you simplify two lines that overlap, you can get two lines that *almost* overlap.  For pcb2gcode, that looks like two paths that run next each other, sometimes overlapping, sometimes crossing, just by a little.

That's horrible for the euler path because it causes the mill head to wind around a bunch and zipzag trying to capture all these little lines!  Worse yet, it'll often have to lift and lower to deal with it.  That sucks really bad.

I'll try to only optimize at the end and hopefully that will solve the issue.

closed 2020-06-12 by eyal0

c-, 2020-06-12:
Whoa... that *dramatically* cleans up that via test case.

[image]

... and for a less trivial board... well, here's the bCNC statistics before and after:

[image]
[image]

At first glance, that seems to eliminate most of the sharp direction changing that was the impetus for #417.

eyal0, 2020-06-12:
Yup, just wait until the backtracking is in place, it'll do a lot less retracting.

commit ea9b88b3 (2020-06-12): Don't simplify in the middle of generation.

It causes overlapping paths to stop being overlapping.
This makes worse eulerian paths.

This fixes #425 .