pcb2gcode/pcb2gcode #453: Add windows-latest to the CI

eyal0, 2020-06-28:
Github CI supports windows now.  It would be nice to do a Windows build in CI and export a binary so that we can put it on the releases page.  I have no experience in building this code under windows.

palyla, 2020-08-17:
Hi @eyal0, after a lot of pain I was succeeded with windows build (just local build, not ci script). I've downloaded gerbv you recommended, tried to build but make command was failed on linking stage. I looked into .libs folder and found the necessary library after that wrote a pc file for pkg-config, and pcb2gcode was successfully built.  You can download the build by the link: 
https://yadi.sk/d/K66saKHOMHlUlg
Have a nice day :)

eyal0, 2020-12-16:
I just tried doing it in GitHub CI but it keeps failing.  It says that the same function is defined multiple times during the gerbv compilation.  Can you provide some tips on how you did this?  I'm very curious!  If I can adapt it to the CI then I could have release builds frequently.  Thanks @palyla

eyal0, 2020-12-22:
To test out a windows build, I'm using github CI.  I start up a windows VM in github CI and use tmate to run it interactively.

```
    - name: Setup tmate session
      uses: mxschmitt/action-tmate@v3
```

After I'm in, I run these commands:

```
sed -i 's/^CheckSpace/#CheckSpace/g' /etc/pacman.conf
pacman -Sy
pacman --needed -S bash pacman pacman-mirrors msys2-runtime
pacman -Su
pacman --needed -S base-devel git mingw-w64-x86_64-gcc mingw-w64-x86_64-boost mingw-w64-x86_64-gtkmm
pushd ~
git clone --depth=1 https://github.com/eyal0/gerbv.git
pushd gerbv
sh autogen.sh
./configure --disable-update-desktop-database --prefix=/mingw64
make -j 4
make install
popd
popd
```

eyal0, 2020-12-24:
The problem was in gerbv: https://sourceforge.net/p/gerbv/bugs/254/

commit a475ecb4 (2021-01-03): Add Windows CI build (#493)

This fixes #453
closed 2021-01-03 by eyal0